CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter #3: Can't calibrate - reads 0 amps

BarrettLowe, 2020-03-12:
I'm either missing something in the documentation or something is going on with my board. After I get my calibration tuned in on a 60w light bulb, the amperage is wrong when turning the bulb off. With no current flowing, it reads about 0.2 amps. Turning the light on again yields (accurately) 0.44(ish) amps. Completely unplugging the CT clamp from the board yields 0.2 amps as well.

 I have v1.3 of the 6 channel hardware and haven't altered it in any way. Looks like JP8-11 are not connected. I'm using esphome like so:
```
- platform: atm90e32
    cs_pin: 5
    phase_a:
      voltage:
        name: "Volts A"
        id: "ic1Volts"
        accuracy_decimals: 1
      current:
        name: "CT1 Amps"
        id: "ct1Amps"
        accuracy_decimals: 3
# Voltage using a 12VAC power transformer
      gain_voltage: 7063
# Current calibration using a 100A/50ma SCT-013-000
      gain_ct: 18874
    phase_b:
      current:
        name: "CT2 Amps"
        id: "ct2Amps"
# If the wattage is backwards due to the phase 180 degrees from the clamp direction
#        filters:
#          - multiply: -1.0
      gain_ct: 18874
    frequency:
      name: "Freq"
    line_frequency: 60Hz
    gain_pga: 1X
    update_interval: ${update_time}
```

Any idea what I'm doing wrong here?

CircuitSetup, 2020-03-12:
One thing I see wrong is that phase_b needs a "gain_voltage: 7063", like with phase_a. But that shouldn't affect the current reading, only the power calculation. 

Can you confirm the same thing happens on other current channels or is it just one? If only one check the solder joints for that connector and surrounding caps/resistors. It's possible there is a cold joint. I usually only see this when the gain is turned up.

BarrettLowe, 2020-03-13:
Looks like 1 of the power monitoring ICs is working fine. CT1-3 all have this issue while 4-6 do not. Didn't see any cold joints but I ran around both ICs just cleaning things up and putting a little pressure down. Seems to have fixed it for now!

closed 2020-03-13 by BarrettLowe