CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter #4: Rev 1.4 add-on board and GPIO35

presslab-us, 2020-03-28:
The rev 1.4 add-on board has changed to GPIO35 for one of the chip select signals.  This GPIO cannot be used because it is an input only pin on the ESP32.

It's not a problem for me, I just moved to another chip select.  But I thought I should mention it.

CircuitSetup, 2020-03-28:
Thank you, I actually just realized this myself. Itll be corrected in the next revision, probably back to GPIO 2, since there are no other available pins that aren't already used.

presslab-us, 2020-04-06:
It looks like on my ESP32 board, GPIO 2 is used for the blue LED.  In EmonESP this blinks when WiFi is connected.  I have disabled that blinking in code and soldered a jumper from GPIO 35 to GPIO 2.  It seems to work fine; I can even see the LED dim slightly as the CS is toggled.  I haven't looked with a scope, but I'd assume the load of the LED has little effect on the signal integrity.

presslab-us, 2020-04-07:
With GPIO 2 being used for CS, I now see the problem where I cannot reprogram the ESP32 through the USB port while it is plugged into the add-on board.  The source of the problem seems to be that the CS pin is pulled high with a 10k resistor on the add-on board, and this messes up the ESP32 bootloader.

CircuitSetup, 2020-04-12:
Is GPIO 13 or 2 causing the boot issue when pulled high?

presslab-us, 2020-04-12:
GPIO 2 causes the boot issue.  I have disconnected GPIO 35.  The documentation below states that "GPIO2 must also be either left unconnected/floating, or driven Low, in order to enter the serial bootloader."
https://github.com/espressif/esptool/wiki/ESP32-Boot-Mode-Selection

CircuitSetup, 2020-05-08:
GPIO 35 was changed to GPIO 2 in the next revision of the add-on board. I'll make a note to disconnect the jumper for 2 if programming the esp32.

presslab-us, 2020-05-08:
That sounds good.  It also might make sense to swap GPIO 35 for GPIO 15.  And then make the last add-on board GPIO 2.  This way only someone with all 6 add-on boards will need to disconnect the jumper for programming.

CircuitSetup, 2020-05-27:
That would have made more sense. For now its on GPIO 2 in the 3rd position on the jumper. Maybe I'll make that change in the next revision. Closing this for now.

closed 2020-05-27 by CircuitSetup