CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter #6: Using ethernet enabled ESP32 boards with this board?

Blackra1n5, 2020-04-06:
I would prefer using ethernet instead of wifi for reliability.

Would the [WESP32](https://wesp32.com/hardware/) or [Olimex's ESP32-POE-ISO](https://www.olimex.com/Products/IoT/ESP32/ESP32-POE-ISO/resources/ESP32-POE-ISO-GPIO.png) be used with the 6 Channel Energy Meter?
And is there a specific pins I should be using?

I have linked the GPIO pinouts of the 2 boards.

Thank You

presslab-us, 2020-04-06:
I'm somewhat interested in this too.  I would prefer Ethernet as the unit is very close to my main network switch.

Looking at the pinout of both of those boards, it seems like neither would be a drop-in.  I guess it could be sky-wired or an adapter board made.

CircuitSetup, 2020-04-07:
@Blackra1n5 neither of those would directly plug into the meter since the pins are different. You could wire up the SPI pins and power with jumpers, but that would be messy unless you made some kind of pcb adapter.

This looks like it could work though: https://www.electrodragon.com/product/lan8720-ethernet-breakout-board-phy/  but you'd have to solder on different headers to plug into the meter and the esp32.

presslab-us, 2020-04-07:
Looking at the code here:
https://github.com/espressif/arduino-esp32/blob/0f772270fbb32099cad72053755d6b4c7e8e1f82/tools/sdk/include/ethernet/eth_phy/phy.h#L34

I think the RMII pins will conflict with the SPI pins.

robotsflo, 2020-04-12:
The ATM90E32AS can be used with SPI or UART, the UART can be the solution with Ethernet LAN8720 ?

CircuitSetup, 2020-04-14:
> Looking at the code here:
> https://github.com/espressif/arduino-esp32/blob/0f772270fbb32099cad72053755d6b4c7e8e1f82/tools/sdk/include/ethernet/eth_phy/phy.h#L34
> 
> I think the RMII pins will conflict with the SPI pins.

Yes, it would, since SPI uses 19 for MISO, which could be changed if you don't mind soldering a wire on the pcb, and changing it in software.

CircuitSetup, 2020-04-14:
> The ATM90E32AS can be used with SPI or UART, the UART can be the solution with Ethernet LAN8720 ?

> The ATM90E32AS can be used with SPI or UART, the UART can be the solution with Ethernet LAN8720 ?

It wouldn't matter since the pcb is already setup for SPI. Either way you you would have to solder in a new wire or 2.

robotsflo, 2020-04-14:
> Yes, it would, since SPI uses 19 for MISO, which could be changed if you don't mind soldering a wire on the pcb, and changing it in software.

Would you have some shematics and code of this tweak ? I think this would be valuable for people will be interested by ethernet feature.

CircuitSetup, 2020-06-25:
> Would you have some shematics and code of this tweak ? I think this would be valuable for people will be interested by ethernet feature.

Sorry I didn't reply to your question earlier. Honestly, I wouldn't recommend doing this just yet. I've been researching this for a new meter, and there are several things that have to work correctly. 

If you can wait until the new meter (it's actually being made for 3-phase power, but can be configured for split single phase), it will have a built in ethernet jack + an on board ESP32. I don't have an ETA, but I am going to start testing next week.

closed 2020-06-25 by CircuitSetup