CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter #7: Power supply glitch

presslab-us, 2020-04-09:
I have two energy meters, one with no add-on boards and one with 3 add-on boards.  On the unit with 3 add-on boards the unit has been locking up and would require a power cycle to work again.  The other unit has not locked up.  I suspected the difference was the current draw from one vs the other.

In diagnosing this issue I have been looking at the power supplies as a source of the problem.  I have captured the signals with my scope as shown below.  Yellow is the 3.3V at C36 (AC coupled).  Red is the rectified incoming voltage at C33.  Blue is the un-rectified voltage at J4.  The lower yellow trace is a zoom of the glitch event.

It can be seen that the 3.3V supply has a glitch due to the voltage at C33 dropping too far.  This seems to happen during a period of high current draw (likely WiFi transmission).  The incoming voltage in blue seems fine so I don't believe it is a line voltage problem.

[image]

I have added an additional 100uF to C33 and I have not observed a lock up after that.  To be clear the captured waveforms are with an unmodified unit.

I know board space is limited but I'm actually not even sure an additional 100uF is sufficient.  The industry accepted standard for hold up time is one cycle, or 16.7 ms at 60 Hz (even worse at 50 Hz).  If the input power lost a cycle (e.g. switching to battery backup) while the WiFi were transmitting, I would guess 200uF is not sufficient for this scenario.  In addition, it is possible to add 3 more add-on boards, further increasing the current draw and the need for more capacitance.

I hope my comments do not come across as overly critical.  Thanks for your time!

CircuitSetup, 2020-04-09:
Thanks so much for detailing this. I had observed a dip in my own testing, but it wasnt causing a problem. I may have overlooked testing with as many add-ons as you have while transmitting data over wifi.

Have you tried putting the cap in parallel with C36? It may be more efficient this way (and you wouldnt need a higher voltage cap like C33).

presslab-us, 2020-04-09:
I have not tried that, but in terms of energy content I think it would be a wash.  The higher voltage cap would have less capacitance but it would be offset by the lower current drawn from it because of the buck converter.  The disadvantage of putting more capacitance in parallel with C36 is that the 3.3V supply would still fall out of regulation, whereas with C33 the 3.3V would remain stable.

CircuitSetup, 2020-04-09:
Right, that makes sense. I'll have to do some more testing of this.

Just thinking about it a bit - it would be a lot easier to add an additional cap on the 3.3v line on each add-on board. This way total capacitance would increase as more add-ons/load is added. 

I'll also test using a 220uf or 300uf cap in place of the 100uf c33. 2 seperate caps may perform better as you have it though.

CircuitSetup, 2020-04-30:
Is this still causing your esp32 to reset, or was that resolved with the other code changes?

presslab-us, 2020-05-01:
This problem was causing a lock up and not a reset.  I have not seen one lock up since adding additional capacitance.  Before, it would usually lock up after only a few hours.

CircuitSetup, 2020-05-04:
Sorry, I meant to say lock up. Did you happen to capture the interval at which the glitch was happening? I'm trying to figure out if this was the wifi transmitting (or maybe even SPI communicating), and if the brownout detector was triggered or not. I can't seem to reproduce the same conditions. 

I've previously observed that the esp32 will eat more power in AP mode (if it doesn't connect to the main AP after so many tries, it'll go into AP mode), which is another factor in power consumption.

presslab-us, 2020-05-04:
The interval wasn't necessarily consistent.  I think it happened when a few things lined up.  I could see my scope trigger (at the minimum) maybe every 30 minutes or so.

Looking at the voltage noise on the 3.3V supply and the rate that the bus capacitor is falling, it looks like this occurs during high current demand.  The thing I can think of that draws the highest current is the WiFi transmitter.

Maybe you have a bit higher line voltage, or lower transformer ratio?

CircuitSetup, 2020-09-03:
FYI, I believe this issue is solved with the addition of a 100uf cap on the 3.3V line going into the add-on boards.

closed 2020-09-03 by CircuitSetup