CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter #11: EmonESP AP Mode and Watchdog

CircuitSetup, 2020-05-09:
If the ESP32 goes into AP mode, the watchdog is triggered because [WIFI_CLIENT_DISCONNECT_RETRY ](https://github.com/CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter/blob/d6541ecf00eec54e139f56ebfdceabcc1d8c3c4a/Software/EmonESP/src_6chan/wifi.h#L92)is set to 10 seconds. 

Need to write some code to handle the watchdog if WIFI_CLIENT_DISCONNECT_RETRY > 5000 (5 seconds)

presslab-us, 2020-05-14:
Maybe it would be a good idea to get PlatformIO working instead of the Arduino IDE?  That way these defines could be set as a compiler option.

CircuitSetup, 2020-05-14:
I've been thinking about that. The WDT was also being triggered by async_tcp, and setting [CONFIG_ASYNC_TCP_USE_WDT](https://github.com/me-no-dev/AsyncTCP/blob/ca8ac5f919d02bea07b474531981ddbfd64de97c/src/AsyncTCP.h#L36) to 0 prevents it from being triggered.

caronte-crc, 2020-09-18:
I found a problem installing src_6chan on an esp32. It reboots after wifi-setup.
The error on the monitor serie say "task watchdog got triggered. the following tasks did not reset the watchdog in time"
Looking for the source of the error found that just commenting out a delay fix the problem.
…
  // Bring up the web server
  web_server_setup();
  //delay(500); <-- Fix continuos reboot.
…

closed 2021-02-04 by CircuitSetup