CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter #12: Formatting of MQTT data causes socket error and disconnection from MQTT broker

CircuitSetup, 2020-05-14:
The formatting of mqtt_topic_prefix, I believe, is causing a disconnection from the MQTT broken.  Have to investigate this further.

CircuitSetup, 2020-05-14:
@presslab-us were you seeing this on your end while testing? I'm seeing random characters in what is being sent. I'm guessing one of the arrays is getting filled with null and need some kind of termination instead. 

I reverted to the previous code (with Strings) in mqtt_publish, and it worked without issue.

presslab-us, 2020-05-15:
I checked out master and tested it and I didn't see any problems.  What text are you using for base topic and prefix?  I am using "emon/main" for base topic, and I tested both "" and "123" for prefix.

In any case, I did see a potential bug.  Because `sprintf()` has variable arguments the compiler may not know to properly cast the `String` to a `char *`.  Why don't you try this and see if it helps:
```diff
diff --git a/Software/EmonESP/src_6chan/mqtt.cpp b/Software/EmonESP/src_6chan/mqtt.cpp
index c80b2fd..c5a9f57 100644
--- a/Software/EmonESP/src_6chan/mqtt.cpp
+++ b/Software/EmonESP/src_6chan/mqtt.cpp
@@ -101,7 +101,7 @@ void mqtt_publish(const char * data)
 {
   const char * data_ptr = data;
   char * topic_ptr = mqtt_topic_prefix;
-  topic_ptr += sprintf(mqtt_topic_prefix, "%s/%s", mqtt_topic, mqtt_feed_prefix);
+  topic_ptr += sprintf(mqtt_topic_prefix, "%s/%s", mqtt_topic.c_str(), mqtt_feed_prefix.c_str());
 
   do
   {
```

CircuitSetup, 2021-02-04:
this was fixed in the latest version

closed 2021-02-04 by CircuitSetup