CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter #14: How to properly connect current transformers to each input

manutenfruits, 2020-05-27:
Hi, I want to use the 6 channel meter to measure the following:

 - Input of the house, split phase (2x clamps)
 - Washer/Dryer breaker, split phase (2x clamps)
 - breaker #3 (1 clamp)
 - breaker #4 (1 clamp)

I started by connecting it as follows:

- Chip 1:
    - Phase A: whole house phase 1
    - Phase B: whole house phase 2 (inverted clamp).
    - Phase C: breaker #3 
- Chip 2:
    - Phase A: washer/dryer phase 1
    - Phase B: washer/dryer phase 2 (inverted clamp).
    - Phase C: breaker #4 

I get proper readings for the whole house and for the washer/dryer, but I get 0A for the two separate breakers. I want to understand what I'm doing wrong before I open the electric panel again.

Is the third measurement on each chip supposed to be in the same phase as the other two? I took it off the panel and tried each input separately with a resistive load and it reads correctly on every port/input, and power is always a positive number.

CircuitSetup, 2020-05-27:
What software are you using? If ESPHome, can you post your config?

manutenfruits, 2020-05-27:
For more context, I ended up opening the panel again and reversed those CT clamps that read negative, even if it didn't make much sense to me. I started pointing the arrow towards the load on those clamps that were on the same phase, and reversed in those that were on the other phase, but I think I had to make some adjustments. When I was done, I was getting readings on the last two breakers (both on phase C of their respective chips), but after less than an hour they went to zero (and I had already closed the panel).

Here's the relevant part of the config:
```

spi:
  clk_pin: 18
  miso_pin: 19
  mosi_pin: 23
sensor:
  - platform: atm90e32
    cs_pin: 5
    phase_a: # Black Input
      voltage:
        name: "Voltage A"
        id: voltage_a
      current:
        name: "Current A"
        id: current_a
      power:
        name: "Active Power A"
        id: active_power_phase_a
      reactive_power:
        name: "Reactive Power A"
      power_factor:
        name: "Power Factor A"
      gain_voltage: 7314
      gain_ct: 21111
    phase_b: # Washer/Dryer black
      voltage:
        name: "Voltage Washer/Dryer A"
      current:
        name: "Current Washer/Dryer A"
        id: active_power_washer_a
      power:
        name: "Active Power Washer/Dryer A"
      gain_voltage: 7362
    phase_c: # NID
      voltage:
        name: "Voltage NID"
      current:
        name: "Current NID"
      power:
        name: "Active Power NID"
      gain_voltage: 7314
      gain_ct: 21111
    frequency:
      name: "Line 1 Frequency"
    line_frequency: 60Hz
    gain_pga: 2X
    chip_temperature:
      name: "Temperature"
    update_interval: 5s
  - platform: atm90e32
    cs_pin: 4
    phase_a: # Red Input
      voltage:
        name: "Voltage B"
        id: voltage_b
      current:
        name: "Current B"
        id: current_b
      power:
        name: "Active Power B"
        id: active_power_phase_b
      reactive_power:
        name: "Reactive Power B"
      power_factor:
        name: "Power Factor B"
      gain_voltage: 7314
      gain_ct: 21111
    phase_b: # Washer/Dryer red
      voltage:
        name: "Voltage Washer/Dryer B"
      current:
        name: "Current Washer/Dryer B"
        id: active_power_washer_b
      power:
        name: "Active Power Washer/Dryer B"
      gain_voltage: 7362
    phase_c: # Aux/Kitchen GFCI
      voltage:
        name: "Voltage Kitchen GFCI"
      current:
        name: "Current Kitchen GFCI"
      power:
        name: "Active Power Kitchen GFCI"
      gain_voltage: 7362
    frequency:
      name: "Line 2 Frequency"
    line_frequency: 60Hz
    gain_pga: 2X
    update_interval: 5s
```

CircuitSetup, 2020-05-27:
Your config looks okay (btw, you probably only need one voltage reported - otherwise you're just reporting the same number over and over, and using up ESP32 memory)

It sounds like the physical connection isn't being made very well with the CTs on 3 and 4 (phase c IC1, and phase a IC2). Can you take a close look at the 2 pins on the 3.5mm jacks on the same side as the "+" symbol on the pcb? The opposite pins are not used, other than to hold the jacks on the board. They may be a cold solder joints that I missed in QA.

manutenfruits, 2020-05-28:
I did a pass on the 3.5mm joints when I first got the board because one of them almost came off when I put the jack in, and I checked again and they seem fine. I think it must be something making a weak contact somewhere else, I swapped the phase c jacks for testing and seemed to work so I left it that way.

closed 2020-05-28 by manutenfruits