CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter #15: [power_factor] is an invalid option for [phase_b]. Please check the indentation

elyorkhakimov, 2020-08-06:
I have ran into indentation issue and can't seem to resolve it.
Here is a portion of YAML file I'm trying to run:

```
#IC1 Main
  - platform: atm90e32
    cs_pin: 5
    phase_a:
      voltage:
        name: ${disp_name} Volts A
        id: ic1Volts
        accuracy_decimals: 2
      current:
        name: ${disp_name} CT1 Amps
        id: ct1Amps
# The max value for current that the meter can output is 65.535. If you expect to measure current over 65A,
# divide the gain_ct by 2 (120A CT) or 4 (200A CT) and multiply the current and power values by 2 or 4 by uncommenting the filter below
#        filters:
#          - multiply: 2
      power:
        name: ${disp_name} CT1 Watts
        id: ct1Watts
#        filters:
#          - multiply: 2
      gain_voltage: ${voltage_cal}
      gain_ct: ${current_cal}
    phase_b:
      current:
        name: ${disp_name} CT2 Amps
        id: ct2Amps
      power:
        name: ${disp_name} CT2 Watts
        id: ct2Watts
      reactive_power:
        name: CT2 Reactive Power
#       power_factor:
#        name: CT2 Power Factor
      gain_voltage: ${voltage_cal}
      gain_ct: ${current_cal}
    phase_c:
      current:
        name: ${disp_name} CT3 Amps
        id: ct3Amps
      power:
        name: ${disp_name} CT3 Watts
        id: ct3Watts
      gain_voltage: ${voltage_cal}
      gain_ct: ${current_cal}
    frequency:
      name: ${disp_name} Freq A
    line_frequency: 60Hz
    gain_pga: 1X
    update_interval: ${update_time}
```

I'm getting this output when trying to run it:

```
root@tuyaconvert:/home/user# esphome 6chan_energy_meter-add-on_jp8-jp11.yaml run
INFO Reading configuration 6chan_energy_meter-add-on_jp8-jp11.yaml...
INFO Detected timezone 'UTC' with UTC offset 0
Failed config

sensor.atm90e32: [source 6chan_energy_meter-add-on_jp8-jp11.yaml:65]
  platform: atm90e32
  cs_pin: 5
  phase_a:
    voltage:
      name: cs_nrg_mtr Volts A
      id: ic1Volts
      accuracy_decimals: 2
    current:
      name: cs_nrg_mtr CT1 Amps
      id: ct1Amps
    power:
      name: cs_nrg_mtr CT1 Watts
      id: ct1Watts
    gain_voltage: 7305
    gain_ct: 27518
  phase_b:  [source 6chan_energy_meter-add-on_jp8-jp11.yaml:87]
    current:
      name: cs_nrg_mtr CT2 Amps
      id: ct2Amps
    power:
      name: cs_nrg_mtr CT2 Watts
      id: ct2Watts

    [reactive_power] is an invalid option for [phase_b]. Please check the indentation.
    reactive_power:  [source 6chan_energy_meter-add-on_jp8-jp11.yaml:94]
      name: CT2 Reactive Power
    gain_voltage: 7305
    gain_ct: 27518
  phase_c:
    current:
      name: cs_nrg_mtr CT3 Amps
      id: ct3Amps
    power:
      name: cs_nrg_mtr CT3 Watts
      id: ct3Watts
    gain_voltage: 7305
    gain_ct: 27518
  frequency:
    name: cs_nrg_mtr Freq A
  line_frequency: 60Hz
  gain_pga: 1X
  update_interval: 10s
root@tuyaconvert:/home/user#
```

What am I doing wrong?

elyorkhakimov, 2020-08-06:
Never mind, turns out I needed to run ESPhome dev branch

CircuitSetup, 2020-08-06:
@elyorkhakimov yes, power factor and a couple others are still in the dev build.

closed 2020-08-06 by CircuitSetup