CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter #17: Expandable 6 Channel ESP32 Energy Meter Add-on Board

timwing, 2020-09-03:
I got my Expandable 6 Channel ESP32 Energy Meter Main Board up and running well then realized I needed more channels :>) I added the Expandable 6 Channel ESP32 Energy Meter Add-on Board that is providing data on channels 10, 11 & 12 but not giving data on on channels 7, 8 & 9. I am using the supplied "6chan_energy_meter-add-on_jp8-jp11.yaml" and have tried moving my CT's around but still unable to get anything on 7, 8 & 9. What could I be doing wrong?

I am using ESPHome and viewing data via HomeAssistant. Any feedback would be much appreciated.

timwing, 2020-09-03:
@CircuitSetup, That was it! Looks like 5 of the 6 CS (outboard side) pins were not connected at the board. All fixed and working now. Thank you!

Tim

timwing, 2020-09-06:
@CircuitSetup So I am having some additional issues with the boards. I have had two of the 3.5mm jacks come off the board...got all those soldered back on to the main & add-on boards. Then everything was working until a day later when I had the 9-12vac come off the main board. Now that is soldered back on but now I am unable to maintain power to the unit. Works then it shuts off.

CircuitSetup, 2020-09-06:
Hum, this definitely shouldn't have happened. Are both of your boards green?

timwing, 2020-09-06:
Are you referring to the LED or the color of the printed circuit board?

Both of my printed circuit boards are green.

CircuitSetup, 2020-09-06:
Yes, the circuit board. I ask because there is a new batch of black boards that were manufactured out of house.

For the green ones, these were in house. I'm wondering if I overlooked your board in QA. When it shuts off does the green led turn off too, or just the esp32?

timwing, 2020-09-06:
Well now I get no lights on either board or esp32. I do have 11.05vac in the transformer plug.

CircuitSetup, 2020-09-06:
When the transformer plug came off, did it take any trace with it? 

If you measure the output (there is a + and - over the output pins) of the black component next to the plug you should get a DC voltage. 

Either way, if its dead, ill gladly send you a new main board. Let me know your order number if thats the case.

timwing, 2020-09-06:
I am a little discouraged by the parts connections coming loose. At this point I think replacement would be the best option. Order #200054

CircuitSetup, 2020-09-06:
You got it. I'll send a new main board asap. It'll be black, but is exactly the same as the green.

CircuitSetup, 2020-09-08:
This is in the mail!

closed 2020-09-08 by CircuitSetup