CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter #18: Issue with <esp_task_wdt.h> and a Wifi Redeclaration

errolsancaktar, 2020-09-21:
Just pulled newest version and trying to build via platformio.  getting an error that it cant find <esp_task_wdt.h> which doesnt make sense since its part of arduino-esp32.
It appears to be looking for the feedLoopWDT which i dont see declared anywhere:
```src/http.cpp:78:19: error: 'feedLoopWDT' was not declared in this scope
Also getting two other interesting issues that dont seem to make sense.

```src/http.cpp:59:51: error: no matching function for call to 'BearSSL::WiFiClientSecure::connect(const char*&, int&, int)'

CircuitSetup, 2020-09-21:
I just updated the platformio.ini file and got everything to build (along with some other improvements). I'm not quite sure why this was happening. 

You can ignore the warnings that are output while building.

errolsancaktar, 2020-09-23:
That did work for me as well however now i have no mqtt.  shows connected yes for a couple seconds then switches back to no,.  nothing posts to mqtt.

errolsancaktar, 2020-09-23:
Also is there any logging methodology over the network?

CircuitSetup, 2020-09-23:
@errolsancaktar did you update spiffs as well? There is a new setting for json based mqtt that may need to be set.

CircuitSetup, 2020-09-23:
As far as logging, I was thinking of putting in some kind of virtual terminal so you can see what is going on remotely. Would that work?

errolsancaktar, 2020-09-23:
Yeah that would be super helpful. If you want help with a pr lemme know. Also I'm guessing no way to upload spiffs without pulling from breaker panel?

CircuitSetup, 2020-09-23:
You should be able to upload the spiffs.bin via the web interface. 

I'm working on getting a debug output in the interface.

errolsancaktar, 2020-09-27:
i'm getting an 'update FAIL' when i try to upload spiffs.bin.  is there another step i should take?

CircuitSetup, 2020-10-02:
how was the spiffs.bin compiled? 

I'm still working on the debug output.

errolsancaktar, 2020-10-02:
Just through platformio no changes to code base

CircuitSetup, 2021-03-23:
@errolsancaktar did you ever figure out what was wrong with this? Updating the 6 channel code has been stalled lately.