CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter #20: Chip selects in the 36 channel example

sillygoose, 2020-11-05:
These don't appear to be right, i get an error on the 35 as 
```    ESP32: GPIO35 (34-39) can only be used as an input pin.```

and 27 doesn't appear in the expansion board J4 and J7 schematics pinouts.

```
...
#IC1 AddOn 2
  - platform: atm90e32
    cs_pin: 27  <==== perhaps 2?
...
#IC1 AddOn 3
  - platform: atm90e32
    cs_pin: 35  <==== perhaps 12?
...
```

I ran into some other quirks, not sure if I don't understand something or if my hardware is acting up.  Main board and 3 expansion boards and new ESP32.

First configured with the chip selects for the three boards you sent:
0/16    2/17    12/21

Result:
IO12 did not respond (blank states displayed for the 3 CT inputs on the third expansion board, first atm90e32)

Changed chip selects to: 
0/16   2/17    13/21

Result:
IO13 responded but problem moved to IO2 (blank states displayed for the 3 CT inputs on the second expansion board, first atm90e32) -- weird?

Changed chip selects to 
0/16   2/17   15/21

Result:
IO15 responded but IO2 did not respond (blank states for the 3 CT inputs on the second expansion board, first atm90e32)

Changed chip selects to 
0/16   14/17   15/21

Result:
This set of chip selects works and all states are reported by ESPHome.  I'll live with this for now but sure would like to know if this a software or hardware issue.

CircuitSetup, 2020-11-06:
What version of the add-on boards do you have? The pin assignments changed a bit over time because, like GPIO 35, you can't use it as a CS pin. That was my mistake. 

For the current pin assignments, take a look here: https://github.com/CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter#add-on-boards

sillygoose, 2020-11-07:
I have v1.4 rev 1 boards, at least the ones I have access to.  The last order I got from you is packed away in a case and I prefer not to mess with it.

Looking at the silkscreen for CS chip selects I see, 0, 27, 2? (hidden under header), 13, 14, and 15.  Since the schematic is v1.3 rev 1 perhaps they have changed?  I started with the chip selects in your example before digging into the schematic and trying those values.

CircuitSetup, 2021-02-04:
Sorry I didn't see your last question sooner. Yes, the CS pins changed a bit between 1.3 and 1.4 rev1. See the link above. Let me know if you have any other questions!

closed 2021-02-04 by CircuitSetup