CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter #21: Information flipping back and forth for live data

errolsancaktar, 2020-11-21:
[image]
[image]

i'm getting data flapping back and forth for some reason.  The voltage also seems to be way off.  i have it configured for 
ct1: Input
CT2: Input
CT3: Solar input
CT4: Solar input
CT5: AC
CT6: Dryer

errolsancaktar, 2020-11-21:
strangely my mqtt is only showing the topic with connected, no sub data.

CircuitSetup, 2020-11-21:
Is this with the latest version available here? If so, can you confirm the calibration values?

errolsancaktar, 2020-11-22:
I pulled from the master branch.  so i should turn the ones that show negative amps around?  also have you seen any other instances of the CT calibration selector not showing up?  (sometimes i see a jquery error in console)
CT1/2 - 120A - 41880
CT3/4 - 100A - 27961
CT5/6 - 50A - 15420

CircuitSetup, 2020-12-08:
Sorry, I overlooked your last response! 

Yes, anything showing as negative should be turned around. 

It's possible that sometimes the files aren't loading properly for the interface due to connection or memory issues, or there is a bug somewhere. I haven't seen anything, nor have I seen anything reported.

errolsancaktar, 2020-12-08:
i'll swap out the esp32 and flash new to it.  also CT3 and 4 are for the solar panels, should those show negative or positive?

errolsancaktar, 2020-12-09:
Tried new board and also flipped the cts, no change.  Is it possible it's the board?

CircuitSetup, 2020-12-09:
You're still getting negative values even though the CTs were flipped? This would only happen on solar when you are exporting energy that you aren't using. For individual circuits it should never be negative, because either way that circuit is consuming power, whether it be from the grid or from solar. 

The only way it could be the board is if the readings are 0 or 65.535. It sounds like the issue is in the software somewhere.  However, I've never seen temp and frequency off like they are in the first post. 

Are you able to switch to ESPHome to see if you get similar readings?

errolsancaktar, 2020-12-09:
i'm not sure what you mean by switching to esphome sorry

CircuitSetup, 2020-12-09:
ESPHome is another set of software that will work with the meter. If you don't have Home Assistant running, it is a bit of a pain to set up. 

However, I see that the readings for one side of the meter are all 65535, This indicates that something isn't communicating with the ESP32 properly or something is wrong with the meter.  It may be easier to send you a new one to test. Do you have your original order number handy?

errolsancaktar, 2020-12-09:
Is this it? 188072

CircuitSetup, 2020-12-10:
No, but I found your order. I'll get you a new one (v1.4 rev1, black pcb) mailed out today.

CircuitSetup, 2021-02-04:
Closing this since a new meter solved the issue.

closed 2021-02-04 by CircuitSetup