CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter #23: Voltage and VAR Discrepancy/Non-Zeros on no current

drewdavis1, 2021-02-04:
Just got my 2x 6 channel setup (all components from CircuitSetup store) and wondering about some data I'm getting. Not sure if I'm missing something or what. My setup is with a 60W incandescent light bulb for test and calibration.  Kill-A-Wat is reading 0.43A, 51.8W @ 118.2V.  All of the CT clamps are on the hot wire for the bulb for testing.

CT1 - CT2: 200A CT (SCT-024)
CT3 - CT12: 80A CT (SCT-010)

As you can see in the first screenshot, the Current and Power look pretty good, however the Voltage and Reactive Power have quite a range. 
[image]

In the second screenshot, this is with the bulb completely unplugged, no current at all. As you can see, quite a few non-zero values. 
[image]

Questions:
 1. What would cause the VAR to have such a swing in readings from the
    same source? 
 2. The variance in voltage, is that just due to the
    polling frequency of ESPHome? With just the one reference voltage, I was expecting  the ports on the same chip to read the same voltage
 3. With no current, what would cause so many values to be non-zero?

ESPHome YAML below for reference.

Thanks in advance!

```yaml
substitutions:
  device_name: power_panel_meter
  friendly_name: Power Panel Meter
  update_time: 10s
  voltage_cal: '7344'
esphome:
  name: ${device_name}
  platform: ESP32
  board: nodemcu-32s
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
logger:
api:
ota:
web_server:
spi:
  clk_pin: 18
  miso_pin: 19
  mosi_pin: 23
#IC1 Main
  - platform: atm90e32
    cs_pin: 5
    line_frequency: 60Hz
    gain_pga: 1X
    update_interval: ${update_time}
    frequency:
      name: ${friendly_name} Port 1-3 Frequency
    chip_temperature: 
      name: ${friendly_name} Port 1-3 Chip Temp
    #Port 1
    phase_a:
      voltage:
        name: ${friendly_name} CT1 Voltage
      current:
        name: ${friendly_name} CT1 Current
        accuracy_decimals: 3
      power:
        name: ${friendly_name} CT1 Power 
      reactive_power:
        name: ${friendly_name} CT1 Reactive Power
      power_factor:
        name: ${friendly_name} CT1 Power Factor   
      gain_voltage: 7344
      gain_ct: 29532  
    #Port 2 
    phase_b:
      voltage:
        name: ${friendly_name} CT2 Voltage
      current:
        name: ${friendly_name} CT2 Current
        accuracy_decimals: 3
      power:
        name: ${friendly_name} CT2 Power 
      reactive_power:
        name: ${friendly_name} CT2 Reactive Power
      power_factor:
        name: ${friendly_name} CT2 Power Factor     
      gain_voltage: 7344
      gain_ct: 34105
    #Port 3
    phase_c:
      voltage:
        name: ${friendly_name} CT3 Voltage
      current:
        name: ${friendly_name} CT3 Current
        accuracy_decimals: 3
      power:
        name: ${friendly_name} CT3 Power
      reactive_power:
        name: ${friendly_name} CT3 Reactive Power
      power_factor:
        name: ${friendly_name} CT3 Power Factor
      gain_voltage: 7344
      gain_ct: 41660
#IC2 Main
  - platform: atm90e32
    cs_pin: 4
    line_frequency: 60Hz
    gain_pga: 1X
    update_interval: ${update_time}
    frequency:
      name: ${friendly_name} Port 4-6 Frequency
    chip_temperature: 
      name: ${friendly_name} Port 4-6 Chip Temp
    #Port 4
    phase_a:
      voltage:
        name: ${friendly_name} CT4 Voltage
      current:
        name: ${friendly_name} CT4 Current
        accuracy_decimals: 3
      power:
        name: ${friendly_name} CT4 Power 
      reactive_power:
        name: ${friendly_name} CT4 Reactive Power
      power_factor:
        name: ${friendly_name} CT4 Power Factor 
      gain_voltage: 7344
      gain_ct: 41660
    #Port 5 
    phase_b:
      voltage:
        name: ${friendly_name} CT5 Voltage
      current:
        name: ${friendly_name} CT5 Current
        accuracy_decimals: 3
      power:
        name: ${friendly_name} CT5 Power 
      reactive_power:
        name: ${friendly_name} CT5 Reactive Power
      power_factor:
        name: ${friendly_name} CT5 Power Factor  
      gain_voltage: 7344
      gain_ct: 41660
    #Port 6
    phase_c:
      voltage:
        name: ${friendly_name} CT6 Voltage
      current:
        name: ${friendly_name} CT6 Current
        accuracy_decimals: 3
      power:
        name: ${friendly_name} CT6 Power
      reactive_power:
        name: ${friendly_name} CT6 Reactive Power
      power_factor:
        name: ${friendly_name} CT6 Power Factor
      gain_voltage: 7344
      gain_ct: 41660
#IC1 AddOn 1
  - platform: atm90e32
    cs_pin: 0
    line_frequency: 60Hz
    gain_pga: 1X
    update_interval: ${update_time}
    frequency:
      name: ${friendly_name} Port 7-9 Frequency
    chip_temperature: 
      name: ${friendly_name} Port 7-9 Chip Temp    
    #Port 7
    phase_a:
      voltage:
        name: ${friendly_name} CT7 Voltage
      current:
        name: ${friendly_name} CT7 Current
        accuracy_decimals: 3
      power:
        name: ${friendly_name} CT7 Power 
      reactive_power:
        name: ${friendly_name} CT7 Reactive Power
      power_factor:
        name: ${friendly_name} CT7 Power Factor
      gain_voltage: 7344
      gain_ct: 41660
    #Port 8
    phase_b:
      voltage:
        name: ${friendly_name} CT8 Voltage
      current:
        name: ${friendly_name} CT8 Current
        accuracy_decimals: 3
      power:
        name: ${friendly_name} CT8 Power 
      reactive_power:
        name: ${friendly_name} CT8 Reactive Power
      power_factor:
        name: ${friendly_name} CT8 Power Factor
      gain_voltage: 7344
      gain_ct: 41660
    #Port 9
    phase_c:
      voltage:
        name: ${friendly_name} CT9 Voltage
      current:
        name: ${friendly_name} CT9 Current
        accuracy_decimals: 3
      power:
        name: ${friendly_name} CT9 Power
      reactive_power:
        name: ${friendly_name} CT9 Reactive Power
      power_factor:
        name: ${friendly_name} CT9 Power Factor
      gain_voltage: 7344
      gain_ct: 41660
#IC2 AddOn 1
  - platform: atm90e32
    cs_pin: 16
    line_frequency: 60Hz
    gain_pga: 1X
    update_interval: ${update_time}
    frequency:
      name: ${friendly_name} Port 10-12 Frequency
    chip_temperature: 
      name: ${friendly_name} Port 10-12 Chip Temp  
    #Port 10
    phase_a:
      voltage:
        name: ${friendly_name} CT10 Voltage
      current:
        name: ${friendly_name} CT10 Current
        accuracy_decimals: 3
      power:
        name: ${friendly_name} CT10 Power 
      reactive_power:
        name: ${friendly_name} CT10 Reactive Power
      power_factor:
        name: ${friendly_name} CT10 Power Factor
      gain_voltage: 7344
      gain_ct: 41660
    #Port 11
    phase_b:
      voltage:
        name: ${friendly_name} CT11 Voltage
      current:
        name: ${friendly_name} CT11 Current
        accuracy_decimals: 3
      power:
        name: ${friendly_name} CT11 Power 
      reactive_power:
        name: ${friendly_name} CT11 Reactive Power
      power_factor:
        name: ${friendly_name} CT11 Power Factor
      gain_voltage: 7344
      gain_ct: 41660
    #Port 12
    phase_c:
      voltage:
        name: ${friendly_name} CT12 Voltage
      current:
        name: ${friendly_name} CT12 Current
        accuracy_decimals: 3
      power:
        name: ${friendly_name} CT12 Power
      reactive_power:
        name: ${friendly_name} CT12 Reactive Power
      power_factor:
        name: ${friendly_name} CT12 Power Factor
      gain_voltage: 7344
      gain_ct: 41660
```

CircuitSetup, 2021-02-04:
1. The variance has to do with the different size CTs, gain, a combination of interference, timing of ADC readings, and manufacturing differences in the CTs, and on-board resistors (which are 1%). The 200A CTs are not going to be as accurate at low or 0 power. Physically they are much larger, and depending on what they are next to, may pick up interference. If you disconnect them, and still get a large swings in readings, there may be something wrong with the meter.
2. Each voltage has a separate ADC on the chip. Each chip (you have 4) has a separate set of 1% resistors. This is why CTs 1-3, 4-6, 7-9, 10-12 are all very close, but slightly different from each other. You really don't need more than 1 voltage reading since they're all coming from the same source.
3. Same reason as 1.

FYI, you can add filters to the data in ESPHome - so you could say "if value is 0.01, then 0" https://esphome.io/components/sensor/index.html

Hope this helps! Let me know if there's something that can be explained better.

drewdavis1, 2021-02-04:
OK, great, thank you for the info and quick response. That does help. Also, thank you for making such a fantastic product for the community!

Now for the ultimate question: ESPHome straight to HomeAssistant and use Grafana for a additional analysis OR emonESP with EmonCMS and the HA addons to integrate the data into HA.... haha. 

Thanks!!!

CircuitSetup, 2021-03-22:
You're welcome! Going to close this for now, but if you are still having issues, feel free to open a new issue.

closed 2021-03-22 by CircuitSetup