CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter #24: wrong value power factor !!

oussama415, 2021-02-05:
I use the counter with the ES32 , I have a problem with the power factor sometimes for loads I have the right value, sometimes for other loads I have a lower value !!!
for example i should have 0.98 but the function eic.GetPowerFactorA() return 0.52 
how do i solve this problem?

CircuitSetup, 2021-02-09:
What kind of load are you using to test this? Also, what else are you using to measure the power factor?

oussama415, 2021-02-09:
I put an asynchronous motor 1.1KW it gave me good results also for a 120W led lamp
and when I put a set of 2 to 3 LEDs, at the current level it's ok but for the power factor no.
I compared my measurements with a certified smart meter

CircuitSetup, 2021-02-16:
Are you able to verify the real power and apparent power (VA) of the LED lamp?  The power factor can also be decreased if the current transformer is backwards.

oussama415, 2021-02-18:
I tested the current sensor in both directions, always the same result !!!
Here the example : 

-------------------Phase 1 : ------------------------
Voltage 1 : 218.97V
Current 1 : 0.50A
Active Power 1 : 0.109 KW
Reactive Power 1 : 0.018 KVAR
Apparent Power 1 : 0.110KVA
Power Factor 1 : 0.9870
Déphasage 1 : 0.16
Temp : 28.00°C
Frequency : 50.00Hz
-------------------Phase 2 :  ------------------------
Voltage 2 : 226.38V
Current 2 : 0.33A
Active Power 2 : 0.039 KW
Reactive Power 2 : 0.065 KVAR
Apparent Power 2 : 0.075 KVA
Power Factor 2 : 0.5130
Déphasage 2 : 1.03
Temp : 25.00°C
Frequency : 49.95Hz
-------------------Phase 3 :  ------------------------
Voltage 3 : 218.85V
Current 3 : 0.00A
Active Power 3 : 0.000 KW
Reactive Power 3 : 0.000 KVAR
Apparent Power 3 : 0.000 KVA
Power Factor 3 : 0.0560
Déphasage 3 : 1.51
Temp : 28.00°C
Frequency : 50.00Hz

in phase 2 instead of having 0.98 I had 0.50!

CircuitSetup, 2021-08-15:
We're you able to figure anything out with this? I assume you're reading the current the corresponding voltage phases? If not that will also throw off the Power factor reading.

kgalilio, 2021-08-15:
Hi John,

That's too advanced for me. I am using the same code on github:
https://github.com/CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter/blob/master/Software/ESPHome/6chan_energy_meter-add-on_jp8-jp11.yaml

It worked for me, and I am getting correct values for 3 installations (3
mainboards, and 5 addons). The only problem am having is with CT8. So I
don't think it is a Power factor issue.

CircuitSetup, 2021-08-15:
> Hi John, That's too advanced for me. I am using the same code on github: [...]

This was a different setup than yours with a 3 phase system.