CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter #25: Odd Voltage Reading

johnsethwatkins, 2021-02-08:
I've set up my 6 channel v1.2 board which I bought some time ago.

I bridged JP8-JP11.

I am EU based with nominal 220V supply. The actual measured V is 221v today.

I am using a SCT-013-000 100A clamp

ESPHome yaml below. Measured voltage is showing in logs/Home Assistant as 62.5V.

I would be very grateful for any help in identifying what I have done wrong with the deployment. 

```
substitutions:
# Change the disp_name to something you want  
  disp_name: 6Cenergymonitor
# Interval of how often the power is updated
  update_time: 10s
# Current Transformers:
#  20A/25mA SCT-006: 11143
#  30A/1V SCT-013-030: 8650
#  50A/1V SCT-013-050: 15420
#  80A/26.6mA SCT-010: 41660
#  100A/50ma SCT-013-000: 27518
#  120A/40mA: SCT-016: 41787
#  200A/100mA SCT-024: 27518
#  current_cal: '27518'
  current_cal: '27518'
# Jameco 9VAC Transformer: 
#  For meter versions: 
#  >= v1.3: 7305
  voltage_cal: '37080'

# mqtt:
#  broker: !secret mqtt_broker
#  username: !secret mqtt_user
#  password: !secret mqtt_pass

spi:
  clk_pin: 18
  miso_pin: 19
  mosi_pin: 23

#IC1
sensor:
  - platform: wifi_signal
    name: "WiFi Signal Sensor"
    update_interval: 30s
  - platform: atm90e32
    cs_pin: 5
    phase_a:
      voltage:
        name: ${disp_name} Volts A
        id: ic1Volts
        accuracy_decimals: 1
      current:
        name: ${disp_name} CT1 Amps
        id: ct1Amps
# The max value for current that the meter can output is 65.535. If you expect to measure current over 65A, 
# divide the gain_ct by 2 (120A CT) or 4 (200A CT) and multiply the current and power values by 2 or 4 by uncommenting the filter below
#        filters:
#          - multiply: 2
      power:
        name: ${disp_name} CT1 Watts
        id: ct1Watts
#        filters:
#          - multiply: 2
      gain_voltage: ${voltage_cal}
      gain_ct: ${current_cal}
    phase_b:
      current:
        name: ${disp_name} CT2 Amps
        id: ct2Amps
      power:
        name: ${disp_name} CT2 Watts
        id: ct2Watts
      gain_voltage: ${voltage_cal}
      gain_ct: ${current_cal}
    phase_c:
      current:
        name: ${disp_name} CT3 Amps
        id: ct3Amps
      power:
        name: ${disp_name} CT3 Watts
        id: ct3Watts
      gain_voltage: ${voltage_cal}
      gain_ct: ${current_cal}
    frequency:
      name: ${disp_name} Freq A
    line_frequency: 50Hz
    gain_pga: 2X
    update_interval: ${update_time}
#IC2
  - platform: atm90e32
    cs_pin: 4
    phase_a:
#this voltage is only needed if monitoring 2 voltages
      voltage:
        name: ${disp_name} Volts B
        id: ic2Volts
        accuracy_decimals: 1
      current:
        name: ${disp_name} CT4 Amps
        id: ct4Amps
      power:
        name: ${disp_name} CT4 Watts
        id: ct4Watts
      gain_voltage: ${voltage_cal}
      gain_ct: ${current_cal}
    phase_b:
      current:
        name: ${disp_name} CT5 Amps
        id: ct5Amps
      power:
        name: ${disp_name} CT5 Watts
        id: ct5Watts
      gain_voltage: ${voltage_cal}
      gain_ct: ${current_cal}
    phase_c:
      current:
        name: ${disp_name} CT6 Amps
        id: ct6Amps
      power:
        name: ${disp_name} CT6 Watts
        id: ct6Watts
      gain_voltage: ${voltage_cal}
      gain_ct: ${current_cal}
    frequency:
      name: ${disp_name} Freq B
    line_frequency: 50Hz
    gain_pga: 2X
    update_interval: ${update_time}
        
#Total Amps   
  - platform: template
    name: ${disp_name} Total Amps
    id: totalAmps
    lambda: return id(ct1Amps).state + id(ct2Amps).state + id(ct3Amps).state + id(ct4Amps).state + id(ct5Amps).state + id(ct6Amps).state ;
    accuracy_decimals: 2
    unit_of_measurement: A
    icon: "mdi:flash"
    update_interval: ${update_time}         
#Total Watts
  - platform: template
    name: ${disp_name} Total Watts
    id: totalWatts
    lambda: return id(ct1Watts).state + id(ct2Watts).state + id(ct3Watts).state + id(ct4Watts).state + id(ct5Watts).state + id(ct6Watts).state ;
    accuracy_decimals: 1
    unit_of_measurement: W
    icon: "mdi:flash-circle"
    update_interval: ${update_time}
#kWh
  - platform: total_daily_energy
    name: ${disp_name} Total kWh
    power_id: totalWatts
    filters:
      - multiply: 0.001
    unit_of_measurement: kWh
    
switch:
  - platform: restart
    name: ${disp_name} Restart  
time:
  - platform: sntp
    id: sntp_time
```

CircuitSetup, 2021-02-09:
What are the specs of the AC transformer that you are using, and how did you calibrate it to the meter?

johnsethwatkins, 2021-02-09:
Thanks for the reply. 

I bought the AC transformer at the same time as I bought your board, and given that it's v1.2 I think that was at least 18 months ago if not more. I don't know what brand the transformer is, as to fit it into the fuse board I dremelled it out of the case to mount it into a better shaped 3D printed one. Therefore the brand is lost in time. I know that it was 9V AC rated and at least 500ma as that was your spec at the time and I was shopping for this transformer specifically and not re-using something I already had.

I don't have an oscilloscope to check the waveform, but as it's open I can see it's only a coil without circuitry, so I don't immediately see how the waveform can be affected in the step-down. On a multi-meter it reads 10.1V AC.

For calibration I have the transformer V (10.1) and the incoming supply V (221 yesterday) and I play with the voltage_cal value in esphome code. The highest value the code will accept is 65000, which only gets me to 118V measured.

The A calibration works perfectly, I'm at 26500 to get an accurate reading.

If everything points to the transformer I am happy to buy one of your recommended units listed on your readme, I just wanted to ask your advice before disappearing down that direction. 

For the clamps I'm using (SCT-013-000 100A) I have I don't need to cut the onboard burden resistors do I (v1.2)?

CircuitSetup, 2021-02-15:
So it is only the B Volts are reading correctly, but the B Volts are not? They should be exactly the same assuming you are using 1 transformer, and the voltage jumpers are not cut.

johnsethwatkins, 2021-02-17:
No, both are wrong.

I am feeding the board 10VAC from the transformer. As I write this I have just one clamp in-use, position CT1.

In code I have voltage_cal: '65535', which is the highest espHome will allow.

In both the raw log files and Home Assistant itself I am showing;

VoltsA 119.6V
VoltsB 116.0V

True measured voltage with multimeter on fuseboard is 219V

johnsethwatkins, 2021-02-24:
Good morning,

Any further thoughts on this please?

CircuitSetup, 2021-02-24:
Sorry for not replying sooner. 

Since you have an older meter, the voltage reading is a little different than it is now. The difference between A and B can be accounted for by the variance in resistors. This can be corrected by tuning the calibration. 

To get to 220V, you'll have to double the output in the ESPHome config. That will look like this: 
```
    phase_a:
      voltage:
        name: ${disp_name} Volts A
        id: ic1Volts
        accuracy_decimals: 1
        filters:
          - multiply: 2.0
```

Also, since the voltage isn't able to be brought up via the calibration gain, the power output will also need to be doubled, like this: 
```
      power:
        name: ${disp_name} CT1 Watts
        id: ct1Watts
        filters:
          - multiply: 2
```
for each phase

johnsethwatkins, 2021-02-24:
Fixed, working, happy! Thank you.

closed 2021-02-24 by johnsethwatkins