CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter #27: Board Version Compatibility

johnsethwatkins, 2021-02-25:
I have a v1.2 board up and running.

I have an EU 220V 3 phase supply in my house. I have read that to properly monitor the supply and my 3 phase equipment I need to stack two boards. Two boards will be useful in any event as my supply is not as good as I would like and it is common to find >20v variations across the phases which I'd like to track independently.

If I now purchase a new board from you it will be v1.x (1.4?) that you will send to me.

Does v1.2 stack with the current version (physically and in terms of components), OR, do I need to order two new boards to complete my set-up.

Thanks
John.

CircuitSetup, 2021-02-26:
Since you'd be measuring 3 voltages, it would be best to have all of the same version of meter. While it will work, the calibration for the v1.2 board will be much different from the newer v1.4 meters. 

If you want, I can give you a large discount on a new v1.4 main board since you were an early adopter 😊 Please send me a message via the Contact form on the site, and I'll get you a coupon code.

johnsethwatkins, 2021-02-26:
Many thanks. I have ordered a new board and a new expansion board to go with it.

closed 2021-02-26 by johnsethwatkins