CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter #28: Question about Split Phase 220v

blacktirion, 2021-03-01:
Hello,

I didn't see this explicitly spelled out in the Readme.MD / Wiki:

Do I need two clamps to measure a US based 220v circuit? Is there logic in ESPhome to deal with that? I seem to be getting mixed answers on this. If I need two clamps, do I need to do the severing operation to measure two voltages?

presslab-us, 2021-03-07:
If your 240V circuit is single phase (no current on neutral) you can use one sensor.  If it's a split phase circuit (with a neutral) then you will need two sensors.

Some devices (like dryers) can look like single phase but are really split phase.  On older homes the ground connection is used as the neutral.

I believe with ESPHome you can use a multiply filter to multiply the power results by 2 to get the correct result with one sensor.
https://esphome.io/components/sensor/index.html#offset-multiply

Usually the voltages in split phase are close enough that you don't need two voltage measurements.  That's how IoTaWatt works by default.

CaptClaude, 2021-03-08:
95% of all residential distribution in the US & Canada is split-phase 240Vac. From the neutral to each phase, the voltage _should_ be 120V, but that will depend on a lot of factors, most of which you have no control over. For example, I measured my voltages at my panel last week and they were 110-130. For the purposes of monitoring energy usage, most home monitoring systems you can buy simply assume that the line-to-neutral voltage is 120V and don't bother to measure it.  If you want to be super accurate, measure both phases, but the 120V assumption is pretty accurate.  
Keeping an eye on your line voltages is only important if you are suspecting that sags or transients are causing problems. And the effort required to do that is beyond what most of us are going to build.
[This is a pretty good introduction to the 120/240v residential distribution at your breaker panel.](https://theengineeringmindset.com/120-240v-split-phase-us-can/)

CircuitSetup, 2021-03-08:
> One more thing: You cannot measure the current flowing to a 240V load (dryer, stovetop, over, A/C, etc) with one current transformer (CT) clamped around both hot wires. You'll read zero because the two phase currents cancel each other magnetically. You do need two clamps (properly called CTs), one for each phase for correct measurements. However, since the voltage is the same on each phase (more or less) then the current flowing in each phase should be the same (if it's not, you have a problem) so it seems to me that measuring one phase and doubling the value would work just fine.

This is true, but if there is enough wire available, and the CT is large enough, you can pass both wires through 1 CT in opposite directions. Do not attempt this if the wire is a heavy gauge. Also, in most cases the load is balanced between both phases from the appliance, but some run electronics on 1 phase only. This likely wouldn't be an issue though since these electronics only use mA at a time.

I [updated the readme to include a section on this.](https://github.com/CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter#measuring-dual-pole-240v-circuits)

presslab-us, 2021-03-08:
A couple observations from my installation, regarding 240V appliances:

My dryer pulls about 150 watts from one phase only for the drum motor.  Because my home is older this current flows through the ground wire.  I needed to measure the current from both wires for this appliance.

My range, although it does have a dedicated neutral, only draw a few watts from this for the electronics so I ignored it and only measure one wire.

Because I couldn't route the two conductors through one cable clamp, I used an audio "Y" cable to combine two current sensors into one input.  As with two conductors in one clamp, care must be taken to ensure the measurement range of the input is not exceeded.
https://www.monoprice.com/product?p_id=30895

closed 2021-03-22 by CircuitSetup