CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter #44: Dead link on github page and lowest power consumption that can be measured

bartgrefte, 2021-06-12:
There's a dead link on https://github.com/CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter , ctrl+f for "here for some more advanced configurations".

In addition to that, I'm curious about something, what's the lowest power consumption a setup using the 6-Channel ESP32 Energy Meter can measure? Don't see that in the specs, or I overlooked it.

CircuitSetup, 2021-06-15:
Thanks, that link is fixed. 

The lowest the meter can register depends on the accuracy of the CT and if it is calibrated or not. Generally, with the YHDC CTs, the accuracy is going to be within 2% (1% CT accuracy, 1% meter burden resistor accuracy) for amperage. Then factoring in the accuracy of the voltage measurement (the newest meters have a 0.5% resistors for the voltage dividers), and about 1% for the Jameco Transformers.

So, it can go down pretty low to about 1-2 watts.

bartgrefte, 2021-06-19:
As low as 1-2 watts is perfect, thanks for the info :)

closed 2021-07-06 by CircuitSetup