CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter #51: Odd Power readings on one CT

kgalilio, 2021-08-15:
I've got one Main Board with two Addon boards stacked on top. All are working fine, with the exception of Addon 1 CT8 (middle left).
The CT Clamp is clamped to monitor an Air Conditioner. With 18 Amps and 230V, I am expecting to see values around 4,000 Watts; however, I am getting very low values (fluctuating between 2W and 80W).
If the Voltage, Frequency and Amps are correct and using the same power and calibration factors for all sensors, what causes this one to misread?
I've changed 3 CT clamps already, and they all produce similar results.

[image]
[image]

CircuitSetup, 2021-08-15:
How is the CT reading the 2 phases of the 2 pole circuit? Something is likely backwards since a low power reading would indicate a low power factor.

kgalilio, 2021-08-15:
Do you mean the clamp could be installed backwards? If that's the case, shouldn\t it cause the values to read negative?

CircuitSetup, 2021-08-15:
Is the one CT around both wires? If so, they should be going in opposite directions through the CT.

kgalilio, 2021-08-15:
I use single wire only.
It's a 230V system, Common and Live. The clamp is around the Live wire.

CircuitSetup, 2021-08-15:
Ohh, I see. 

Okay, try turning the CT around on the wire.

kgalilio, 2021-08-15:
Tried that. Similar fluctuating small values but negative!

CircuitSetup, 2021-08-15:
If you switch the CT on the wire to a different channel on the meter then what happens?

miguelangel-nubla, 2022-03-22:
I am also hitting this, even the same CT8. 230V Live/common. 233.1 V reading
```
energy-meter-main CT8 Amps	0.48 A	
energy-meter-main CT8 Watts	9.14 W	
```

CircuitSetup, 2022-03-22:
> I am also hitting this, even the same CT8. 230V Live/common. 233.1 V reading
> 
> ```
> energy-meter-main CT8 Amps	0.48 A	
> energy-meter-main CT8 Watts	9.14 W	
> ```

This is possible if the power factor reading is low. Are you getting a reading for that?

closed 2022-06-01 by CircuitSetup