CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter #62: Communication with ATM90E32 failed

XtremeOwnageDotCom, 2021-09-13:
So- everything was working nicely as expected.

I picked up an addon board, and incidentally placed the ESP32 one row down too far into the pin, which the documentation specifically says do not place the ESP32 into. The ESP responded by bricking itself.

Well, I replaced the ESP32, with another ESP32-WROOM-32D, and placed the board into the proper socket this time. Well, after troubleshooting for a bit, I removed the addon board, and reverted to just placing it into the main board, in its original configuration.

After tinkering a bit more, and removing all non-essential configuration, I have reached the limit of my troubleshooting capabilities, in that, the logs only tell me "[13:20:03][E][atm90e32:132]: Communication with ATM90E32 failed!"

When I accidentally incorrectly installed the ESP32 when adding the addon-board, did this damage the underlying circuit? If not- what other debugging steps should be performed?

This is the ESP32 I am attempting to use currently. The pinout seems to match the original one. 

https://www.amazon.com/gp/product/B0718T232Z

It is labeled as ESP32-WROOM-32D

## Config
```yaml
substitutions:
  devicename: "energy_meter"
  friendly_name: "Energy Meter"
  ip_address: 10.100.3.112
  gateway: 10.100.3.1
# Interval of how often the power is updated
  update_time: 15s

# Current Transformers:
#  20A/25mA SCT-006: 11143
#  30A/1V SCT-013-030: 8650
#  50A/1V SCT-013-050: 15420
#  80A/26.6mA SCT-010: 41660
#  100A/50ma SCT-013-000: 27518
#  120A/40mA: SCT-016: 41787
#  200A/100mA SCT-024: 27518
# Jameco 9VAC Transformer: 42620
  voltage_cal: '7305'
  mains_current_cal: '41787'
  normal_current_cal: '11143'

  <<: !include secrets.yaml

<<: !include config/common.yaml

logger:
  level: VERY_VERBOSE
  
esphome:
  name: ${devicename}
  platform: ESP32
  board: nodemcu-32s

spi:
  clk_pin: 18
  miso_pin: 19
  mosi_pin: 23

sensor:
  - platform: wifi_signal
    name: ${friendly_name} WiFi
    update_interval: 60s
#IC1 - Main Board   
  - platform: atm90e32
    cs_pin: 5
    #CT1
#    phase_a:
#      voltage:
#        name: ${friendly_name} Volts
#        id: ic1Volts
#        accuracy_decimals: 1
#      current:
[... 163 lines cut ...]
[13:14:34][C][ota:030]:   Address: 10.100.3.112:3232
[13:14:34][C][ota:032]:   Using Password.
[13:14:34][C][api:095]: API Server:
[13:14:34][C][api:096]:   Address: 10.100.3.112:6053
[13:14:34][C][wifi_signal.sensor:009]: WiFi Signal 'Energy Meter WiFi'
[13:14:34][C][wifi_signal.sensor:009]:   Device Class: 'signal_strength'
[13:14:34][C][wifi_signal.sensor:009]:   State Class: 'measurement'
[13:14:34][C][wifi_signal.sensor:009]:   Unit of Measurement: 'dBm'
[13:14:34][C][wifi_signal.sensor:009]:   Accuracy Decimals: 0
[13:14:34][V][wifi_signal.sensor:009]:   Unique ID: '94b97eda9788-wifisignal'
[13:14:38][VV][api.service:327]: on_ping_request: PingRequest {}
[13:14:38][VV][api.service:043]: send_ping_response: PingResponse {}
[13:14:43][VV][api.service:327]: on_ping_request: PingRequest {}
[13:14:43][VV][api.service:043]: send_ping_response: PingResponse {}
[13:14:48][VV][api.service:327]: on_ping_request: PingRequest {}
[13:14:48][VV][api.service:043]: send_ping_response: PingResponse {}
[13:14:53][VV][api.service:327]: on_ping_request: PingRequest {}
[13:14:53][VV][api.service:043]: send_ping_response: PingResponse {}
[13:14:58][VV][api.service:327]: on_ping_request: PingRequest {}
[13:14:58][VV][api.service:043]: send_ping_response: PingResponse {}
[13:15:03][VV][api.service:327]: on_ping_request: PingRequest {}
[13:15:03][VV][api.service:043]: send_ping_response: PingResponse {}
[13:15:08][VV][api.service:327]: on_ping_request: PingRequest {}
[13:15:08][VV][api.service:043]: send_ping_response: PingResponse {}
```

CircuitSetup, 2021-09-20:
I have to check, but I believe you have to have at least one phase for ESPHome to connect to the meter. 

Also, have you tried just connecting the add-on board without the main board, and with ESP32 plugged in - powered by a USB cable? (you have to change the cs pins in the config to what they are set to on the add-on board) 

You can try the same as above with the main board without the AC transformer plugged in. It's possible the power supply was burned out. If this is the case you should still be able to power everything via the ESP32 USB connection.

XtremeOwnageDotCom, 2021-09-20:
I have tried it, without the add-on board, while the main-board was powered via 9v. The ESP does start, and I do see green lights on the main board. However- it failed to communicate with the ATM90E32.

When powering the ESP via USB, it still exhibits the same issue.

If- you can point me at applicable test-points, I can probe and see if they are receiving the proper voltage. I however, do not have a oscilloscope capable of viewing the I2C / SPI bus.

XtremeOwnageDotCom, 2021-09-20:
[image]

CircuitSetup, 2021-09-20:
Okay, how about just the add-on board with the cs pins changed in the config?

XtremeOwnageDotCom, 2021-09-20:
With addon-board only- configured to use CS 0, and 16

```
[16:59:38][I][app:102]: ESPHome version 2021.9.1 compiled on Sep 20 2021, 16:58:35
[16:59:38][C][wifi:501]: WiFi:
[16:59:38][C][wifi:361]:   SSID: [redacted]
[16:59:38][C][wifi:362]:   IP Address: 10.100.3.112
[16:59:38][C][wifi:364]:   BSSID: [redacted]
[16:59:38][C][wifi:365]:   Hostname: 'energy_meter'
[16:59:38][C][wifi:369]:   Signal strength: -68 dB ▂▄▆█
[16:59:38][C][wifi:373]:   Channel: 5
[16:59:38][C][wifi:374]:   Subnet: 255.255.255.0
[16:59:38][C][wifi:375]:   Gateway: 10.100.3.1
[16:59:38][C][wifi:376]:   DNS1: 0.0.0.0
[16:59:38][C][wifi:377]:   DNS2: 0.0.0.0
[16:59:38][C][spi:081]: SPI bus:
[16:59:38][C][spi:082]:   CLK Pin: GPIO18 (Mode: OUTPUT)
[16:59:38][C][spi:083]:   MISO Pin: GPIO19 (Mode: INPUT)
[16:59:38][C][spi:084]:   MOSI Pin: GPIO23 (Mode: OUTPUT)
[16:59:38][C][spi:085]:   Using HW SPI: YES
[16:59:38][C][logger:193]: Logger:
[16:59:38][C][logger:194]:   Level: VERY_VERBOSE
[16:59:38][C][logger:195]:   Log Baud Rate: 115200
[16:59:38][C][logger:196]:   Hardware UART: UART0
[16:59:38][C][:129]: ATM90E32:
[16:59:38][C][atm90e32:130]:   CS Pin: GPIO0 (Mode: OUTPUT)
[16:59:38][E][atm90e32:132]: Communication with ATM90E32 failed!
[16:59:38][C][atm90e32:134]:   Update Interval: 15.0s
[16:59:38][C][atm90e32:156]:   Frequency 'Energy Meter Freq'
[16:59:38][C][atm90e32:156]:     State Class: 'measurement'
[16:59:38][C][atm90e32:156]:     Unit of Measurement: 'Hz'
[16:59:38][C][atm90e32:156]:     Accuracy Decimals: 1
[16:59:38][C][atm90e32:156]:     Icon: 'mdi:current-ac'
[16:59:38][C][:129]: ATM90E32:
[16:59:38][C][atm90e32:130]:   CS Pin: GPIO16 (Mode: OUTPUT)
[16:59:38][C][atm90e32:134]:   Update Interval: 15.0s
[16:59:38][C][restart:021]: Restart Switch 'Energy Meter Restart'
[16:59:38][C][restart:021]:   Icon: 'mdi:restart'
[16:59:38][C][homeassistant.time:010]: Home Assistant Time:
[16:59:38][C][homeassistant.time:011]:   Timezone: 'CST6CDT5,M3.2.0/2,M11.1.0/2'
[16:59:38][C][ota:029]: Over-The-Air Updates:
[16:59:38][C][ota:030]:   Address: 10.100.3.112:3232
[16:59:38][C][ota:032]:   Using Password.
[16:59:38][C][api:135]: API Server:
[16:59:38][C][api:136]:   Address: 10.100.3.112:6053
[16:59:38][C][wifi_signal.sensor:009]: WiFi Signal 'Energy Meter WiFi'
[16:59:38][C][wifi_signal.sensor:009]:   Device Class: 'signal_strength'
[16:59:39][C][wifi_signal.sensor:009]:   State Class: 'measurement'
[16:59:39][C][wifi_signal.sensor:009]:   Unit of Measurement: 'dBm'
[16:59:39][C][wifi_signal.sensor:009]:   Accuracy Decimals: 0
[16:59:39][V][wifi_signal.sensor:009]:   Unique ID: '94b97eda9788-wifisignal'
[16:59:43][VV][scheduler:151]: Running interval 'update' with interval=15000 last_execution=4294964766 (now=12471)
[16:59:43][VV][spi:095]: Enabling SPI Chip on pin 16...
[16:59:43][VV][spi:016]: Disabling SPI Chip on pin 16...
[16:59:43][VV][atm90e32:176]: read16_ 0x0000 output 0x0001
[16:59:53][VV][api.service:327]: on_ping_request: PingRequest {}
[16:59:53][VV][api.service:043]: send_ping_response: PingResponse {}
```

Actually- oddly enough, it appears it is detecting the one on 16. Just- not the one on 0. Interesting....

XtremeOwnageDotCom, 2021-09-21:
So- I had an interesting theory that perhaps there may be something weird with the pin configurations. So- I made this config to test my theory: 

```yaml
sensor:
#IC1 - Main Board   
  - platform: atm90e32
    cs_pin: 0
    line_frequency: 60Hz
  - platform: atm90e32
    cs_pin: 1
    line_frequency: 60Hz
  - platform: atm90e32
    cs_pin: 2
    line_frequency: 60Hz
  - platform: atm90e32
    cs_pin: 3
    line_frequency: 60Hz
  - platform: atm90e32
    cs_pin: 4
    line_frequency: 60Hz
  - platform: atm90e32
    cs_pin: 5
    line_frequency: 60Hz
  - platform: atm90e32
    cs_pin: 9
    line_frequency: 60Hz
  - platform: atm90e32
    cs_pin: 10
    line_frequency: 60Hz
  - platform: atm90e32
    cs_pin: 12
    line_frequency: 60Hz
  - platform: atm90e32
    cs_pin: 13
    line_frequency: 60Hz
  - platform: atm90e32
    cs_pin: 14
    line_frequency: 60Hz
  - platform: atm90e32
    cs_pin: 15
    line_frequency: 60Hz
  - platform: atm90e32
    cs_pin: 16
    line_frequency: 60Hz
  - platform: atm90e32
    cs_pin: 17
    line_frequency: 60Hz
  - platform: atm90e32
    cs_pin: 18
    line_frequency: 60Hz
  - platform: atm90e32
    cs_pin: 19
    line_frequency: 60Hz
  - platform: atm90e32
    cs_pin: 21
    line_frequency: 60Hz
  - platform: atm90e32
    cs_pin: 22
    line_frequency: 60Hz
  - platform: atm90e32
    cs_pin: 23
    line_frequency: 60Hz
  - platform: atm90e32
    cs_pin: 25
    line_frequency: 60Hz
  - platform: atm90e32
    cs_pin: 26
    line_frequency: 60Hz
  - platform: atm90e32
    cs_pin: 27
[... 162 lines cut ...]
[19:08:04][E][atm90e32:132]: Communication with ATM90E32 failed!
[19:08:04][C][atm90e32:134]:   Update Interval: 60.0s
[19:08:04][C][:129]: ATM90E32:
[19:08:04][C][atm90e32:130]:   CS Pin: GPIO22 (Mode: OUTPUT)
[19:08:04][E][atm90e32:132]: Communication with ATM90E32 failed!
[19:08:04][C][atm90e32:134]:   Update Interval: 60.0s
[19:08:04][C][:129]: ATM90E32:
[19:08:04][C][atm90e32:130]:   CS Pin: GPIO23 (Mode: OUTPUT)
[19:08:04][E][atm90e32:132]: Communication with ATM90E32 failed!
[19:08:04][C][atm90e32:134]:   Update Interval: 60.0s
[19:08:05][C][:129]: ATM90E32:
[19:08:05][C][atm90e32:130]:   CS Pin: GPIO25 (Mode: OUTPUT)
[19:08:05][E][atm90e32:132]: Communication with ATM90E32 failed!
[19:08:05][C][atm90e32:134]:   Update Interval: 60.0s
[19:08:05][C][:129]: ATM90E32:
[19:08:05][C][atm90e32:130]:   CS Pin: GPIO26 (Mode: OUTPUT)
[19:08:05][E][atm90e32:132]: Communication with ATM90E32 failed!
[19:08:05][C][atm90e32:134]:   Update Interval: 60.0s
[19:08:05][C][:129]: ATM90E32:
[19:08:05][C][atm90e32:130]:   CS Pin: GPIO27 (Mode: OUTPUT)
[19:08:05][E][atm90e32:132]: Communication with ATM90E32 failed!
[19:08:05][C][atm90e32:134]:   Update Interval: 60.0s
```

Testing with main board only, produces none of the above sensors working.

CircuitSetup, 2021-09-21:
It sounds like the meters are damaged unfortunately. I would send you a replacement, but I do not have any in stock at the moment.

XtremeOwnageDotCom, 2021-09-22:
I shot a PM via your website's email function.

BrandoCommando, 2021-12-22:
I'm actually having the same issues with the "latest" board. Received the 6c main board Dec 13th, realized the 9VDC transformers I have wouldn't work, ordered the 9VAC Jaleco one, realized the 15pin ESP32s I have wouldn't work, ordered a 19pin one, and then got it all hooked up to a known load.
It was reporting "Sending state nan A with 2 decimals of accuracy" for all CTs. I thought it may have been the burden resistor thing, but noticed this issue on Github. So I enabled VERY_VERBOSE debugging, and noticed "Component atm90e32.sensor was marked as failed." for both sides. I tried the idea of telling the ESP there was an atm90e32 on every pin, and they all returned the same failure.
Other than trying a DC transformer and a 15pin ESP32 on this board, I don't think I did anything that may have damaged any of the meter's pins, so maybe my ESP32 (ESP32_DevKitC_V4) is bad? Should I try replacing that?

eaconner, 2021-12-22:
I’m having a similar issue, received my v1.4 board on December 20, 2021 and one atm90e32 is working but the other is unresponsive. Very frustrating after waiting 3 months for this circuit.

XtremeOwnageDotCom, 2021-12-22:
All I can say, is good luck to the both of you. 

I never received a response from my PM, nor an update here. I ended up saying screw it, and ordered an Iotawatt. Arrived a week later, and has been chugging along ever since working fine.

CircuitSetup, 2021-12-22:
@BrandoCommando, @eaconner, @XtremeOwnageDotCom  New meters coming your way to test.

CircuitSetup, 2021-12-22:
> All I can say, is good luck to the both of you.
> 
> I never received a response from my PM, nor an update here (after 3 months). I ended up saying screw it, and ordered an Iotawatt. Arrived a week later, and has been chugging along ever since working fine.
> 
> My now defunct circuitsetup device makes an interesting paperweight on my desk.

I'm sorry I did not get back to you sooner. I literally just got in more meters after a long wait fixing supply issues.

> To the manufacturer / admin / etc, I recommend you add a tad of circuitry to prevent the ESP from being slotted in correctly.... ya know, since the slots are exactly one slot larger than a standard ESP.... and being one slot off apparently cooks the chips.

Yes, putting AC voltage to an ESP32 does not do it good. Normally there is a pin in the bottom connector to prevent this from happening.

missmah, 2022-01-19:
> I have to check, but I believe you have to have at least one phase for ESPHome to connect to the meter.

Do you mean defined in the YAML? Or physical CTs actually connected? I see a bunch of "sending new state" and "received new state", in the log via ESPHome inside HomeAssistant, but I don't see any devices in home assistant, and I'm wondering if this is because I don't have any CTs physically connected yet? (I'm running via USB power right now)

alex-murashko, 2022-06-25:
I experienced the same issue. 
I have main and addon board. ESP failed to communicate with all four IC's. 

In my case, I had the ESP32 not fully inserted into socket.  

Once I have it fully inserted the issues are gone.

fjaquish, 2023-01-08:
I know this is an old thread, but I have a very similar problem. I upgraded to a second addon board and now all 6 ATM90E32 chips fail to communicate. I did initially have the CS in the wrong position on the the second addon board on first powerup.  This seemed to have bricked my NodeMCU. I replaced the nodemcu but now none of the PM chips communicate. 

Tx

Here is the log, all ATM90E32s fail to communicate. 

INFO Reading configuration /config/esphome/energy-monitor.yaml...
WARNING GPIO4 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Detected timezone 'America/New_York'
INFO Starting log output from 192.168.1.35 using esphome API
INFO Successfully connected to 192.168.1.35
[14:22:36][I][app:102]: ESPHome version 2022.12.3 compiled on Jan  7 2023, 12:08:09
[14:22:36][C][wifi:504]: WiFi:
[14:22:36][C][wifi:362]:   Local MAC: C0:49:EF:E5:FA:54
[14:22:36][C][wifi:363]:   SSID: '8vanornum'[redacted]
[14:22:36][C][wifi:364]:   IP Address: 192.168.1.35
[14:22:36][C][wifi:366]:   BSSID: F8:BB:BF:A8:D4:E3[redacted]
[14:22:36][C][wifi:367]:   Hostname: 'energy-monitor'
[14:22:36][C][wifi:369]:   Signal strength: -57 dB ▂▄▆█
[14:22:36][C][wifi:373]:   Channel: 1
[14:22:36][C][wifi:374]:   Subnet: 255.255.255.0
[14:22:36][C][wifi:375]:   Gateway: 192.168.1.254
[14:22:36][C][wifi:376]:   DNS1: 165.19.68.30
[14:22:36][C][wifi:377]:   DNS2: 0.0.0.0
[14:22:37][C][logger:293]: Logger:
[14:22:37][C][logger:294]:   Level: DEBUG
[14:22:37][C][logger:295]:   Log Baud Rate: 115200
[14:22:37][C][logger:296]:   Hardware UART: UART0
[14:22:37][C][spi:101]: SPI bus:
[14:22:37][C][spi:102]:   CLK Pin: GPIO18
[14:22:37][C][spi:103]:   MISO Pin: GPIO19
[14:22:37][C][spi:104]:   MOSI Pin: GPIO23
[14:22:37][C][spi:106]:   Using HW SPI: YES
[14:22:37][C][template.sensor:023]: Template Sensor '6CH Total House Amps'
[14:22:37][C][template.sensor:023]:   State Class: ''
[14:22:37][C][template.sensor:023]:   Unit of Measurement: 'A'
[14:22:37][C][template.sensor:023]:   Accuracy Decimals: 2
[14:22:37][C][template.sensor:023]:   Icon: 'mdi:flash'
[14:22:37][C][template.sensor:024]:   Update Interval: 10.0s
[14:22:37][C][template.sensor:023]: Template Sensor '6CH Total House Watts'
[14:22:37][C][template.sensor:023]:   State Class: ''
[14:22:37][C][template.sensor:023]:   Unit of Measurement: 'W'
[14:22:37][C][template.sensor:023]:   Accuracy Decimals: 1
[14:22:37][C][template.sensor:023]:   Icon: 'mdi:flash-circle'
[14:22:37][C][template.sensor:024]:   Update Interval: 10.0s
[14:22:37][C][template.sensor:023]: Template Sensor '6CH Total Solar Amps'
[14:22:37][C][template.sensor:023]:   State Class: ''
[14:22:37][C][template.sensor:023]:   Unit of Measurement: 'A'
[14:22:37][C][template.sensor:023]:   Accuracy Decimals: 2
[14:22:37][C][template.sensor:023]:   Icon: 'mdi:flash'
[14:22:37][C][template.sensor:024]:   Update Interval: 10.0s
[14:22:37][C][template.sensor:023]: Template Sensor '6CH Total Solar Watts'
[14:22:37][C][template.sensor:023]:   State Class: ''
[14:22:37][C][template.sensor:023]:   Unit of Measurement: 'W'
[14:22:37][C][template.sensor:023]:   Accuracy Decimals: 1
[14:22:37][C][template.sensor:023]:   Icon: 'mdi:flash-circle'
[14:22:37][C][template.sensor:024]:   Update Interval: 10.0s
[14:22:37][D][sensor:127]: '6CH Total HotWater Amps': Sending state nan A with 2 decimals of accuracy
[14:22:37][C][template.sensor:023]: Template Sensor '6CH Total Garage Amps'
[14:22:37][C][template.sensor:023]:   State Class: ''
[14:22:37][C][template.sensor:023]:   Unit of Measurement: 'A'
[14:22:37][C][template.sensor:023]:   Accuracy Decimals: 2
[14:22:37][C][template.sensor:023]:   Icon: 'mdi:flash'
[14:22:37][C][template.sensor:024]:   Update Interval: 10.0s
[14:22:37][C][template.sensor:023]: Template Sensor '6CH Total Garage Watts'
[14:22:37][C][template.sensor:023]:   State Class: ''
[14:22:37][C][template.sensor:023]:   Unit of Measurement: 'W'
[14:22:37][C][template.sensor:023]:   Accuracy Decimals: 1
[... 74 lines cut ...]
[14:22:37][C][atm90e32:151]:     Device Class: 'power'
[14:22:37][C][atm90e32:151]:     State Class: 'measurement'
[14:22:37][C][atm90e32:151]:     Unit of Measurement: 'W'
[14:22:37][C][atm90e32:151]:     Accuracy Decimals: 2
[14:22:37][C][atm90e32:156]:   Frequency '6CH Freq'
[14:22:37][C][atm90e32:156]:     State Class: 'measurement'
[14:22:37][C][atm90e32:156]:     Unit of Measurement: 'Hz'
[14:22:37][C][atm90e32:156]:     Accuracy Decimals: 1
[14:22:38][C][atm90e32:156]:     Icon: 'mdi:current-ac'
[14:22:38][D][sensor:127]: '6CH Total Hot Water Watts': Sending state nan W with 1 decimals of accuracy
[14:22:38][C][:129]: ATM90E32:
[14:22:38][C][atm90e32:130]:   CS Pin: GPIO4
[14:22:38][E][atm90e32:132]: Communication with ATM90E32 failed!
[14:22:38][C][atm90e32:134]:   Update Interval: 10.0s
[14:22:38][C][atm90e32:136]:   Current A '6CH Garage L2 Amps'
[14:22:38][C][atm90e32:136]:     Device Class: 'current'
[14:22:38][C][atm90e32:136]:     State Class: 'measurement'
[14:22:38][C][atm90e32:136]:     Unit of Measurement: 'A'
[14:22:38][C][atm90e32:136]:     Accuracy Decimals: 2
[14:22:38][C][atm90e32:137]:   Power A '6CH Garage L2 Watts'
[14:22:38][C][atm90e32:137]:     Device Class: 'power'
[14:22:38][C][atm90e32:137]:     State Class: 'measurement'
[14:22:38][C][atm90e32:137]:     Unit of Measurement: 'W'
[14:22:38][C][atm90e32:137]:     Accuracy Decimals: 2
[14:22:38][C][atm90e32:143]:   Current B '6CH Solar L2 Amps'

XtremeOwnageDotCom, 2023-01-08:
> <CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter/issues/62/1374817996
> @github.com>

Sadly, my resolution to this problem was to chunk the hardware into the
trash.

Ended up picking up an Iotawatt. It's been ticking every since.

Tbh, it's a design issue which should probably be addressed... as it's
EXTREMELY easy to do this.

fjaquish, 2023-01-08:
Sorry to hear that. As I am sure you did, I have a fair amount of time and money in this setup, it worked great for a year. 

I did some further testing this morning, I plugged the NodeMCU into each board separately powered by the USB port. Only one of the three boards communicates with the PM chip. My main board and one addon don't communicate at all. I guess they are cooked, very frustrating.

XtremeOwnageDotCom, 2023-01-08:
@fjaquish  yea, its a shame. I had high hopes for this one. Despite this [Comment](https://github.com/CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter/issues/62#issuecomment-999885293), nothing ever did occur on this issue.

If you have a heat gun, and a few components, might be worthwhile trying to replace the atm chip. Otherwise, 

Might be time to check out [Other Options](https://static.xtremeownage.com/blog/2022/2023-home-assistant-energy-monitoring/)

CircuitSetup, 2023-01-08:
> Sorry to hear that. As I am sure you did, I have a fair amount of time and money in this setup, it worked great for a year.
>
> I did some further testing this morning, I plugged the NodeMCU into each [...]

I'm sorry you're having issues with the meters. Have you tried using another esp32? It's more likely that something is wrong with that then a bunch of metering chips not responding. Having duplicate CS pins will not hurt anything. Be sure that the esp32 has 38 pins as well.

CircuitSetup, 2023-01-08:
> 

I'm sorry it didn't work out. I agree there should be something more to protect circuitry when inserting the esp32 into the wrong slot. The meters do come with a pin in the bottom slot to prevent this from happening.

fjaquish, 2023-01-08:
Hi, As I stated earlier, on initial powerup my first NodeMCU bricked.  It is bothersome because it always worked before. I was flashing with an older version of ESPHome, I believe that is why when I added the new sensors I was getting stack overflows. I upgraded the ESPHome and flashed the new yaml via usb. 

This morning I tested each board.  One of the addon board communicates with the ATM90E32 chips, the other two just return NA and the log shows a fail to connect. If I had another NodeMCU, I would pop it in and see, but I don't. I also have repeatedly plugged the NodeMCU into the board that works and it seems to connect without trouble.

fjaquish, 2023-01-08:
Just to be clear, at no time did I plug the NodeMCU into the wrong slot, my error was not having CS configured correctly on my first powerup. I did not expect that this would cause anything to fry?

fjaquish, 2023-01-09:
> <CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter/issues/62/1374859958

Hi John,

I am looking for a recommendation on what to do next?  Suggestions? Tx, FJJ

fjaquish, 2023-01-19:
Just to close out this thread. 

Circuit Setup replaced my Mainboard and one Add-on board that was not communicating with the ATM chips. 

I started with a fresh NodeMCU and installed the new hardware and everything appears to be working as it should. 

I wish I knew why my first setup did not work, I hate open threads, but at least it is all working now.

Kudos to Circuit Setup for standing behind their product.

FJJ

XtremeOwnageDotCom, 2023-01-19:
> Just to close out this thread.

Eh, the thread was originally opened due to mine having the same issue as yours. 

To fully clarify, mine is completely dead STILL. Despite above comments saying replacement hardware was shipped- that never occured. If it did- It never reached my doorstep.

fjaquish, 2023-01-19:
XtremeOwnageD, I can only relate my experience,  I am sorry that you did not get yours resolved. .

closed 2025-02-24 by CircuitSetup