CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter #63: Sensors reading with a very small negative value

profucius, 2021-09-16:
Hello, I am using your 6C main board to read the energy through four different lines in my power box. AC for upstairs, AC for downstairs, washing machine, and dryer. I've got everything configured and working well, except for one issue I cant seem to figure out.

When the appliances are currently "off", the 6C unit is sending a very small negative value in the watts sensor. I would ignore this except it gets amplified up to the kWh reading and therefore the energy cost reading in the Energy dashboard within Home Assistant.

[image]

I've triple checked my yaml within ESPHome and everything seems to be correct, it's mostly the same copied from your example file. What can I do to make this value absolute?

By the way, when the appliances turn on, I get a positive watt reading, so the issue is not that the clamps are on backwards (I've double checked that).

CircuitSetup, 2021-09-19:
See this post in the discussion section: https://github.com/CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter/discussions/59

profucius, 2021-09-20:
Thanks for this link. I ended up resolving the issue using the advice. For anyone else who wants to know how I did it:

Editing the ESPhome yaml for the energy-meter, towards the bottom, below the last `phase_c` section" (specifically, below the last `update_interval` line), I added a new section. Here is a copy of what is in my yaml now, but of course this would need to be edited for your own purposes. The idea being that you rename the `name `and `id` fields to match your own sensor, and then rename the id within the `if (id(xxxxx).state` within `lamba` so that it references your original sensor id.

`
#===================================================================#
### Convert watts to positive numbers (since the meter sometimes reads negatives)

# Dryer Watts Positive
  - platform: template
    name: ${disp_name} Dryer - Watts Positive
    id: DryerWattsPositive
    lambda: |-
      if (id(ct1Watts).state < 0) {
        return 0;
      } else {
        return id(ct1Watts).state ;
      }
    accuracy_decimals: 1
    unit_of_measurement: W
    icon: "mdi:flash-circle"
    update_interval: ${update_time}

# Washer Watts Positive
  - platform: template
    name: ${disp_name} Washer - Watts Positive
    id: WasherWattsPositive
    lambda: |-
      if (id(ct2Watts).state < 0) {
        return 0;
      } else {
        return id(ct2Watts).state ;
      }
    accuracy_decimals: 1
    unit_of_measurement: W
    icon: "mdi:flash-circle"
    update_interval: ${update_time}

# BLANK3 Watts Positive
  - platform: template
    name: ${disp_name} BLANK3 - Watts Positive
    id: ct3WattsPositive
    lambda: |-
      if (id(ct3Watts).state < 0) {
        return 0;
      } else {
        return id(ct3Watts).state ;
      }
    accuracy_decimals: 1
    unit_of_measurement: W
    icon: "mdi:flash-circle"
    update_interval: ${update_time}

# BLANK4 Watts Positive
  - platform: template
    name: ${disp_name} BLANK4 - Watts Positive
    id: ct4WattsPositive
    lambda: |-
      if (id(ct4Watts).state < 0) {
        return 0;
      } else {
        return id(ct4Watts).state ;
      }
    accuracy_decimals: 1
    unit_of_measurement: W
    icon: "mdi:flash-circle"
    update_interval: ${update_time}

# AC Down Watts Positive
  - platform: template
    name: ${disp_name} AC Down - Watts Positive
    id: ACDownWattsPositive
    lambda: |-
      if (id(ct5Watts).state < 0) {
        return 0;
      } else {
        return id(ct5Watts).state ;
      }
    accuracy_decimals: 1
    unit_of_measurement: W
    icon: "mdi:flash-circle"
    update_interval: ${update_time}

# AC Up Watts Positive
  - platform: template
    name: ${disp_name} AC Up - Watts Positive
    id: ACUpWattsPositive
    lambda: |-
      if (id(ct6Watts).state < 0) {
        return 0;
      } else {
        return id(ct6Watts).state ;
      }
    accuracy_decimals: 1
    unit_of_measurement: W
    icon: "mdi:flash-circle"
    update_interval: ${update_time}
`

closed 2021-09-20 by profucius