CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter #75: Components/Hardware Details

kimza90, 2022-01-15:
hi there,

I need your help!
Which hardware do you have select?
Do you have details or maybe a datasheet/shopping link?
The BOM list is missing up to several components.

best regards
kim

missmah, 2022-01-16:
Indeed, is there slack proper BOM for this project? The Eagle files seem to be missing a lot of important information. Thanks!

CircuitSetup, 2022-01-16:
There is a BOM located here: https://octopart.com/bom-tool/EwHlAzfY

There are some parts that are out of date, however and can be substituted. The burden resistors used lately are 0.1% tolerance.

kimza90, 2022-01-16:
Thank you very much.
do you have BOM for the adon board ??

CircuitSetup, 2022-02-17:
> Thank you very much. do you have BOM for the adon board ??

Sorry I overlooked your reply. The add-on board is the same, but has 2 long 20pin headers, 1 short header, 2 12pin 2 row 2mm connectors w/ 2mm shunts, and no power supply parts (C31-C35, R22, R23, L1, D1, BR1, IC3, DC connector)

kimza90, 2022-02-18:
Thank you for your feedback. can I only use add on board if I do not need volt measurement? add power via usb power to ESP32

CircuitSetup, 2022-02-18:
The add-on board can be powered by the ESP32 usb port, yes. However, the meter's power calculations will not be correct without a voltage reference. This is okay if you only need the current measurement, or you want to calculate power in code.

kimza90, 2022-02-18:
I see that ATM90E32AS-AU-R (IC1 - IC2) has a very long delivery time have you found a replacement that works?

CircuitSetup, 2022-02-18:
> I see that ATM90E32AS-AU-R (IC1 - IC2) has a very long delivery time have you found a replacement that works?
I've run into problems recently getting them. Unfortunately the ATM90E32AS and ATM90E36 are the only ones that can be used. I believe the 36 needs one of the unconnected pins grounded though.

closed 2022-06-01 by CircuitSetup