CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter #79: Reboot Loop with ESP32 emon

github768484, 2022-01-23:
Can you tell me why I get this continuing loop? 

ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4
Starting AP
WiFi interface ready
WiFi client started
E (6693) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (6693) task_wdt:  - loopTask (CPU 1)
E (6693) task_wdt: Tasks currently running:
E (6693) task_wdt: CPU 0: IDLE0
E (6693) task_wdt: CPU 1: IDLE1
E (6693) task_wdt: Aborting.
abort() was called at PC 0x4016fe84 on core 0

ELF file SHA256: 0000000000000000

Backtrace: 0x4008880c:0x3ffbf8f0 0x40088a89:0x3ffbf910 0x4016fe84:0x3ffbf930 0x40086f81:0x3ffbf950 0x401867af:0x3ffbca10 0x401717ff:0x3ffbca30 0x4008b249:0x3ffbca50 0x40089a9a:0x3ffbca70

Rebooting...

CircuitSetup, 2022-01-23:
It looks like the watchdog timer is getting triggered. Try disabling it.

github768484, 2022-01-23:
That resolved the issue. Thank you. What is the downside of disabling the watchdog?

closed 2025-02-24 by CircuitSetup