CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter #80: Home Assistant Energy Page

iannecj, 2022-02-03:
This had been reporting on Energy Page completely however now while bars in individual devices are still reporting, the total grid energy graph is not.  When I go to Energy configuration in home assistant I see its complaining:

Last reset missing
The following entities have state class 'measurement' but 'last_reset' is missing:
sensor.pm_main_total_daily_kwh_farmbell

Is there new configuration in ESPhome YAML file?  I don't see this last reset aspect in the docs.

Meter has been flawless by the way thanks!

Thanks,

Chris

CircuitSetup, 2022-02-03:
[It looks like there was a fix to the default behavior of Total Daily Energy.](https://github.com/esphome/esphome/pull/2163) 

Try changing _state_class: measurement_ to _state_class: total_increasing_

CircuitSetup, 2022-02-04:
Yes, I'll definitely change it in the examples. Thanks for bringing this to light!

CircuitSetup, 2022-02-04:
oh, I forgot to ask. Was state_class added automatically (as stated in the forum thread), or is that something you added? I'm trying to figure out if it explicitly needs to be in the config.

iannecj, 2022-02-04:
I had for the main total energy sensor in YAML and I had added it at some point.  I believe from an online example.
device_class: "energy"
state_class: "measurement"

That did not work and it complained of no last reset parameter as we said.

Based on your suggestion I changed to:
    device_class: "energy"
    state_class: "total_increasing"

And it works but it may have also worked if I had just deleted  state_class: "measurement".  Didn't try that.

CircuitSetup, 2022-02-04:
got it! Thanks for your help!

closed 2022-02-04 by CircuitSetup