CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter #85: EmonESP unable to set port of EmonCMS server

mmallozzi, 2022-02-16:
When I try to enter a custom port in the traditional way for the EmonCMS server, such as `192.168.1.3:27015`, the code seems to treat the entire string as the hostname field, and uses hardcoded port 80 to try to connect to. So ultimately it tries to connect to `192.168.1.3:27015:80` and spits out errors accordingly.

This might just require pulling in a fix from upstream: https://github.com/openenergymonitor/EmonESP/pull/102

Also, what are the changes in the EmonESP fork? Is it possible to just use the mainstream EmonESP?

mmallozzi, 2022-02-17:
For anyone looking for a temporary workaround, I modified emoncms.cpp in the emoncms_publish method to pass my desired port (27015) as the optional third parameter to get_http in the "Plain old HTTP" branch. For me (puled down master a couple days ago) that was on line 63.

CircuitSetup, 2022-02-18:
Thanks for following up with this. I'll take a look at what needs to be changed to fix it. 

The version for the 6 channel meter adds calibration fields, and the additional current channels depending on how many add-on boards are used, if any. Unfortunately, the original EmonESP would not work properly with the 6 channel meter.

closed 2025-01-27 by CircuitSetup