CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter #90: gerber files

rikvandersteen, 2022-02-23:
hello  can annybody help me i tried to make the gerber files  of the 3 pcb but eagle showing corrupted files 
can annybody send me the 3 gerber files of the pcb   thank you

wwk-github, 2022-02-23:
The sch/brd files open fine.
Are you sure your local files were properly cloned from git?

This is what I can see:
[image]

wwk-github, 2022-02-23:
What are your files sha-sums and sizes?
Below are mine.
Can you post yours?

```
ls -allh *.sch       
-rw-r--r--  1 w  staff   1.7M Feb 18 11:45 6_channel_esp32_energy_meter.sch
-rw-r--r--  1 w  staff   1.1M Feb 18 11:45 6_channel_esp32_energy_meter_add-on.sch
-rw-r--r--  1 w  staff   122K Feb  6 20:00 6_channel_remote.sch

ls -allh *.brd
-rw-r--r--  1 w  staff   408K Feb 18 11:45 6_channel_esp32_energy_meter.brd
-rw-r--r--  1 w  staff   409K Feb 18 11:45 6_channel_esp32_energy_meter_add-on.brd
-rw-r--r--  1 w  staff    72K Feb  6 20:00 6_channel_remote.brd
```

```
sha256sum *.brd
35deda746f9961dd08c735d46e2f291d8f1bf4efde3d03f0b2587383172ce62d  6_channel_esp32_energy_meter.brd
69cdaad99e61140cdce5e5c0269944e3e960628cdd9e059a9ee8a10d0256957e  6_channel_esp32_energy_meter_add-on.brd
e86a60d9b7105e089e79f66f8ae5ff17c1a7a39bbab307fb9d7cc6e5155cf838  6_channel_remote.brd

sha256sum *.sch
2a6023082aea383277a7d66575dab7634dd4943533fb5e119c5752a5ffcf0232  6_channel_esp32_energy_meter.sch
193387096dce433dfaf05c84899caee2ffde81996a00a506214c58d903ddb7e1  6_channel_esp32_energy_meter_add-on.sch
d84772ee3e0509e569aec4f8186837aa21ff999bb6d52e04eafb36cc9cf8ceca  6_channel_remote.sch
```

wwk-github, 2022-02-24:
Maybe you should reinstall your EAGLE app...
I do not have much experience with EAGLE, so check those files before using them elsewhere.

[6_channel_esp32_energy_meter_2022-02-24.zip](https://github.com/CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter/files/8131385/6_channel_esp32_energy_meter_2022-02-24.zip)
[6_channel_esp32_energy_meter_add-on_2022-02-24.zip](https://github.com/CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter/files/8131386/6_channel_esp32_energy_meter_add-on_2022-02-24.zip)
[6_channel_remote_2022-02-24.zip](https://github.com/CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter/files/8131387/6_channel_remote_2022-02-24.zip)
.

rikvandersteen, 2022-02-24:
Thank you. I have checked your files.     And the give.the same error on jlcpcb.com. When I want to order them

CircuitSetup, 2022-02-24:
> Thank you. I have checked your files. And the give.the same error on jlcpcb.com. When I want to order them

When you are exporting via the CAM processor, is it set to Gerber RS-274X, and "Export as ZIP"? You can also load their cam file located here: https://raw.githubusercontent.com/JLCPCBofficial/jlcpcb-eagle/master/cam/jlcpcb_2_layer_v9.cam

closed 2022-10-21 by CircuitSetup