CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter #106: Max CT mA output - readme says 33mA peak max, but recommended models sold by your company go all the way up to 100mA.

alexruffell, 2022-07-13:
The readme states that the CTs must not exceed 33mA peak output, yet the models suggested go up to 100mA. What is the correct maximum mA output value? I have 30 and 40A 240V (USA) breakers I want to monitor and am trying to pick the correct CTs. I had found datasheets for a 50A 100mA CT and a 50A 50mA CT but am having trouble actually finding them for sale

https://github.com/CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter/blob/master/README.md

```
Current Transformers (any combination of the following, or any current transformer that does not exceed 720mV RMS, or 33mA output)
[SCT-006 20A/25mA Micro](https://circuitsetup.us/index.php/product/20a-25ma-micro-current-transformer-yhdc-sct-006-6mm/) (6mm opening - 3.5mm connectors)
[SCT-010 80A/26.6mA Mini](https://circuitsetup.us/index.php/product/80a-26-6ma-mini-current-transformer-yhdc-sct-010-10mm/) (10mm opening - 3.5mm connectors)
[SCT-013-000 100A/50mA](https://circuitsetup.us/index.php/product/100a-50ma-current-transformer-yhdc-sct-013/) (13mm opening - 3.5mm connectors)
[SCT-016 120A/40mA](https://circuitsetup.us/index.php/product/120a-40ma-current-transformer-yhdc-sct-016-with-3-5mm-jack-16mm-opening/) (16mm opening - 3.5mm connectors)
[Magnelab SCT-0750-100](https://amzn.to/2IF8xnY) (screw connectors - must sever burden resistor connection on the back of the board since they have a built in burden resistor).
[SCT-024 200A/100mA](https://www.poweruc.pl/products/split-core-current-transformer-sct024ts-rated-input-400a) (24mm opening - terminal output)
Others can also be used as long as they're rated for the amount of power that you are wanting to measure, and have a current output no more than 720mV RMS, or 33mA at peak output.
```

alexruffell, 2022-07-15:
Hi John,

The confusing bit is regarding the output of the CTs. The documentation says:

`Current Transformers (any combination of the following, or any current transformer that does not exceed 720mV RMS, or 33mA output)`

However several of the CTs you recommend have 40, 50 and 100mA outputs. I found some YHDC CTs that are 50A and have either a 50mA or 100mA output (not specified on amazon product page) so based on the CTs you recommend I figured they would be compatible. I should get a tad bit of added resolution since the CTs' measured current range is closer to what may flow on those breakers compared to using the 80/100A CTs.

I plan on measuring:

200A main breaker
2 x 100A CTs

40A Oven breaker
2 x 50A CTs

30A Dryer breaker
2 x 50A CTs

When the addon boards are back in stock, I may expand the system to also measure the 2 HVAC systems.

closed 2022-10-21 by CircuitSetup

Azylog, 2023-10-02:
Hello,
For the records : the max voltage ATM90E32 can measure is 720mV (with gain=1). Considering the 6-channel have 22Ω burden resistors, that means the maximum current you can flow through the burden is I=U/R=0.720/22 = 32.7mA

Using a 5Ω burden would allow to use SCT up to 144mA but with a lower resolution.

Did I miss something ?