CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter #107: Isn't reading Voltage at all, Frequency varies wildly

peely, 2022-07-30:
Hi there, my Expandable 6 Channel has just arrived and i'm struggling to get it setup correctly, i'm hoping you can be of help :) 

ESPHome flashed, reading coming back into Home Assistant via MQTT, all working. But the readings are wrong. I've used the https://github.com/CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter/blob/master/Software/ESPHome/6chan_energy_meter_main_board.yaml as my base, and modified with my local values (In the UK, so voltage and frequency are different). Please see this PR for my diff https://github.com/peely/Expandable-6-Channel-ESP32-Energy-Meter/pull/1/files

This is resulting in readings sent over MQTT that have a wildly varing Frequency, 
[image]

And the Voltage reading of near 0 always 
[image]

I'm well aware that this will be a PEBCAK issue, i'm just not sure what config value I can change to fix this :sweat_smile: 

I've got a [50A CT Clamp](https://www.amazon.co.uk/gp/product/B07MY3NNFR) on CT1, and I also have a [30A clamp](https://www.amazon.co.uk/gp/product/B07MY361ZW) on CT2, both on the same live test wire at the moment (I wanted to see the effect clamp size had on readings, my screenshots are only of the 50A clamp). I'm not using the power supply suggested, just another 9V power supply I already had, which my volt meter tells me is outputting 9.12V

Is there anymore information I can provide to help the troubleshooting? I was following [this guide](https://www.youtube.com/watch?v=n2XZzciz0s4) linked to by the [Home Assistant Energy page](https://www.home-assistant.io/docs/energy/electricity-grid/). Thank you in advanced.

CircuitSetup, 2022-08-03:
I was going to ask if the transformer was AC/AC, and not AC/DC :)  Common mistake - glad you figured it out!

peely, 2022-08-04:
Yes indeed I did, with help :) And now i've had time to re-test it with the new power transformer and it's working great! Thank you very much. I'll close this issue now.

closed 2022-08-04 by peely