CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter #111: Communication with ATM90E32 failed

max-rousseau, 2022-08-23:
Hello - I recently picked up a v1.4 rev1 board (the black one). As I was trying, unsuccessfully so, to calculate the right calibration values on a small test circuit I have (just one light bulb) the board eventually started to say that the communication failed with the sensors. I did swap them out as I was trying to figure the right calibration between all 6 plugs, to see if the measurements were the same.

To simplify I now just power it with usb power and have only one sensor in CT1. Still no dice.

> INFO OTA successful
> INFO Successfully uploaded program.
> INFO Starting log output from 192.168.xx using esphome API
> INFO Successfully connected to 192.168.xx
> [07:54:37][I][app:102]: ESPHome version 2022.8.0 compiled on Aug 23 2022, 07:53:54
> [07:54:37][C][wifi:491]: WiFi:
> [07:54:37][C][wifi:353]:   Local MAC: 24:0A:C4:58:E4:54
> [07:54:37][C][wifi:354]:   SSID: [redacted]
> [07:54:37][C][wifi:355]:   IP Address: 192.168.xx
> [07:54:37][C][wifi:357]:   BSSID: [redacted]
> [07:54:37][C][wifi:358]:   Hostname: 'edison'
> [07:54:37][C][wifi:360]:   Signal strength: -46 dB ▂▄▆█
> [07:54:37][C][wifi:364]:   Channel: 1
> [07:54:37][C][wifi:365]:   Subnet: 255.255.255.0
> [07:54:37][C][wifi:366]:   Gateway: 192.168.30.1
> [07:54:37][C][wifi:367]:   DNS1: 0.0.0.0
> [07:54:37][C][wifi:368]:   DNS2: 0.0.0.0
> [07:54:37][C][logger:275]: Logger:
> [07:54:37][C][logger:276]:   Level: DEBUG
> [07:54:37][C][logger:277]:   Log Baud Rate: 115200
> [07:54:37][C][logger:278]:   Hardware UART: UART0
> [07:54:37][C][spi:097]: SPI bus:
> [07:54:37][C][spi:098]:   CLK Pin: GPIO18
> [07:54:37][C][spi:099]:   MISO Pin: GPIO19
> [07:54:37][C][spi:100]:   MOSI Pin: GPIO23
> [07:54:37][C][spi:102]:   Using HW SPI: YES
> [07:54:37][C][:129]: ATM90E32:
> [07:54:37][C][atm90e32:130]:   CS Pin: GPIO5
> [07:54:37][E][atm90e32:132]: Communication with ATM90E32 failed!
> [07:54:37][C][atm90e32:134]:   Update Interval: 10.0s
> [07:54:37][C][atm90e32:135]:   Voltage A '6C Volts A'
> [07:54:37][C][atm90e32:135]:     Device Class: 'voltage'
> [07:54:37][C][atm90e32:135]:     State Class: 'measurement'
> [07:54:37][C][atm90e32:135]:     Unit of Measurement: 'V'
> [07:54:37][C][atm90e32:135]:     Accuracy Decimals: 1
> [07:54:37][C][atm90e32:136]:   Current A '6C CT1 Amps'
> [07:54:37][C][atm90e32:136]:     Device Class: 'current'
> [07:54:37][C][atm90e32:136]:     State Class: 'measurement'
> [07:54:37][C][atm90e32:136]:     Unit of Measurement: 'A'
> [07:54:37][C][atm90e32:136]:     Accuracy Decimals: 2
> [07:54:37][C][atm90e32:137]:   Power A '6C CT1 Watts'
> [07:54:37][C][atm90e32:137]:     Device Class: 'power'
> [07:54:37][C][atm90e32:137]:     State Class: 'measurement'
> [07:54:37][C][atm90e32:137]:     Unit of Measurement: 'W'
> [07:54:37][C][atm90e32:137]:     Accuracy Decimals: 2
> [07:54:37][C][atm90e32:156]:   Frequency '6C Freq A'
> [07:54:37][C][atm90e32:156]:     State Class: 'measurement'
> [07:54:37][C][atm90e32:156]:     Unit of Measurement: 'Hz'
> [07:54:37][C][atm90e32:156]:     Accuracy Decimals: 1
> [07:54:37][C][atm90e32:156]:     Icon: 'mdi:current-ac'

CircuitSetup, 2022-08-23:
Hi Max - can you provide your config file? Also, what ESP32 are you using?

max-rousseau, 2022-08-23:
Of course, sorry for not including initially. Pretty vanilla. The board I am using now is Esprsesif ESP32-WROOM-32D. I think this one's a bit older I could try with another one but didn't dig deeper on that front given at least initially it was working.

```
substitutions:
  disp_name: 6C
  update_time: 10s
  current_cal: '26275' # gain_ct # old 27961
  voltage_cal: '15970' # gain_voltage # old '15970'

esphome:
  name: Foo

esp32:
  board: esp32dev
  framework:
    type: arduino

logger:
  # level: VERY_VERBOSE
  
# Enable Home Assistant API
api:
  encryption:
    key: "[redacted]"

ota:
  password: "[redacted]"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  manual_ip:
    static_ip: 192.168.[redacted]
    gateway: 192.168.[redacted]
    subnet: 255.255.255.0

spi:
  clk_pin: 18
  miso_pin: 19
  mosi_pin: 23

sensor:

  - platform: wifi_signal
    name: ${disp_name} WiFi
    update_interval: 60s

  - platform: atm90e32
    cs_pin: 5
    phase_a:
      voltage:
        name: ${disp_name} Volts A
        id: ic1Volts
        accuracy_decimals: 1
      current:
        name: ${disp_name} CT1 Amps
        id: ct1Amps
      power:
        name: ${disp_name} CT1 Watts
        id: ct1Watts
      gain_voltage: ${voltage_cal}
      gain_ct: ${current_cal}

    frequency:
      name: ${disp_name} Freq A
    line_frequency: 60Hz
    gain_pga: 1X
    update_interval: ${update_time}

```

CircuitSetup, 2022-08-23:
Your config looks okay. The only other thing I would check before trying a different ESP32, is trying to connect to the other IC on the meter. Just change `cs_pin: 4`. If that connects, but `cs_pin: 5` does not, then it's an issue with the meter, and I'll send you a new one.

max-rousseau, 2022-08-24:
So I tried another ESP32 (have alot), it was one of the older DEVKITv1 which fits the wider headers. I aligned the 3v headers to the top left corner but I'm pretty sure it ended up cooking the chip (smelled like it). 

Surprisingly enough - I put the smaller 32D back in and it detected the one sensor in CT1 but with really funky readings. Upon rebooting it, now it does not show the communication failed error but there are no readings coming through. Only the wifi signal.

CircuitSetup, 2022-08-24:
Okay, that's strange that the ESP32 popped. As long as it has the same number of pins it should have worked. Maybe it was just on its way out? 

Also very strange that it connected momentarily with the original. It does have 19 pins per side, correct? (should be 1 row empty on the bottom of the meter)

Did you try with `cs_pin: 4`?

max-rousseau, 2022-08-24:
The one that cooked up was the earlier devkit v1 model with 15 pins on each side, while the 3V pin was aligned with the board, perhaps something else downstream didn't line up great. The current one I have in there that works on and off is a 19 pin each side (32D model)

Tried pin 4 but no dice:
> [12:53:48][C][:129]: ATM90E32:
> [12:53:48][C][atm90e32:130]:   CS Pin: GPIO4
> [12:53:48][E][atm90e32:132]: Communication with ATM90E32 failed!

To be fair however, I did try go back to pin 5 this morning and now that is also communications failed. I literally didn't touch anything on the board besides turning off the power last night and turning the test bench back on today... flaky behavior definitely throwing me off.

max-rousseau, 2022-08-25:
After a few reboots there are some readings coming through but they are pretty wild. The load on the test bench is a standard incandescent 75W bulb

> 
> [17:41:48][D][sensor:127]: '6C CT1 Watts': Sending state 8.19200 W with 2 decimals of accuracy
> [17:41:48][D][sensor:127]: '6C Freq A': Sending state 68.46000 Hz with 1 decimals of accuracy
> [17:41:58][D][sensor:127]: '6C Volts A': Sending state 5.80000 V with 1 decimals of accuracy
> [17:41:59][D][sensor:127]: '6C CT1 Amps': Sending state 0.00600 A with 2 decimals of accuracy
> [17:41:59][D][sensor:127]: '6C CT1 Watts': Sending state 3.55232 W with 2 decimals of accuracy
> [17:41:59][D][sensor:127]: '6C Freq A': Sending state 0.53000 Hz with 1 decimals of accuracy
> [17:42:02][D][sensor:127]: '6C WiFi': Sending state -53.00000 dBm with 0 decimals of accuracy
> [17:42:08][D][sensor:127]: '6C Volts A': Sending state 0.01000 V with 1 decimals of accuracy
> [17:42:08][D][sensor:127]: '6C CT1 Amps': Sending state 0.00000 A with 2 decimals of accuracy
> [17:42:08][D][sensor:127]: '6C CT1 Watts': Sending state -20954.15234 W with 2 decimals of accuracy
> [17:42:08][D][sensor:127]: '6C Freq A': Sending state 0.53000 Hz with 1 decimals of accuracy
> [17:42:18][D][sensor:127]: '6C Volts A': Sending state 0.32000 V with 1 decimals of accuracy
> [17:42:18][D][sensor:127]: '6C CT1 Amps': Sending state 0.57700 A with 2 decimals of accuracy
> [17:42:18][D][sensor:127]: '6C CT1 Watts': Sending state 8.84736 W with 2 decimals of accuracy
> [17:42:18][D][sensor:127]: '6C Freq A': Sending state 69.77000 Hz with 1 decimals of accuracy
> [17:42:28][D][sensor:127]: '6C Volts A': Sending state 0.01000 V with 1 decimals of accuracy
> [17:42:29][D][sensor:127]: '6C CT1 Amps': Sending state 0.00000 A with 2 decimals of accuracy
> [17:42:29][D][sensor:127]: '6C CT1 Watts': Sending state 0.00032 W with 2 decimals of accuracy
> [17:42:29][D][sensor:127]: '6C Freq A': Sending state 118.62000 Hz with 1 decimals of accuracy
> [17:42:38][D][sensor:127]: '6C Volts A': Sending state 0.31000 V with 1 decimals of accuracy
> [17:42:38][D][sensor:127]: '6C CT1 Amps': Sending state 0.00100 A with 2 decimals of accuracy
> [17:42:38][D][sensor:127]: '6C CT1 Watts': Sending state -0.00224 W with 2 decimals of accuracy
> [17:42:38][D][sensor:127]: '6C Freq A': Sending state 0.53000 Hz with 1 decimals of accuracy
> [17:42:48][D][sensor:127]: '6C Volts A': Sending state 0.33000 V with 1 decimals of accuracy
> [17:42:48][D][sensor:127]: '6C CT1 Amps': Sending state 0.57600 A with 2 decimals of accuracy
> [17:42:48][D][sensor:127]: '6C CT1 Watts': Sending state -20964.31055 W with 2 decimals of accuracy
> [17:42:48][D][sensor:127]: '6C Freq A': Sending state 43.24000 Hz with 1 decimals of accuracy
> [17:42:58][D][sensor:127]: '6C Volts A': Sending state 0.33000 V with 1 decimals of accuracy
> [17:42:58][D][sensor:127]: '6C CT1 Amps': Sending state 0.00000 A with 2 decimals of accuracy
> [17:42:58][D][sensor:127]: '6C CT1 Watts': Sending state -20.97152 W with 2 decimals of accuracy
> [17:42:58][D][sensor:127]: '6C Freq A': Sending state 36.54000 Hz with 1 decimals of accuracy
> [17:43:02][D][sensor:127]: '6C WiFi': Sending state -47.00000 dBm with 0 decimals of accuracy

CircuitSetup, 2022-10-21:
Was the above without the AC transformer hooked up? When volts are left floating, it can show results like this for power and frequency.

max-rousseau, 2022-10-25:
Yes the device was powered by the AC transformer.

CircuitSetup, 2022-10-25:
Okay, is the ESP32 also being powered by micro USB?

Kt527, 2024-04-20:
I had same issue. Do you solve the problem and how? All the atm90 chip error fail to communicate. I did try usb power or from power adapter.

closed 2025-01-27 by CircuitSetup