CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter #112: Communication with ATM90E32 failed Error

haas844, 2022-10-09:
Hi,

I recently tried setting up the 6ch main board (v1.4, rev1) to work with Homeassistant through ESPHome, and keep getting the following error for both cs_pin: 4 and cs_pin: 5:

[E][atm90e32:132]: Communication with ATM90E32 failed!

I am currently using the Jameco 157041 power supply, with a HiLetgo ESP-WROOM-32 board (https://www.amazon.com/gp/product/B0718T232Z). 

When I check the logs, I get regular updates on the WiFi status and nothing for the CT sensors. Homeassistant currently shows all the sensors as 'Unknown'. 

### Config
```
substitutions: 
  disp_name: 6C
  update_time: 10s
  current_cal: '27961'
  voltage_cal: '7305'

esphome:
  name: homeenergymeter

esp32:
  board: esp32dev
  framework:
    type: arduino

logger:

api:
  encryption:
    key: "redacted"

ota:
  password: "redacted"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

spi:
  clk_pin: 18
  miso_pin: 19
  mosi_pin: 23

sensor:
  - platform: wifi_signal
    name: ${disp_name} WiFi
    update_interval: 60s

  - platform: atm90e32
    cs_pin: 5
    phase_a:
      voltage:
        name: ${disp_name} Volts A
        id: ic1Volts
        accuracy_decimals: 1
      current:
        name: ${disp_name} CT1 Amps
        id: ct1Amps
      gain_voltage: ${voltage_cal}
      gain_ct: ${current_cal}
    phase_b:
      current:
        name: ${disp_name} CT2 Amps
        id: ct2Amps
      gain_ct: ${current_cal}
    phase_c:
      current:
        name: ${disp_name} CT3 Amps
        id: ct3Amps
      gain_ct: ${current_cal}
    frequency:
      name: ${disp_name} Freq A
    line_frequency: 60Hz
    gain_pga: 1X
    update_interval: ${update_time}

  - platform: atm90e32
    cs_pin: 4
    phase_a:
      voltage:
        name: ${disp_name} Volts B
        id: ic2Volts
        accuracy_decimals: 1
      current:
        name: ${disp_name} CT4 Amps
        id: ct4Amps
      gain_voltage: ${voltage_cal}
      gain_ct: ${current_cal}
    phase_b:
      current:
        name: ${disp_name} CT5 Amps
        id: ct5Amps
      gain_ct: ${current_cal}
    phase_c:
      current:
        name: ${disp_name} CT6 Amps
        id: ct6Amps
      gain_ct: ${current_cal}
    frequency:
      name: ${disp_name} Freq B
    line_frequency: 60Hz
    gain_pga: 1X
    update_interval: ${update_time}  

switch:
  - platform: restart
    name: ${disp_name} Restart
```
### Logs
```
INFO Reading configuration /config/esphome/homeenergymeter.yaml...
WARNING GPIO4 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Starting log output from homeenergymeter.local using esphome API
INFO Successfully connected to homeenergymeter.local
[13:55:04][I][app:102]: ESPHome version 2022.9.4 compiled on Oct  9 2022, 13:43:41
[13:55:04][C][wifi:502]: WiFi:
[13:55:04][C][wifi:360]:   Local MAC: [redacted]
[13:55:04][C][wifi:361]:   SSID: [redacted]
[13:55:04][C][wifi:362]:   IP Address: [redacted]
[13:55:04][C][wifi:364]:   BSSID: [redacted]
[13:55:04][C][wifi:365]:   Hostname: 'homeenergymeter'
[13:55:04][C][wifi:367]:   Signal strength: -62 dB ▂▄▆█
[13:55:04][C][wifi:371]:   Channel: 6
[13:55:04][C][wifi:372]:   Subnet: 255.255.255.0
[13:55:04][C][wifi:373]:   Gateway: 192.168.1.1
[13:55:04][C][wifi:374]:   DNS1: 8.8.8.8
[13:55:04][C][wifi:375]:   DNS2: 8.8.4.4
[13:55:04][C][logger:275]: Logger:
[13:55:04][C][logger:276]:   Level: DEBUG
[13:55:04][C][logger:277]:   Log Baud Rate: 115200
[13:55:04][C][logger:278]:   Hardware UART: UART0
[13:55:04][C][spi:101]: SPI bus:
[13:55:04][C][spi:102]:   CLK Pin: GPIO18
[13:55:04][C][spi:103]:   MISO Pin: GPIO19
[13:55:04][C][spi:104]:   MOSI Pin: GPIO23
[13:55:04][C][spi:106]:   Using HW SPI: YES
[13:55:04][C][:129]: ATM90E32:
[13:55:04][C][atm90e32:130]:   CS Pin: GPIO5
[13:55:04][E][atm90e32:132]: Communication with ATM90E32 failed!
[13:55:04][C][atm90e32:134]:   Update Interval: 10.0s
[13:55:04][C][atm90e32:135]:   Voltage A '6C Volts A'
[13:55:04][C][atm90e32:135]:     Device Class: 'voltage'
[13:55:04][C][atm90e32:135]:     State Class: 'measurement'
[13:55:04][C][atm90e32:135]:     Unit of Measurement: 'V'
[13:55:04][C][atm90e32:135]:     Accuracy Decimals: 1
[13:55:04][C][atm90e32:136]:   Current A '6C CT1 Amps'
[13:55:04][C][atm90e32:136]:     Device Class: 'current'
[13:55:04][C][atm90e32:136]:     State Class: 'measurement'
[13:55:04][C][atm90e32:136]:     Unit of Measurement: 'A'
[13:55:04][C][atm90e32:136]:     Accuracy Decimals: 2
[13:55:04][C][atm90e32:143]:   Current B '6C CT2 Amps'
[13:55:04][C][atm90e32:143]:     Device Class: 'current'
[13:55:04][C][atm90e32:143]:     State Class: 'measurement'
[13:55:04][C][atm90e32:143]:     Unit of Measurement: 'A'
[13:55:04][C][atm90e32:143]:     Accuracy Decimals: 2
[13:55:04][C][atm90e32:150]:   Current C '6C CT3 Amps'
[13:55:04][C][atm90e32:150]:     Device Class: 'current'
[13:55:04][C][atm90e32:150]:     State Class: 'measurement'
[13:55:04][C][atm90e32:150]:     Unit of Measurement: 'A'
[13:55:04][C][atm90e32:150]:     Accuracy Decimals: 2
[13:55:04][C][atm90e32:156]:   Frequency '6C Freq A'
[13:55:04][C][atm90e32:156]:     State Class: 'measurement'
[13:55:04][C][atm90e32:156]:     Unit of Measurement: 'Hz'
[13:55:04][C][atm90e32:156]:     Accuracy Decimals: 1
[13:55:04][C][atm90e32:156]:     Icon: 'mdi:current-ac'
[13:55:04][C][:129]: ATM90E32:
[13:55:04][C][atm90e32:130]:   CS Pin: GPIO4
[13:55:04][E][atm90e32:132]: Communication with ATM90E32 failed!
[13:55:04][C][atm90e32:134]:   Update Interval: 10.0s
[13:55:04][C][atm90e32:135]:   Voltage A '6C Volts B'
[13:55:04][C][atm90e32:135]:     Device Class: 'voltage'
[13:55:04][C][atm90e32:135]:     State Class: 'measurement'
[13:55:04][C][atm90e32:135]:     Unit of Measurement: 'V'
[13:55:05][C][atm90e32:135]:     Accuracy Decimals: 1
[13:55:05][C][atm90e32:136]:   Current A '6C CT4 Amps'
[13:55:05][C][atm90e32:136]:     Device Class: 'current'
[13:55:05][C][atm90e32:136]:     State Class: 'measurement'
[13:55:05][C][atm90e32:136]:     Unit of Measurement: 'A'
[13:55:05][C][atm90e32:136]:     Accuracy Decimals: 2
[13:55:05][C][atm90e32:143]:   Current B '6C CT5 Amps'
[13:55:05][C][atm90e32:143]:     Device Class: 'current'
[13:55:05][C][atm90e32:143]:     State Class: 'measurement'
[13:55:05][C][atm90e32:143]:     Unit of Measurement: 'A'
[13:55:05][C][atm90e32:143]:     Accuracy Decimals: 2
[13:55:05][C][atm90e32:150]:   Current C '6C CT6 Amps'
[13:55:05][C][atm90e32:150]:     Device Class: 'current'
[13:55:05][C][atm90e32:150]:     State Class: 'measurement'
[13:55:05][C][atm90e32:150]:     Unit of Measurement: 'A'
[13:55:05][C][atm90e32:150]:     Accuracy Decimals: 2
[13:55:05][C][atm90e32:156]:   Frequency '6C Freq B'
[13:55:05][C][atm90e32:156]:     State Class: 'measurement'
[13:55:05][C][atm90e32:156]:     Unit of Measurement: 'Hz'
[13:55:05][C][atm90e32:156]:     Accuracy Decimals: 1
[13:55:05][C][atm90e32:156]:     Icon: 'mdi:current-ac'
[13:55:05][C][restart:022]: Restart Switch '6C Restart'
[13:55:05][C][restart:022]:   Icon: 'mdi:restart'
[13:55:05][C][mdns:100]: mDNS:
[13:55:05][C][mdns:101]:   Hostname: homeenergymeter
[13:55:05][C][ota:089]: Over-The-Air Updates:
[13:55:05][C][ota:090]:   Address: homeenergymeter.local:3232
[13:55:05][C][ota:093]:   Using Password.
[13:55:05][C][api:138]: API Server:
[13:55:05][C][api:139]:   Address: homeenergymeter.local:6053
[13:55:05][C][api:141]:   Using noise encryption: YES
[13:55:05][C][wifi_signal.sensor:009]: WiFi Signal '6C WiFi'
[13:55:05][C][wifi_signal.sensor:009]:   Device Class: 'signal_strength'
[13:55:05][C][wifi_signal.sensor:009]:   State Class: 'measurement'
[13:55:05][C][wifi_signal.sensor:009]:   Unit of Measurement: 'dBm'
[13:55:05][C][wifi_signal.sensor:009]:   Accuracy Decimals: 0
[13:55:09][D][sensor:127]: '6C WiFi': Sending state -61.00000 dBm with 0 decimals of accuracy
[13:56:09][D][sensor:127]: '6C WiFi': Sending state -62.00000 dBm with 0 decimals of accuracy
[13:57:09][D][sensor:127]: '6C WiFi': Sending state -61.00000 dBm with 0 decimals of accuracy
[13:58:09][D][sensor:127]: '6C WiFi': Sending state -61.00000 dBm with 0 decimals of accuracy
[13:59:09][D][sensor:127]: '6C WiFi': Sending state -61.00000 dBm with 0 decimals of accuracy
[14:00:09][D][sensor:127]: '6C WiFi': Sending state -61.00000 dBm with 0 decimals of accuracy
[14:01:09][D][sensor:127]: '6C WiFi': Sending state -61.00000 dBm with 0 decimals of accuracy
[14:02:09][D][sensor:127]: '6C WiFi': Sending state -61.00000 dBm with 0 decimals of accuracy
[14:03:09][D][sensor:127]: '6C WiFi': Sending state -62.00000 dBm with 0 decimals of accuracy
```

CircuitSetup, 2022-10-21:
I"m sorry the meter is giving you issues. Everything looks okay in the logs. 

Can you confirm that the ESP32 is being powered by the meter?

haas844, 2022-10-21:
Hi,

I believe it was an issue with the ESP32 I was using, I switched to the following board and everything started working as it should:

KeeYees ESP32S
https://www.amazon.com/gp/aw/d/B07QCP2451

Thanks!

closed 2022-10-21 by haas844