CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter #125: Off topic - estimate time for new stock

ray-matos-mpf, 2023-01-07:
Estimated time for boards to be back in stock?

Azylog, 2023-03-20:
Hello,
Are you still making/selling the 6C energy meter ? I sent an e-mail 2 weeks ago about getting it by myself and got no answer.
News about stock is also pretty old on the shop.
I understand perfectly you don't want/can start new batches as it takes some time to check highly variable stocks and so on.

My main question is : should we get it by our own so you can focus on software/anything else ?

I received 5 assembled PCB from JLCPCB last friday. Just missing the female headers which I soldered myself.
And I just ordered 20 add-ons cards, now JLCPCB is almost out of stock for ATM90E32AS ;-(.

You made a really nice open-hardware stuff, ordering it directly is making me a bit ashamed ...

I'm already thinking of some modifications that could be some fun : add ESP32 directly on the board, remove J16, check if BR1 should really be 2A (hard to find), try to get rid of those stackable headers, etc ...

Thanks,
Julien

CircuitSetup, 2023-03-20:
The 6 channel energy meters will be in stock this Thursday! 

@Azylog feel free to make your own! I thought about putting the module directly on the pcb, but this would make everything larger, and would eliminate the stackable feature of the add-on boards. BR1 does not need to be 2A, also.

Azylog, 2023-03-21:
Hey ! Thanks for your answer ! Glad to hear you could had some stock.

You don't necessarily to put a MODULE on the board. As you already provide 3.3V, it could simply be an IC, an 40Mhz XTAL and a few 0603 condensators. The main concern here is probably the antenna.
Wemos have already done most of the work here with ESP32-S2FH4 : https://www.wemos.cc/en/latest/_static/files/sch_s2_mini_v1.0.0.pdf
But I wonder if that could still be routed on a 2 layers PCB ? I'll try in a few weeks and try to share the results. Off course, we should keep the stackable add-on feature by routing power, VA and all SPI pins (MOSI, MISO, CKL and 10xCS if I'm right).

For now, I'll focus on software. For example, I'm trying to define current_cal from a select in esphome.

And finally, about BR1 : LMR14020 is rated for 2A, that's probably why you defined it as 50V/2A. Any idea about the max current used by IC3 with 5 add-on boards ?
I could measure it when I receive my add-on boards but you probably already have done it ?

Thanks !

(and perhaps should I open a new thread, sorry)

holblin, 2023-08-12:
Stocks seems to be still an issue 😕 any way to get a two phases, 30 or 36 channels bundle 😶
I'm doing a panel change and wanted to throw it while the electrician is there!

Azylog, 2023-08-12:
I still have a few boards from my batch as I made more than used actually. I can sell them but only in EU and in euros, this is not my business and I won't struggle with customs. And provide an invoice, of course.

holblin, 2023-08-13:
I'm in US :-/

closed 2024-07-25 by CircuitSetup