CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter #127: Power (watts) not calculated for IC2 on addon board

is343, 2023-01-16:
Hello! I just setup my addon board and am running into an issue where the addon board's IC2 is picking up current values but is not calculating any power.
I am measuring 2 voltages with 2 adapters plugged into different phase outlets. I severed the JP13 and JP12 connections on both the mainboard and the addon board. The instructions don't specifically mention also severing the addon boards so maybe this is where the issue is from? I was speculating to myself that maybe it's an issue of it not getting any voltage so it's calculating based on 0 volts, but I have no idea if that is the case or not.

Both boards are v1.4.

Everything is working fine for CT 1-10. It's only CT 11, 12, and 13 that are doing this.

Any idea what might be going on with this?

[image]

I've checked over my yaml a few times now. I don't think I'm missing anything, but here is the the main sensor setup:

```
sensor:
  - platform: wifi_signal
    name: ${disp_name} WiFi
    update_interval: 60s
#IC1 Main
  - platform: atm90e32
    cs_pin: 5
    frequency:
      name: ${disp_name} Frequency
    line_frequency: 60Hz
    gain_pga: 1X
    update_interval: ${update_time}
    chip_temperature: 
      name: ${disp_name} Port 1-3 Chip Temp
    # Port 1
    # MAINS INPUT A
    phase_a:
      voltage:
        name: ${disp_name} Volts A
        id: ic1Volts
        accuracy_decimals: 1
      current:
        name: ${disp_name} CT1 Amps
        id: ct1Amps
        accuracy_decimals: 3
# The max value for current that the meter can output is 65.535. If you expect to measure current over 65A, 
# divide the gain_ct by 2 (120A CT) or 4 (200A CT) and multiply the current and power values by 2 or 4 by uncommenting the filter below
#        filters:
#          - multiply: 2
      power:
        name: ${disp_name} CT1 Watts
        id: ct1Watts
        accuracy_decimals: 2
#        filters:
#          - multiply: 2
      reactive_power:
        name: ${disp_name} CT1 Reactive Power
        id: ct1ReactivePower
      power_factor:
        name: ${disp_name} CT1 Power Factor
        id: ct1PowerFactor
      gain_voltage: ${voltage_cal}
      gain_ct: ${current_cal_80A}
    # Port 2
    # ELECTRIC RANGE A
    phase_b:
      current:
        name: ${disp_name} CT2 Amps
        id: ct2Amps
        accuracy_decimals: 3
      power:
        name: ${disp_name} CT2 Watts
        id: ct2Watts
        accuracy_decimals: 2
      reactive_power:
        name: ${disp_name} CT2 Reactive Power
        id: ct2ReactivePower
      power_factor:
        name: ${disp_name} CT2 Power Factor
        id: ct2PowerFactor
      gain_voltage: ${voltage_cal}
      gain_ct: ${current_cal_120A}
    # Port 3
    # WATER HEATER A
    phase_c:
      current:
        name: ${disp_name} CT3 Amps
        id: ct3Amps
        accuracy_decimals: 3
      power:
        name: ${disp_name} CT3 Watts
        id: ct3Watts
        accuracy_decimals: 2
      reactive_power:
        name: ${disp_name} CT3 Reactive Power
        id: ct3ReactivePower
      power_factor:
        name: ${disp_name} CT3 Power Factor
        id: ct3PowerFactor
      gain_voltage: ${voltage_cal}
      gain_ct: ${current_cal_80A}

# #IC2 Main
  - platform: atm90e32
    cs_pin: 4
    # frequency:
    #   name: ${disp_name} Port 4-6 Frequency
    line_frequency: 60Hz
    gain_pga: 1X
    update_interval: ${update_time}
    chip_temperature: 
      name: ${disp_name} Port 4-6 Chip Temp
    # Port 4
    # MAINS INPUT B
    phase_a:
      voltage:
        name: ${disp_name} Volts B
        id: ic2Volts
        accuracy_decimals: 1
      current:
        name: ${disp_name} CT4 Amps
        id: ct4Amps
        accuracy_decimals: 3
      power:
        name: ${disp_name} CT4 Watts
        id: ct4Watts
      reactive_power:
       name: ${disp_name} CT4 Reactive Power
       id: ct4ReactivePower
      power_factor:
        name: ${disp_name} CT4 Power Factor
        id: ct4PowerFactor
      gain_voltage: ${voltage_cal}
      gain_ct: ${current_cal_120A}
    # Port 5
    # WASHER/DRYER
    phase_b:
      current:
        name: ${disp_name} CT5 Amps
        id: ct5Amps
        accuracy_decimals: 3
      power:
        name: ${disp_name} CT5 Watts
        id: ct5Watts
        accuracy_decimals: 2
      reactive_power:
        name: ${disp_name} CT5 Reactive Power
        id: ct5ReactivePower
      power_factor:
        name: ${disp_name} CT5 Power Factor
        id: ct5PowerFactor
      gain_voltage: ${voltage_cal}
      gain_ct: ${current_cal_80A}
    # Port 6
    # ELECTRIC RANGE B
    phase_c:
      current:
        name: ${disp_name} CT6 Amps
        id: ct6Amps
        accuracy_decimals: 3
      power:
        name: ${disp_name} CT6 Watts
        id: ct6Watts
      reactive_power:
       name: ${disp_name} CT6 Reactive Power
       id: ct6ReactivePower
      power_factor:
        name: ${disp_name} CT6 Power Factor
        id: ct6PowerFactor
      gain_voltage: ${voltage_cal}
      gain_ct: ${current_cal_80A}
        

#IC1 AddOn 1
  - platform: atm90e32
    cs_pin: 0
    line_frequency: 60Hz
    gain_pga: 1X
    update_interval: ${update_time}
    # frequency:
    #   name: ${disp_name} Port 7-9 Frequency
    chip_temperature: 
      name: ${disp_name} Port 7-9 Chip Temp
    #Port 7
    # HVAC UNIT 1 A
    phase_a:
      current:
        name: ${disp_name} CT7 Amps
        id: ct7Amps
        accuracy_decimals: 3
      # HAS NEGATIVE VALS AT LOW USAGE
      power:
        name: ${disp_name} CT7 Watts
        id: ct7Watts
        accuracy_decimals: 2
      reactive_power:
        name: ${disp_name} CT7 Reactive Power
        id: ct7ReactivePower
      power_factor:
        name: ${disp_name} CT7 Power Factor
        id: ct7PowerFactor
      gain_voltage: ${voltage_cal}
      gain_ct: ${current_cal_80A}
    # Port 8
    # HVAC UNIT 2 A
    phase_b:
      current:
        name: ${disp_name} CT8 Amps
        id: ct8Amps
        accuracy_decimals: 3
      # HAS NEGATIVE VALS AT LOW USAGE
      power:
        name: ${disp_name} CT8 Watts
        id: ct8Watts
        accuracy_decimals: 2
      reactive_power:
        name: ${disp_name} CT8 Reactive Power
        id: ct8ReactivePower
      power_factor:
        name: ${disp_name} CT8 Power Factor
        id: ct8PowerFactor
      gain_voltage: ${voltage_cal}
      gain_ct: ${current_cal_80A}
    # Port 9
    # BEDROOM / OFFICE OUTLETS
    phase_c:
      current:
        name: ${disp_name} CT9 Amps
        id: ct9Amps
        accuracy_decimals: 3
      power:
        name: ${disp_name} CT9 Watts
        id: ct9Watts
      reactive_power:
       name: ${disp_name} CT9 Reactive Power
       id: ct9ReactivePower
      power_factor:
        name: ${disp_name} CT9 Power Factor
        id: ct9PowerFactor
      gain_voltage: ${voltage_cal}
      gain_ct: ${current_cal_80A}

#IC2 AddOn 1
  - platform: atm90e32
    cs_pin: 16
    line_frequency: 60Hz
    gain_pga: 1X
    update_interval: ${update_time}
    # frequency:
    #   name: ${disp_name} Port 10-12 Frequency
    chip_temperature: 
      name: ${disp_name} Port 10-12 Chip Temp
    #Port 10
    # WATER HEATER B
    phase_a:
      current:
        name: ${disp_name} CT10 Amps
        id: ct10Amps
        accuracy_decimals: 3
      # HAS NEGATIVE VALS AT LOW USAGE
      power:
        name: ${disp_name} CT10 Watts
        id: ct10Watts
        accuracy_decimals: 2
      reactive_power:
        name: ${disp_name} CT10 Reactive Power
        id: ct10ReactivePower
      power_factor:
        name: ${disp_name} CT10 Power Factor
        id: ct10PowerFactor
      gain_voltage: ${voltage_cal}
      gain_ct: ${current_cal_80A}
    # Port 11
    # HVAC UNIT 1 B
    phase_b:
      current:
        name: ${disp_name} CT11 Amps
        id: ct11Amps
        accuracy_decimals: 3
      # HAS NEGATIVE VALS AT LOW USAGE
      power:
        name: ${disp_name} CT11 Watts
        id: ct11Watts
        accuracy_decimals: 2
      reactive_power:
        name: ${disp_name} CT11 Reactive Power
        id: ct11ReactivePower
      power_factor:
        name: ${disp_name} CT11 Power Factor
        id: ct11PowerFactor
      gain_voltage: ${voltage_cal}
      gain_ct: ${current_cal_80A}
    # Port 12
    # HVAC UNIT 2 B
    phase_c:
      current:
        name: ${disp_name} CT12 Amps
        id: ct12Amps
        accuracy_decimals: 3
      power:
        name: ${disp_name} CT12 Watts
        id: ct12Watts
      reactive_power:
       name: ${disp_name} CT12 Reactive Power
       id: ct12ReactivePower
      power_factor:
        name: ${disp_name} CT12 Power Factor
        id: ct12PowerFactor
      gain_voltage: ${voltage_cal}
      gain_ct: ${current_cal_80A}
```

CircuitSetup, 2023-02-03:
Sorry I didn't see your issue earlier. 

For the add-on board, severing the jumpers will not pass the voltage to the right side of the board (CT4-6). So, if you want to calculate wattage, you'll either need to hook up VA2 on the add-on to the other transformer, or bridge the jumpers again to get voltage from the first transformer hooked up to the jack on the main board.

is343, 2023-02-03:
@CircuitSetup thank you for the reply. 

Okay, just to confirm, if I rebridge the jumpers, then IC2 will be using the same voltage from IC1, correct?

So, if I want to keep IC2 using the second voltage, I should also hook the addon board up to the second AC source, just like I did with with the main board.

CircuitSetup, 2023-02-03:
> @CircuitSetup thank you for the reply.
>
> Okay, just to confirm, if I rebridge the jumpers, then IC2 will be using the same voltage from IC1, correct?
>
> So, if I want to keep IC2 using the second [...]

Yes, that is correct.

is343, 2023-02-04:
I was able to jump the voltage from the main board to the addon board, and it's now working just fine. Thanks for clearing this up!

closed 2023-02-04 by is343