CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter #128: Help setup 6Channel Board

xefil, 2023-01-21:
Hi all,
I've bought a 6Channel Board 1.4V rev1 to monitor my home.
I'm living in Italy and here we have 220V Single Phase 50 Hz.
I'm testing it for now with only one SCT-006 on CT1 using this connection diagram, due the split core has not a jack mounted, I had to mount it in that way:

[image]

The N.C. is connected to nothing. Only "l" and "k" are connected to the jack and then into CT1.
The split-core is monitoring the hot wire (tested using a phase finder) placing it in the right way (following the arrow on the split core)
The board is powered using a 12V AC transformer, not DC (input AC220-240 50/60Hz, output AC12V/20-60W) 
The ESP I have is an ESP32 (ESP32-WROOM-32).
The program I've uploaded is this:
[powermeter.txt](https://github.com/CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter/files/10473007/powermeter.txt)

For the calibration I'm using a kill-a-watt (gosund smart plug) which is showing me this values during load tests:

[image]

But on the powermeter board I'm reading only ampere values:
[image]

The device I'm using has a load declared of 1370W-1630W and if I'm not wrong, the read ampere value should be correct: 6,54A x 220V = 1438W which makes sense more or less (without calibration).

But all the other values are empty: Watt, Frequence A, Frequence B (why is there Frequence B???), Volts A, Volts B (why is there Volts B?)

What I'm doing wrong?? I'm testing it since many days and have no idea whats wrong.

Thanks a lot!

Simon

CircuitSetup, 2023-01-24:
Which AC transformer are you using for the voltage reference?

xefil, 2023-01-24:
Hello @CircuitSetup ,
What exactly do you mean? The trasformer is one I have my own, exactly this model:
https://www.amazon.it/dp/B09J4RDSLZ
It's connected using the 9-12V AC trasformer voltage input from your doc:
https://raw.githubusercontent.com/CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter/master/Images/6-channel_diagram.png
The orange square.
Thanks, Simon

CircuitSetup, 2023-01-25:
> Hello @CircuitSetup , What exactly do you mean? The trasformer is one I have my own, exactly this model: https://www.amazon.it/dp/B09J4RDSLZ

Unfortunately this transformer will not work since it is actually a high frequency switched power supply. The transformer has to have an iron core, and output a sine wave for the meter to measure the voltage properly. Something more like this: https://www.amazon.it/Alimentazione-Tensione-Ingresso-Trasformatore-alimentazione/dp/B08MTQQKH3
(I haven't tested this transformer in particular, but in theory it should work)

xefil, 2023-01-31:
Hello @CircuitSetup ,
Sorry for the late reply, but I had to purchase the suggested transformer and now it looks much better, but not resolved totally.
First of all, now the Frequency A and B are perfect: 50.0Hz, but other values are wrong.
I've connected a hairdryer to check the load and here the results, reading from a smartplug (flashed with ESPHome) and your board:

SmartPlug:
[image]

6Channel Board:
[image]

Why Volts are set to ~160V and not ~220V ?
Then, the wrong values, I suppose it's a matter of calibration. Where exactly, based on the above sketch, should I work on to calibrate it?

Thanks a lot for your help.

In waiting for news I wish you a nice day.

Simon

CircuitSetup, 2023-01-31:
Yes, the calibration value in your config.yaml for voltage must be increased so volts match your smartplug. Since you have the values already, you can calculate what it should be using this (assuming values above and default voltage_cal):

(your voltage reading / energy monitor voltage reading) * voltage_cal = New voltage_cal
(220.4/160.5) * 7305 = 10031

xefil, 2023-01-31:
Hi @CircuitSetup ,
220.4 is the voltage reading from the smart plug, thats ok.
160.5 is the voltage reading from the 6channel board - (Volts A). Is that correct?
voltage_cal is derived due I know that the output should be 1034.7W as read from smart plug?
So in that case should be: 220.4 / 160.5 = 1,373208722741433
Then 1034.7 / 1.373208722741433 = 753.4906987295826 ??
Should then I use 753.4906987295826 as voltage_cal ?

And what about "Volts B" ?

Thanks again!

Simon

CircuitSetup, 2023-01-31:
If voltage_cal was 1034.7 in your config when the output was 160.5V, then:

(220.4/160.5) * 1034.7 = 1420.86  (round up to 1421)

Volts B should be the same, and will change as long as voltage_cal is the variable for gain_voltage in your config. [See example here.](https://github.com/CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter/blob/505cfde15aa48bdc9f1fbe24e0685af1f4c47e37/Software/ESPHome/6chan_energy_meter_main_board.yaml#L123)

xefil, 2023-01-31:
Sorry @CircuitSetup if I'm not understanding it the right way.
I was thinking that `voltage_cal` (default '7305') should be calculated as written from me:
`( voltage reading from the smart plug) / (voltage reading from the 6channel board) * "xxx" = 1034.7` and I need to calculate "xxx". BTW I've maybe misunderstood. I'll try your values and let you know.
Thanks, Simon

xefil, 2023-01-31:
Ahaa, ok, because, now, using `1421` I have `Volts A` set to `31.3 V`, even worst :-D

xefil, 2023-01-31:
Ok @CircuitSetup maybe understood.
If using `7305` I get `160.5`V, what to use to have `220.4`V?
`(220.4 x 7305) / 160.5 = 10031,28971962617` :-)
Right? ;-)

CircuitSetup, 2023-01-31:
yes, just use 10031 without decimal numbers

xefil, 2023-02-01:
Thanks @CircuitSetup , now it looks better, I'm searching something better to measure Volts to be more accurate.
Meawhile, why there are two "Volts" readings? A and B?
The values are same but not exactly.
Thanks, Simon

CircuitSetup, 2023-02-03:
> Thanks @CircuitSetup , now it looks better, I'm searching something better to measure Volts to be more accurate. Meawhile, why there are two "Volts" readings? A and B? The values are same but not exactly. Thanks, Simon

Glad it's working for you now! There are A and B readings because of the possibility of reading 2 different voltages. If you're not reading 2, then you can just remove the variable from the config.

xefil, 2023-02-04:
Hi @CircuitSetup I'll try to remove the "B" readings then.
Meanwhile I would like to share the results using a hairdryer with two power levels.
On channel1 I've an 80A/26.6mA SCT-010 and I've used only for that block the current_cal set to '41660'. For the other 5 channels I've used 20A/25mA SCT-006 and current_cal: '11143'.

Here results

Low power hairdryer:
[image]

High power hairdryer:
[image]

As you can see on low power the difference is not much, values between 4.31 and 4.39
on high power I read between 6.03 and 6.36. Is that expected?

Are those values, on low and on high normal with these differences each other on same sensors and on same load applied?

What can I do otherwise?

Thanks,

Simon

PS: I know the watt are negative, I need to turn the sensor, even if I'm followinf the arrow on the phase line... strange...

thanks!

xefil, 2023-02-08:
Hi @CircuitSetup , any suggestion? :)

Simon

Azylog, 2023-03-20:
Hello,
Perhaps should you consider CT Clamp have an precision margin (+/- 1%) and the resistor also have a precision (it depends but if I remember well, it's 1% on the 6C).
Did you tried to switch the CT clamp on the inputs to check if error is always the same ?

And another thought : your hairdryer is an inductive (and probably varying) load. You could try with a pure resistive load (like a old 100W filament or halogen bulb) and see if the results are similar.

xefil, 2023-03-20:
Thanks for the suggestions @Azylog ! Meanwhile I've put the board into the wall monitoring the house. Seems working not so bad at all. I'll wait the first month to compare what got into the bills. Even there is a "not included power" due it cames from the cellar in which I've a nas. But I think I can calculate more or less the percent of consumption over a month.
Thanks!
Simon