CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter #132: Platformio Build for emonesp-nodemcu32s fails; two reasons

ajw685, 2023-02-25:
when building in VSCode straight after downloading the repository and opening in platformio, when running Build All I get this error in emonesp-nodemcu32s,:

src/config.cpp:60:32: error: array must be initialized with a brace-enclosed initializer
 String ct_name[NUM_CHANNELS] = "";

changing from  `String ct_name[NUM_CHANNELS] = "";` 

to  

`String ct_name[NUM_CHANNELS] = {""};` 

resolves this first issue
but then,

It seems there have been changes with the Async library that is causing build problems

looks like lib_deps needs t be updated with
https://github.com/me-no-dev/ESPAsyncWebServer.git

i can't get it to build consistently, but do have emonesp running on the board

DougKennaugh, 2023-07-20:
I am having the same issues.
changes from String ct_name[NUM_CHANNELS] = "";
to
String ct_name[NUM_CHANNELS] = {""};

When I now compile I get the following error and dont know how to proceed:

**collect2.exe: error: ld returned 1 exit status
*** [.pio\build\emonesp-nodemcu32s\firmware.elf] Error 1**

Any ideas please?

Kt527, 2023-10-22:
I had same issue. Is any one can help please?

closed 2025-01-27 by CircuitSetup