CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter #134: ATM90E32 not communicating ESP32C3

jrmurdoch, 2023-03-18:
I'm getting a not communicating error, this has been setup with esphome. The device is a ESP32C3 these use the IDF framework instead of Arduino. Can anyone confirm this will work with esphome set to idf framework?

CircuitSetup, 2023-03-18:
I dont believe the c3 is the same pinout as the regular esp32. Does it have 19 pins per side?

jrmurdoch, 2023-03-18:
The pin out is different it's a atthinker module it's narrower so it doesn't fit so I've hooked it up with jumper wires. I've tested lots of wire combinations and double checking etc. So that why I'm wondering if it's the framework

CircuitSetup, 2023-06-05:
were you able to get this working with the latest version of ESPHome?

jrmurdoch, 2023-06-05:
Yes, I got this working with the C3 module.

closed 2025-01-27 by CircuitSetup