CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter #145: AddOn Boards Power Calculation Wildly Incorrect

sepffuzzball, 2023-07-31:
Hello! I have to assume that the issue that I have is some form of local configuration or incorrect setup but I've wracked my brain for a while and haven't been able to figure it out.

For context, I have been using the 6-channel Energy Meter with ESPHome for almost 2 years now and it's been great, no issues there. Yesterday I finally added 5 add-on boards and took the time to wire things up. Seemingly no problem there either.

But looking at the data being sent to ESPHome, only the original energy meter sensors seem to be providing a 'correct' (or close enough) value, and anything sourced from the add-on boards is reporting a miniscule amount of power usage (though I certainly wish my AC units only took 1-2 watts).

I'm using SCT013 CTs for all 31 of the breakers I'm monitoring and CTs 1-6 seem to be in-line with what I'm expecting.

Maybe I missed something in the documentation for setup? I didn't sever any jumpers or anything, just made sure that the correct pin selection jumpers were setup on the add-on boards.

Example ESPHome log excerpt:
`[13:07:38][D][sensor:094]: 'Power Monitor CT1 Watts B23 Upstairs1': Sending state 241.73599 W with 2 decimals of accuracy
[13:07:38][D][sensor:094]: 'Power Monitor CT2 Watts B25 Living': Sending state 25.30368 W with 2 decimals of accuracy
[13:07:38][D][sensor:094]: 'Power Monitor CT3 Watts B27 Fox': Sending state 124.26944 W with 2 decimals of accuracy
[13:07:38][D][sensor:094]: 'Power Monitor CT28 Watts B14 Washing Machine': Sending state 0.00384 W with 2 decimals of accuracy
[13:07:38][D][sensor:094]: 'Power Monitor CT29 Watts B10/12 Air Conditioner1': Sending state 0.27360 W with 2 decimals of accuracy
[13:07:38][D][sensor:094]: 'Power Monitor CT30 Watts B30 Office': Sending state 0.63456 W with 2 decimals of accuracy
[13:07:38][D][sensor:094]: 'Power Monitor CT16 Watts B16 Bath GFI': Sending state 0.00032 W with 2 decimals of accuracy
[13:07:38][D][sensor:094]: 'Power Monitor CT17 Watts B6/8 Air Conditioner2': Sending state 1.33376 W with 2 decimals of accuracy
[13:07:38][D][sensor:094]: 'Power Monitor CT18 Watts B22 Kitchen GFI Refigerator': Sending state 0.10112 W with 2 decimals of accuracy
[13:07:39][D][sensor:094]: 'Power Monitor CT4 Watts B28 Upstairs2': Sending state 534.17566 W with 2 decimals of accuracy
[13:07:39][D][sensor:094]: 'Power Monitor CT5 Watts B34 Server1': Sending state 178.48639 W with 2 decimals of accuracy
[13:07:39][D][sensor:094]: 'Power Monitor CT6 Watts B36 Server2': Sending state 789.75708 W with 2 decimals of accuracy
[13:07:39][D][sensor:094]: 'Power Monitor CT22 Watts B5/7 Air Handler1': Sending state 0.09632 W with 2 decimals of accuracy
[13:07:39][D][sensor:094]: 'Power Monitor CT23 Watts B18 Garbage Disposal': Sending state 0.00032 W with 2 decimals of accuracy
[13:07:39][D][sensor:094]: 'Power Monitor CT24 Watts B24 Dishwasher': Sending state 0.00032 W with 2 decimals of accuracy
[13:07:40][D][sensor:094]: 'Power Monitor CT13 Watts B29 Smoke Detectors': Sending state 0.00928 W with 2 decimals of accuracy
[13:07:40][D][sensor:094]: 'Power Monitor CT14 Watts B14 Washing Machine': Sending state 0.00192 W with 2 decimals of accuracy
[13:07:40][D][sensor:094]: 'Power Monitor CT15 Watts B9/11 Air Handler2': Sending state 1.26368 W with 2 decimals of accuracy
[13:07:46][D][sensor:094]: 'Power Monitor CT19 Watts B19 Microwave': Sending state 0.00000 W with 2 decimals of accuracy
[13:07:46][D][sensor:094]: 'Power Monitor CT20 Watts B13/15 Water Heater': Sending state 0.00000 W with 2 decimals of accuracy
[13:07:46][D][sensor:094]: 'Power Monitor CT21 Watts B21 Bar': Sending state 0.02080 W with 2 decimals of accuracy
[13:07:46][D][sensor:094]: 'Power Monitor CT10 Watts B2/4 Oven': Sending state 0.00064 W with 2 decimals of accuracy
[13:07:46][D][sensor:094]: 'Power Monitor CT11 Watts B20 Kitchen GFI': Sending state 0.00320 W with 2 decimals of accuracy
[13:07:46][D][sensor:094]: 'Power Monitor CT12 Watts B32 Fuzz Suite': Sending state 0.09600 W with 2 decimals of accuracy
[13:07:47][D][sensor:094]: 'Power Monitor CT7 Watts B1/3 Cooktop': Sending state 0.00640 W with 2 decimals of accuracy
[13:07:47][D][sensor:094]: 'Power Monitor CT8 Watts B38 Dishwasher & Attic Lights': Sending state 0.08544 W with 2 decimals of accuracy
[13:07:47][D][sensor:094]: 'Power Monitor CT9 Watts B26 Front Bedrooms & Doorbell': Sending state 0.02688 W with 2 decimals of accuracy
[13:07:48][D][sensor:094]: 'Power Monitor CT25 Watts B17 Range Hood & Power Monitor': Sending state 0.00288 W with 2 decimals of accuracy
[13:07:48][D][sensor:094]: 'Power Monitor CT26 Watts B35/37 Dryer': Sending state 0.00160 W with 2 decimals of accuracy
[13:07:48][D][sensor:094]: 'Power Monitor CT27 Watts B33 Kitchen Lights & Garage': Sending state 0.01408 W with 2 decimals of accuracy
[13:07:48][D][sensor:094]: 'Power Monitor CT31 Watts B31 Lounge': Sending state 0.02272 W with 2 decimals of accuracy`

ESPHome YAML Config
`# Example config for when jp8-jp11 are all bridged - this connects all the voltage channels and allows for power to be calculated by the meter directly.
# Boards >= v1.4 jp8-jp11 are removed and have all voltage channels connected

substitutions:
# Change the disp_name to something you want  
  name: power-monitor
  friendly_name: "Power Monitor"
# Interval of how often the power is updated
  update_time: 10s
# Current Transformers:
#  20A/25mA SCT-006: 11143
#  30A/1V SCT-013-030: 8650
#  50A/1V SCT-013-050: 15420
#  80A/26.6mA SCT-010: 41660
#  100A/50ma SCT-013-000: 27518
#  120A/40mA: SCT-016: 41787
#  200A/100mA SCT-024: 27518
  current_cal: '27518'
  #current_cal: '29575'
  #current_cal_big: '55036'
# Jameco 9VAC Transformer: 
#  For meter versions: 
#  >= v1.3: 7305
  voltage_cal: '7305'
  ip: 10.0.13.107
  
[... 368 lines cut ...]
  #      id: ct36Watts
  #    gain_voltage: ${voltage_cal}
  #    gain_ct: ${current_cal}
  #  line_frequency: 60Hz
  #  gain_pga: 1X
  #  update_interval: ${update_time}

text_sensor:
  - platform: version
    name: "${friendly_name} ESPHome Version"
  - platform: wifi_info
    ip_address:
      name: "${friendly_name} IP Address"

# setup a button to restart the device
button:
  - platform: restart
    id: restart_button
    name: "${friendly_name} Restart"

time:
  - platform: sntp
    id: sntp_time`

Thanks for any help or pointers!

CircuitSetup, 2023-07-31:
In your config, there are filters doing a multiply: -1. If you're getting negative power, you should turn the CT around on the wire. This means it's out of phase with the voltage measurement, and getting a low power factor, which will make the power reading low. 

If all of the readings are positive coming out of ESPHome with the multiply: -1, then all CTs are backwards. An easy fix is to turn the voltage transformer around in the wall plug. This will reverse the phase 180 degrees and make power positive. Then remove all multiply: -1 in the config.

sepffuzzball, 2023-07-31:
I can certainly give that a try when I get home, but even on the base board (CT1, CT4 and CT5) I have the filter on and the values are what I'd expect, just negative.

Doesn't seem like that would resolve all the other CTs that are running low though (they were not installed with any direction in mind so they're about 50/50 on what direction they were installed). Is there something else that might cause that? That's pretty consistent on all of the add on boards.

CircuitSetup, 2023-07-31:
None of the jumpers on the back of the board are cut, correct? Also, all boards are v1.4?

Can you add voltage and power factor to 1 or more of the add-on board current channels that are showing low power? If these are also low, then that will pinpoint the problem a bit more.

sepffuzzball, 2023-07-31:
Okay! Also thank you for taking the time to help me with this. I've added voltage and power factor to my induction stovetop, below are some of the relevant bits out of the log. It definitely shows change which is at least good?

Before turning on:
[17:27:06][D][sensor:094]: 'Power Monitor CT7 Voltage B1/3 Cooktop': Sending state 0.13000 V with 2 decimals of accuracy
[17:27:06][D][sensor:094]: 'Power Monitor CT7 Watts B1/3 Cooktop': Sending state 0.00320 W with 2 decimals of accuracy
[17:27:06][D][sensor:094]: 'Power Monitor CT7 Power Factor B1/3 Cooktop': Sending state 0.76900  with 2 decimals of accuracy
[17:27:16][D][sensor:094]: 'Power Monitor CT7 Voltage B1/3 Cooktop': Sending state 0.13000 V with 2 decimals of accuracy
[17:27:16][D][sensor:094]: 'Power Monitor CT7 Watts B1/3 Cooktop': Sending state 0.00320 W with 2 decimals of accuracy
[17:27:16][D][sensor:094]: 'Power Monitor CT7 Power Factor B1/3 Cooktop': Sending state 0.76900  with 2 decimals of accuracy

After turning on: 
[17:30:17][D][sensor:094]: 'Power Monitor CT7 Voltage B1/3 Cooktop': Sending state 0.13000 V with 2 decimals of accuracy
[17:30:17][D][sensor:094]: 'Power Monitor CT7 Watts B1/3 Cooktop': Sending state 0.19040 W with 2 decimals of accuracy
[17:30:17][D][sensor:094]: 'Power Monitor CT7 Power Factor B1/3 Cooktop': Sending state 0.78700  with 2 decimals of accuracy
[17:30:26][D][sensor:094]: 'Power Monitor CT7 Voltage B1/3 Cooktop': Sending state 0.13000 V with 2 decimals of accuracy
[17:30:26][D][sensor:094]: 'Power Monitor CT7 Watts B1/3 Cooktop': Sending state 0.18976 W with 2 decimals of accuracy
[17:30:27][D][sensor:094]: 'Power Monitor CT7 Power Factor B1/3 Cooktop': Sending state 0.78400  with 2 decimals of accuracy

Here's a shot of the back of one of the add-on boards, all jumpers are still bridged. They're all v1.4 rev1 boards as well.
[image]

CircuitSetup, 2023-08-01:
Okay, it looks like the add-ons aren't getting proper voltage. They are plugged into the main board, which is being powered by an AC transformer, correct?  Nothing else besides other meters are using the 2 bottom pins (labeled VA-, VA+ on the bottom of the add-on boards)

sepffuzzball, 2023-08-01:
Yeah, they're plugged into the main board which is powered by a Jameco AC 120V 60Hz to AC 9 Volt adapter plugged right into the board and nothing at all is using VA- or VA+ on any of the add-ons. Would it be worth me sourcing a different AC to AC adapter?

sepffuzzball, 2023-08-02:
Okay, looks like we've managed to self resolve! Had one of my EE friends take a look at it once it sounded like a voltage problem and this is what they said:

The negative line to the AC supply for the add-on boards was bent out and not connected, and the positive AC supply pin on the main board had a plug that locks it out from being plugged into. So neither AC voltage pins were connected to any of the add on boards.

He was able to fix it quickly and now things are looking good! My previous calibration also seems to be quite in-line with the power monitoring I'm getting from my Powerwalls well within expectations! Anyways, I really appreciate you taking the time to help troubleshoot! :)

[image]

closed 2023-08-02 by sepffuzzball

CircuitSetup, 2023-08-02:
> The negative line to the AC supply for the add-on boards was bent out and not connected, and the positive AC supply pin on the main board had a plug that locks it out from being plugged into. So neither AC voltage pins were connected to any of the add on boards.

Glad you figured it out! This was my next question. The plug is put in to prevent people from inserting the ESP32 into that bottom row. If they do, things start smoking :)