CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter #148: Bigger board

holblin, 2023-08-12:
Hi 👋
First of all, thanks for that project!
I was wondering if there is any plans for a bigger board (or different hook up for extensions)
The stacks from what I could see is pretty big when you start to have 30 channels for example.
I saw other designs like the KC868-M16v2 which are way less good (resolution and so on) but which are also with more channels and maybe more user friendly (jumpers). In any case it seems a replacement but it could help for inspiration!
I would be happy to discuss about it more.

Azylog, 2023-08-12:
Hello,
I was wondering if we could replace ATM90E32 with MCP3914 (8 channels). And keep the ESP32 (that could be embedded on the board) for esphome but with a different stacking system : SMD headers would be easier to produce than through-hole long headers (hard to find + hand soldering).

That makes a lot of work, any other idea/suggestion ?

holblin, 2023-08-13:
I think ATM90E32 is already fully compatible esphome so I would stick with it personally.

I'm not an expert but I would have designed the board to be 6 or 12 channels with an SMD header to connect them side by side. If people want to stack them, I would allow the design to have a hex spacer to stack them (with the communication or without).

Also, I would have made all the boards not specialized (main vs extension). The worst case having an ESP32 host board (that could be easily adapted for new ESP32 or other flavors like Raspberry pi).

Though I have 0 experience in selling boards. I have 1 experience in making my own PCB (which is fun, I'm ready to redo it!).
But yeah, I would look for everything SMD mounted with compatible assembly service :-)