CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter #159: Update YAMLs to conform to Home Assistant developer guidelines?

pyrodex, 2023-09-15:
Hello I am seeing the following errors in my HA logs:

`
2023-09-14 20:04:05.616 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.house_energy_monitor_02_total_watts (<class 'homeassistant.components.esphome.sensor.EsphomeSensor'>) is using native unit of measurement 'W' which is not a valid unit for the device class ('energy') it is using; expected one of ['MJ', 'GJ', 'kWh', 'Wh', 'MWh']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+esphome%22
`

Should the ESPHome YAML be adjusted to correct these type of invalid measurement types?

rkm3, 2024-05-27:
I had the same error and think the PR I just issued will fix it. Happy to revise as needed.

CircuitSetup, 2024-05-27:
Fixed in #177

closed 2024-07-25 by CircuitSetup