CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter #167: Replacement for ESP32-DevKitC-32U, which is NRND?

jpwheless, 2024-01-07:
The ESP32-DevKitC-32U (listed in the readme for use with an external antenna) is now Not Recommended for New Designs (NRND) and isn't stocked anymore on Digikey or Mouser.

Is there a newer ESP32 dev board, like an S2, S3, C3, or C6 that would work?  I'm not seeing an official Espressif board with a compatible pinout - most have different numbers of pins, or the 3V3 or CLK pins in places other than the corners.

CircuitSetup, 2024-01-07:
It looks like they put an E on the end of the model numbers. See here: https://www.mouser.com/ProductDetail/Espressif-Systems/ESP32-DevKitC-32UE?qs=sGAEpiMZZMuqBwn8WqcFUj1SFkunHY100yk2Fr1NGzQdaQh4JSK1cA%3D%3D

I dont believe there are any differences between them besides the UE having the latest ESP32 ECO V3.

jpwheless, 2024-01-11:
Thank you, ordered!

closed 2024-01-11 by jpwheless