CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter #169: Current Scaling Not Correct

chksome, 2024-01-23:
Hey CS & Github,

I'm running a 6 channel 1.4 rev1 board with (for the sake of this troubleshooting, although it happens on all of my readings with 20, 80, and 120A clamps across all six channels) SCT006-B6 20A/25mA clamp recently purchased from CS, and ESPHome 2023.12.8. I'm testing with the clamp placed around a wire in my breaker panel that goes to a dedicated circuit with the singular outlet about five feet (and 20ish feet of romex) away. My load is a heat gun with two settings and clamp readings compared against a Kill-A-Watt. It's being run from the 9V AC Jameco transformer purchased from CS.

I can use the formulas from the [ESPHome Page for the 90e32](https://esphome.io/components/sensor/atm90e32.html#current) to calibrate the current and get ESPHome logs to within a tenth of a watt of the Kill-A-Watt. The problem arises when I change the load by bumping the heat gun up to high, for example. At low, the KAW claims 7.06A, and as said, the phase (c from chip 1) matches. If I turn the gun on high the Kill-A-Watt jumps up to 9.8A but the board jumps up to a little over 18A, a huge difference.

Things I've tried:

- Changing the gain_pga to all three values and recalibrating. The gun high current reading changes by differing amounts, but all wildly inaccurate, anywhere from two to nine amps off
- Calibrating at the heat gun high setting. The same thing happens when turning it to low, the reading of the board decreases, but is inaccurate
- Trying to calibrate on a different dedicated circuit
- A completely new set of clamps
- Changing phases across chips
- Different versions of ESPHome. I've tried as far back as 2023.7.1 with various steps in between (when it didn't work, I gave up for a bit and would come back periodically)

I'm open to any and all ideas or other things to try to get this meter to scale properly across loads.

Thanks!

CircuitSetup, 2024-01-23:
This may be the difference between the 6 channel meter measuring VA, whereas the KAW measures Power directly (via a shunt). The difference being that VA includes the total apparent power ((real power + reactive power) * power factor). Does your KAW have a VA button? 

That being said, a hair dryer shouldn't have much reactive power. The 20A CT tends to be less accurate at the top of its range, but 18A for a hair dryer still seems very high. Are you getting the same results with the 80A and 120A CTs?

chksome, 2024-01-24:
It does! With the heat gun on the low setting, the KAW is showing 844VA, 7.02A. The 6C is showing 6.98A at 121.2V, so 854.46VA, perfectly fine.
On high, the KAW is showing 1152VA, 9.72A, 118.4V. The 6C is showing 18.18A at 121.3V, so 2,205.2VA, waaay off.
I'm using 10285 for the gain_ct.

The larger clamps are still a decent amount off when the heat gun gets set to high, but not as bad.
An 80A clamp says:
Low: 6.92A
High: 12.46A
gain_ct: 41996

A 120A clamp says
Low: 6.99A
High: 12.55A
gain_ct: 26834

My full config is below. Please ignore the other gain_cts. I know they have strange values as I've just been playing around. To get these measurements, I'm using:
Chip 1
phase B for the 120A
phase C for the 20A
Chip 2
phase B for the 80A

`substitutions:
  device_name: house-electricity-meter
  pretty_device_name: "House Electricity Meter"
  ct1_name: "phase-1"
  ct2_name: "phase-2"
  ct3_name: "dish-washer"
  ct4_name: "dryer-phase-1"
  ct5_name: "dryer-phase-2"
  ct6_name: "sump-pump"
  ct1_current_gain: "27064"
  ct1_voltage_gain: "7305"
  ct2_current_gain: "26834"
  ct3_current_gain: "10285"
  ct4_current_gain: "41996"
  ct5_current_gain: "26767"
  ct6_current_gain: "11131"
  update_time: 20s

esphome:
  name: ${device_name}

esp32:
  board: nodemcu-32s

<<: !include common/common.yaml
<<: !include common/logger_debug.yaml

spi:
  clk_pin: 18
  miso_pin: 19
  mosi_pin: 23

sensor:
  # IC 1
  - platform: atm90e32
    cs_pin: 5
    phase_a:
      gain_voltage: ${ct1_voltage_gain}
      gain_ct: ${ct1_current_gain}
      voltage:
        name: "${ct1_name}-voltage"
        id: ic1Volts
        accuracy_decimals: 1
      current:
        name: "${ct1_name}-current"
        id: ct1Current
      power:
        name: "${ct1_name}-watts"
      reactive_power:
        name: "${ct1_name}-reactive-power"
      power_factor:
        name: "${ct1_name}-power-factor"
    phase_b:
      gain_voltage: ${ct1_voltage_gain}
      gain_ct: ${ct2_current_gain}
      current:
        name: "${ct2_name}-current"
        id: ct2Current
      power:
        name: "${ct2_name}-watts"
      reactive_power:
        name: "${ct2_name}-reactive-power"
      power_factor:
        name: "${ct2_name}-power-factor"
    phase_c:
      gain_voltage: ${ct1_voltage_gain}
      gain_ct: ${ct3_current_gain}
      current:
        name: "${ct3_name}-current"
      power:
        name: "${ct3_name}-watts"
    line_frequency: 60Hz
    gain_pga: 2X
    current_phases: 3
    update_interval: ${update_time}

  # IC 2
  - platform: atm90e32
    cs_pin: 4
    phase_a:
      gain_voltage: ${ct1_voltage_gain}
      gain_ct: ${ct4_current_gain}
      current:
        name: "${ct4_name}-current"
      power:
        name: "${ct4_name}-watts"
    phase_b:
      gain_voltage: ${ct1_voltage_gain}
      gain_ct: ${ct5_current_gain}
      current:
        name: "${ct5_name}-current"
      power:
        name: "${ct5_name}-watts"
    phase_c:
      gain_voltage: ${ct1_voltage_gain}
      gain_ct: ${ct6_current_gain}
      current:
        name: "${ct6_name}-current"
      power:
        name: "${ct6_name}-watts"
    line_frequency: 60Hz
    gain_pga: 2X
    current_phases: 3
    update_interval: ${update_time}
  # Total Current
  - platform: template
    name: house-total-current
    id: totalCurrent
    lambda: return id(ct1Current).state + id(ct2Current).state;
    accuracy_decimals: 1
    unit_of_measurement: A
    icon: "mdi:flash"
    update_interval: ${update_time}
  # Total Power
  - platform: template
    name: house-total-power
    id: totalPower
    lambda: return id(totalCurrent).state * id(ic1Volts).state;
    accuracy_decimals: 1
    unit_of_measurement: W
    icon: "mdi:flash-circle"
    update_interval: ${update_time}
  # Total kWh
  - platform: total_daily_energy
    name: 'house-daily-kwh'
    power_id: totalPower
    unit_of_measurement: 'kWh'
    state_class: total_increasing
    device_class: energy
    accuracy_decimals: 3
    filters:
      # W to kW
      - multiply: 0.001

# Enable time component to reset energy at midnight
time:
  - platform: homeassistant
    id: homeassistant_time
`

CircuitSetup, 2024-01-29:
Okay, thank you for explaining, and providing your config. 

Please try changing gain_pga: to 1x. 2x should only be used when the amperage output of the CT is too low, and needs more gain.

chksome, 2024-01-30:
Done. If it matters, I'll regularly be measuring total house current (chip 1 phase A and B for 240V split phase) in excess of 100A with two EVs. These new gain_cts are getting pretty high. From my reading of the docs, this may be a problem, but I'm not 100% sure. Based on the readings below, it seems to have made little to no difference. I only included data from these three clamps, one for each type, but each pair is reading really close to its mate. New code is pasted below:

Heat gun low, KAW 6.93A:
Chip 1, Phase B (120A clamp): 7.01A
Chip 1 Phase C (20A clamp): 6.44A
Chip 2 Phase B (80A clamp): 6.92A

Heat gun high, KAW 9.62A:
Chip 1, Phase B (120A clamp): 12.63A
Chip 1 Phase C (20A clamp): 17.98A
Chip 2 Phase B (80A clamp): 12.56A

```yaml
substitutions:
  device_name: house-electricity-meter
  pretty_device_name: "House Electricity Meter"
  ct1_name: "phase-1"
  ct2_name: "phase-2"
  ct3_name: "dish-washer"
  ct4_name: "dryer-phase-1"
  ct5_name: "dryer-phase-2"
  ct6_name: "sump-pump"
  ct1_current_gain: "54679" # 120A
  ct1_voltage_gain: "7305"  # 120A
  ct2_current_gain: "54679" # 120A
  ct3_current_gain: "20540" # 20A
  ct4_current_gain: "54399" # 80A
  ct5_current_gain: "54399" # 80A
  ct6_current_gain: "20540" # 20A
  update_time: 20s

esphome:
  name: ${device_name}

esp32:
  board: nodemcu-32s

<<: !include common/common.yaml
<<: !include common/logger_debug.yaml

spi:
  clk_pin: 18
  miso_pin: 19
  mosi_pin: 23

sensor:
  # IC 1
  - platform: atm90e32
    cs_pin: 5
    phase_a:
      gain_voltage: ${ct1_voltage_gain}
      gain_ct: ${ct1_current_gain}
      voltage:
        name: "${ct1_name}-voltage"
        id: ic1Volts
        accuracy_decimals: 1
      current:
        name: "${ct1_name}-current"
        id: ct1Current
      power:
        name: "${ct1_name}-watts"
      reactive_power:
        name: "${ct1_name}-reactive-power"
      power_factor:
        name: "${ct1_name}-power-factor"
    phase_b:
      gain_voltage: ${ct1_voltage_gain}
      gain_ct: ${ct2_current_gain}
      current:
        name: "${ct2_name}-current"
        id: ct2Current
      power:
        name: "${ct2_name}-watts"
      reactive_power:
        name: "${ct2_name}-reactive-power"
      power_factor:
        name: "${ct2_name}-power-factor"
    phase_c:
      gain_voltage: ${ct1_voltage_gain}
      gain_ct: ${ct3_current_gain}
      current:
        name: "${ct3_name}-current"
      power:
        name: "${ct3_name}-watts"
    line_frequency: 60Hz
    gain_pga: 1X
    current_phases: 3
    update_interval: ${update_time}

  # IC 2
  - platform: atm90e32
    cs_pin: 4
    phase_a:
      gain_voltage: ${ct1_voltage_gain}
      gain_ct: ${ct4_current_gain}
      current:
        name: "${ct4_name}-current"
      power:
        name: "${ct4_name}-watts"
    phase_b:
      gain_voltage: ${ct1_voltage_gain}
      gain_ct: ${ct5_current_gain}
      current:
        name: "${ct5_name}-current"
      power:
        name: "${ct5_name}-watts"
    phase_c:
      gain_voltage: ${ct1_voltage_gain}
      gain_ct: ${ct6_current_gain}
      current:
        name: "${ct6_name}-current"
      power:
        name: "${ct6_name}-watts"
    line_frequency: 60Hz
    gain_pga: 1X
    current_phases: 3
    update_interval: ${update_time}
  # Total Current
  - platform: template
    name: house-total-current
    id: totalCurrent
    lambda: return id(ct1Current).state + id(ct2Current).state;
    accuracy_decimals: 1
    unit_of_measurement: A
    icon: "mdi:flash"
    update_interval: ${update_time}
  # Total Power
  - platform: template
    name: house-total-power
    id: totalPower
    lambda: return id(totalCurrent).state * id(ic1Volts).state;
    accuracy_decimals: 1
    unit_of_measurement: W
    icon: "mdi:flash-circle"
    update_interval: ${update_time}
  # Total kWh
  - platform: total_daily_energy
    name: 'house-daily-kwh'
    power_id: totalPower
    unit_of_measurement: 'kWh'
    state_class: total_increasing
    device_class: energy
    accuracy_decimals: 3
    filters:
      # W to kW
      - multiply: 0.001

# Enable time component to reset energy at midnight
time:
  - platform: homeassistant
    id: homeassistant_time
```

chksome, 2024-01-30:
To eliminate something funny with the heat gun, here are stats with a space heater instead:
KAW: 9.84A
Chip 1, Phase B (120A clamp): 12.93A
Chip 1 Phase C (20A clamp): 18.36A
Chip 2 Phase B (80A clamp): 12.86A