CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter #182: Never ending calibration for 6channel power metter

gonzoyumo, 2024-09-02:
Hello, 

I finally started the setup of my power metter and I might do something wrong but I can't get it to report accurrate current when testing different loads.

I'm doing the calibration following https://github.com/CircuitSetup/Split-Single-Phase-Energy-Meter#calibration and watched the videos but anytime I'm calibrating my CTs correctly with a given load like a heat gun at low settings (7.6A) when I change that load to something bigger like the heat gun at high setting (10.8A) or the kettle (11.4A) then the measurements are off with what the kill a watt reports.

My setup is as follow:
- A power cord with 2 outlets, one is for the  kill A watt and the rest of the setup, the other for the AC adapter to the powe-metter.
- From the kill a watt, I have a custom made power cord with split conductors (romex 14awg) connected to an outlet on which I plug the heat gun and the kettle
- I'm only testing with 3 CTs for now, two 80A/26.6mA - YHDC SCT-010 (CT1 and CT2) and one 120A/40mA - YHDC SCT-016 (CT3) which are all on the load (black wire).
- The AC adapter I'm using is the Jameco 112336 (9v 1500ma). 

After calibrating for the Kettle:
- Kill-A-Watt: 111V, 11.4A
- Power metter:
  - Volts: 110.87V (gain_voltage: 7609)
  - CT1: 11.370A (gain_ct: 41886)
  - CT2: 11.372A (gain_ct: 42066)
  - CT3: 11.373A (gain_ct: 41711)

Then, switching to the heat gun low setting:

- Kill-A-Watt: 116.2V, 7.66A
- Power metter:
  - Volts: 114.84v (gain_voltage: 7609,)
  - CT1: 5.471A
  - CT2: 5.836A
  - CT3: 5.510A 

That's a 2A difference! And of course, if I recalibrate for the heat gun low setting, when I test with high setting or the kettle, then the measurements are off again :/ 

I've tested on IC2 (CT 4, CT5, and CT6) and also by removing the 2 add on boards I have, just in case, but I'm facing the same behavior.

Am I missing something here? Thank you for you assistance.

NB: I tried to open a ticket on https://circuitsetup.us/ but got an error " Block Reason: | A potentially unsafe operation has been detected in your request to this site"

CircuitSetup, 2024-09-09:
I'm sorry the meter is giving you trouble with calibration. Can you post your config file if using ESPHome? 

Also, what are you seeing for the power output in both scenarios?

CircuitSetup, 2024-09-10:
Thanks for supplying that data!

I asked about the power output, because I suspected the kill-a-watt wasn't taking power factor into account. I'm guessing the difference you're seeing in the amperage is because of the power factor. Although, for a resistive load, the PF should be at 1 or really close to 1. 

Are you in an apartment building or on a power system that is not split single phase?

CircuitSetup, 2024-09-10:
Okay, then your voltage and current should be in phase with each other. To be sure, can you try adding the power_factor parameter to your config? You may also want to take a look at reactive_power
See here for details on adding those parameters: https://esphome.io/components/sensor/atm90e32.html

gonzoyumo, 2024-09-10:
@CircuitSetup I think you've nailed it! I did not even know there was a PF indication on the Kill-A-Watt 🤦 

So the fun fact is that when usign the heat gun in `high` mode, PF is 1, but when using in `low` mode, PF drops to `0.68` on the Kill-A-Watt...

The power metter output was indicating a PF of `0.88`. See below:

[image]

I also tested with a toaster/grill which as a PF of 1 on the kill-a-watt and the power metter output was aligned with the kill-A-Watt on all measurements 👍. I wish I had tested that before :/ 

Now I have to further understand what this all mean when tracking energy usage of my house 😄 

Thanks a lot for your help 🙇

closed 2024-09-10 by gonzoyumo