CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter #184: Please help :/

turkoy, 2024-09-18:
I can not get any voltage, ampere, wattage value in any way, what could be the reason for this? I am using it with the following config.

AC POWER 12v 2.2A 
SCT013 Current meter

board 1.4rev1

[image]

`substitutions:
  disp_name: elk-cihaz-1
  update_time: 10s
  current_cal: '8650'
  voltage_cal: '29462'
  
esphome:
  name: elk-cihaz-1

esp32:
  board: nodemcu-32s
  framework:
    type: arduino

spi:
  clk_pin: 18
  miso_pin: 19
  mosi_pin: 23

sensor:
  - platform: wifi_signal
    name: ${disp_name} WiFi
    update_interval: 60s
#IC1
  - platform: atm90e32
    cs_pin: 5
    phase_a:
      voltage:
        name: ${disp_name} Volts A
        id: ic1Volts
        accuracy_decimals: 1
      current:
        name: ${disp_name} CT1 Amps
        id: ct1Amps
      power:
        name: ${disp_name} CT1 Watts
        id: ct1Watts
      gain_voltage: ${voltage_cal}
      gain_ct: ${current_cal}
    phase_b:
      current:
        name: ${disp_name} CT2 Amps
        id: ct2Amps
      power:
        name: ${disp_name} CT2 Watts
        id: ct2Watts
      gain_voltage: ${voltage_cal}
      gain_ct: ${current_cal}
    phase_c:
      current:
        name: ${disp_name} CT3 Amps
        id: ct3Amps
      power:
        name: ${disp_name} CT3 Watts
        id: ct3Watts
      gain_voltage: ${voltage_cal}
      gain_ct: ${current_cal}
    frequency:
      name: ${disp_name} Freq A
    line_frequency: 60Hz
    gain_pga: 1X
    update_interval: ${update_time}
#IC2
  - platform: atm90e32
    cs_pin: 4
    phase_a:
      voltage:
        name: ${disp_name} Volts B
        id: ic2Volts
        accuracy_decimals: 1
      current:
        name: ${disp_name} CT4 Amps
        id: ct4Amps
      power:
        name: ${disp_name} CT4 Watts
        id: ct4Watts
      gain_voltage: ${voltage_cal}
      gain_ct: ${current_cal}
    phase_b:
      current:
        name: ${disp_name} CT5 Amps
        id: ct5Amps
      power:
        name: ${disp_name} CT5 Watts
        id: ct5Watts
      gain_voltage: ${voltage_cal}
      gain_ct: ${current_cal}
    phase_c:
      current:
        name: ${disp_name} CT6 Amps
        id: ct6Amps
      power:
        name: ${disp_name} CT6 Watts
        id: ct6Watts
      gain_voltage: ${voltage_cal}
      gain_ct: ${current_cal}
    frequency:
      name: ${disp_name} Freq B
    line_frequency: 60Hz
    gain_pga: 1X
    update_interval: ${update_time}
        
#Total Amps   
  - platform: template
    name: ${disp_name} Total Amps
    id: totalAmps
    lambda: return id(ct1Amps).state + id(ct2Amps).state + id(ct3Amps).state + id(ct4Amps).state + id(ct5Amps).state + id(ct6Amps).state ;
    accuracy_decimals: 2
    unit_of_measurement: A
    device_class: current
    update_interval: ${update_time}         
#Total Watts
  - platform: template
    name: ${disp_name} Total Watts
    id: totalWatts
    lambda: return id(ct1Watts).state + id(ct2Watts).state + id(ct3Watts).state + id(ct4Watts).state + id(ct5Watts).state + id(ct6Watts).state ;
    accuracy_decimals: 1
    unit_of_measurement: W
    device_class: power
    update_interval: ${update_time}
#kWh
  - platform: total_daily_energy
    name: ${disp_name} Total kWh
    power_id: totalWatts
    filters:
      - multiply: 0.001
    unit_of_measurement: kWh
    device_class: energy
    state_class: total_increasing
switch:
  - platform: restart
    name: ${disp_name} Restart  
time:
  - platform: sntp
    id: sntp_time   `

turkoy, 2024-09-18:
frequency values are constantly moving between 30 and 130 ? and it shows double frequency even though there is only 1 ac power connected.

zdelesde, 2024-09-27:
Perhaps you're using a DC power supply instead of an AC one?

CircuitSetup, 2024-09-27:
> frequency values are constantly moving between 30 and 130 ? and it shows double frequency even though there is only 1 ac power connected.

It sounds like you're not using an AC transformer, but an AC power supply. It has to be a transformer.

turkoy, 2024-09-27:
Thank you very much for your answers.

Yes, it is not written on the adapter, but it may be an AC to DC adapter, I cannot measure the output, I will check it or I will buy a 220V AC to 12V or 9V AC adapter. 

Can we say that the problem is completely caused by this? If I am testing with a dc input adapter.