CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter #186: Negative power reading

icvdok, 2024-11-16:
Hello, I've just setup the board with ESPHome fw. integrated with HA.

The setup is OK, all fine except that in one phase the reading is negative.

[image]

I'm using the 20A:25mA clamp

[image]

What could be wrong with my setup?

```
# CircuitSetup 6 Channel Energy Meter Main Board example config

substitutions:
# Name the device and it's entities
  devicename: hapw06
  friendlyname: hapw06
# Interval of how often the power is updated
  update_time: 15s
# Current Transformers:
#  20A/25mA SCT-006: 11143
#  30A/1V SCT-013-030: 8650
#  50A/1V SCT-013-050: 15420
#  50A/16.6mA SCT-010: 41334
#  80A/26.6mA SCT-010: 41660
#  100A/50ma SCT-013-000: 27518
#  120A/40mA: SCT-016: 41787
#  200A/100mA SCT-024: 27518
#  200A/50mA SCT-024: 55036
  current_cal: '11143'
# Jameco 9VAC Transformer: 
#  For meter versions: 
#  >= v1.3: 7305
  voltage_cal: '14007'

esphome:
  name: ${devicename}
  friendly_name: ${friendlyname}
  project:
    name: circuitsetup.6c-energy-meter-main
    version: "1.4.1"

esp32:
#board: nodemcu-32s
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: !secret api_key

ota:
  - platform: esphome
    password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid_iot
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Hapw05 Fallback Hotspot"
    password: !secret fallback_ap_psw

captive_portal:

web_server:
  port: 80

spi:
  clk_pin: 18
  miso_pin: 19
  mosi_pin: 23

sensor:
  - platform: wifi_signal
    name: ${devicename} WiFi
    update_interval: 60s
#IC1
  - platform: atm90e32
    cs_pin: 5
    phase_a:
      voltage:
        name: ${devicename} Volts A
        id: ic1Volts
        accuracy_decimals: 1
      current:
        name: ${devicename} CT1 Amps
        id: ct1Amps
# The max value for current that the meter can output is 65.535. If you expect to measure current over 65A, 
# divide the gain_ct by 2 (120A CT) or 4 (200A CT) and multiply the current and power values by 2 or 4 by uncommenting the filter below
#        filters:
#          - multiply: 2
      power:
        name: ${devicename} CT1 Watts
        id: ct1Watts
#        filters:
#          - multiply: 2
      gain_voltage: ${voltage_cal}
      gain_ct: ${current_cal}
    phase_b:
      current:
        name: ${devicename} CT2 Amps
        id: ct2Amps
      power:
        name: ${devicename} CT2 Watts
        id: ct2Watts
      gain_voltage: ${voltage_cal}
      gain_ct: ${current_cal}
    phase_c:
      current:
        name: ${devicename} CT3 Amps
        id: ct3Amps
      power:
        name: ${devicename} CT3 Watts
        id: ct3Watts
      gain_voltage: ${voltage_cal}
      gain_ct: ${current_cal}
    frequency:
      name: ${devicename} Freq A
    line_frequency: 60Hz
    gain_pga: 1X
    update_interval: ${update_time}
#IC2
  - platform: atm90e32
    cs_pin: 4
    phase_a:
#this voltage is only needed if monitoring 2 voltages
      voltage:
        name: ${devicename} Volts B
        id: ic2Volts
        accuracy_decimals: 1
      current:
        name: ${devicename} CT4 Amps
        id: ct4Amps
      power:
        name: ${devicename} CT4 Watts
        id: ct4Watts
      gain_voltage: ${voltage_cal}
      gain_ct: ${current_cal}
    phase_b:
      current:
        name: ${devicename} CT5 Amps
        id: ct5Amps
      power:
        name: ${devicename} CT5 Watts
        id: ct5Watts
      gain_voltage: ${voltage_cal}
      gain_ct: ${current_cal}
    phase_c:
      current:
        name: ${devicename} CT6 Amps
        id: ct6Amps
      power:
        name: ${devicename} CT6 Watts
        id: ct6Watts
      gain_voltage: ${voltage_cal}
      gain_ct: ${current_cal}
    frequency:
      name: ${devicename} Freq B
    line_frequency: 60Hz
    gain_pga: 1X
    update_interval: ${update_time}
        
# #Total Amps   
#   - platform: template
#     name: ${devicename} Total Amps
#     id: totalAmps
#     lambda: return id(ct1Amps).state + id(ct2Amps).state + id(ct3Amps).state + id(ct4Amps).state + id(ct5Amps).state + id(ct6Amps).state ;
#     accuracy_decimals: 2
#     unit_of_measurement: A
#     device_class: current
#     update_interval: ${update_time}         
# #Total Watts
#   - platform: template
#     name: ${devicename} Total Watts
#     id: totalWatts
#     lambda: return id(ct1Watts).state + id(ct2Watts).state + id(ct3Watts).state + id(ct4Watts).state + id(ct5Watts).state + id(ct6Watts).state ;
#     accuracy_decimals: 1
#     unit_of_measurement: W
#     device_class: power
#     update_interval: ${update_time}
# #kWh
#   - platform: total_daily_energy
#     name: ${devicename} Total kWh
#     power_id: totalWatts
#     filters:
#       - multiply: 0.001
#     unit_of_measurement: kWh
#     device_class: energy
#     state_class: total_increasing

#Total Amps   
  - platform: template
    name: ${devicename} Total Amps A
    id: totalAmpsA
    lambda: return id(ct1Amps).state + id(ct2Amps).state + id(ct3Amps).state ;
    accuracy_decimals: 2
    unit_of_measurement: A
    device_class: current
    update_interval: ${update_time}         
#Total Watts
  - platform: template
    name: ${devicename} Total Watts A
    id: totalWattsA
    lambda: return id(ct1Watts).state + id(ct2Watts).state + id(ct3Watts).state ;
    accuracy_decimals: 1
    unit_of_measurement: W
    device_class: power
    update_interval: ${update_time}
#kWh
  - platform: total_daily_energy
    name: ${devicename} Total kWh A
    power_id: totalWattsA
    filters:
      - multiply: 0.001
    unit_of_measurement: kWh
    device_class: energy
    state_class: total_increasing

#Total Amps   
  - platform: template
    name: ${devicename} Total Amps B
    id: totalAmpsB
    lambda: return id(ct4Amps).state + id(ct5Amps).state + id(ct6Amps).state ;
    accuracy_decimals: 2
    unit_of_measurement: A
    device_class: current
    update_interval: ${update_time}         
#Total Watts
  - platform: template
    name: ${devicename} Total Watts B
    id: totalWattsB
    lambda: return id(ct4Watts).state + id(ct5Watts).state + id(ct6Watts).state ;
    accuracy_decimals: 1
    unit_of_measurement: W
    device_class: power
    update_interval: ${update_time}
#kWh
  - platform: total_daily_energy
    name: ${devicename} Total kWh B
    power_id: totalWattsB
    filters:
      - multiply: 0.001
    unit_of_measurement: kWh
    device_class: energy
    state_class: total_increasing

switch:
  - platform: restart
    name: ${devicename} Restart  
time:
  - platform: sntp
    id: sntp_time
```

Sandriell, 2024-11-16:
The CT Clamp is on backwards.

icvdok, 2024-11-17:
Thanks Sandriell,  I'll check the orientation again.

icvdok, 2024-11-17:
Looks better now, many thanks, I'll check if it's ok in the next days. Last question, there is a way to reset the KWh counter?

closed 2024-11-21 by icvdok