CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter #189: CT1-3 not reporting with warning about atm90e32.sensor operation time

iamwyza, 2024-12-21:
Recently started tinkering with my newly arrived board.  It's a v1.4 rev 1.

I have a [Aokin ESP32 ESP-32S NodeMCU-32s](https://www.amazon.com/dp/B08NW4JXFM) board hooked up and flashed with ESPHome using the sample from this repo for the main board with no alterations other than the wifi config and setting the current_cal to the be the appropriate one for the CTs (SCT-010).

The board boots fine and reports CT4-6, but does not report CT1-3.  It has these warnings in it:

```
Component atm90e32.sensor took a long time for an operation (78 ms).
Components should block for at most 30 ms.
```

Since only half of these are reporting 

For reference here's the yaml and logs:

```yaml
# CircuitSetup 6 Channel Energy Meter Main Board example config

substitutions:
# Change the disp_name to something you want  
  disp_name: Energy Meter
  friendly_name: CircuitSetup Energy Meter
# Interval of how often the power is updated
  update_time: 10s
# Current Transformers:
#  20A/25mA SCT-006: 11143
#  30A/1V SCT-013-030: 8650
#  50A/1V SCT-013-050: 15420
#  50A/16.6mA SCT-010: 41334
#  80A/26.6mA SCT-010: 41660
#  100A/50ma SCT-013-000: 27518
#  120A/40mA: SCT-016: 41787
#  200A/100mA SCT-024: 27518
#  200A/50mA SCT-024: 55036
  current_cal: '41660'
# Jameco 9VAC Transformer: 
#  For meter versions: 
#  >= v1.3: 7305
  voltage_cal: '7305'

esphome:
  name: energy-meter
  friendly_name: "${friendly_name}"
  project:
    name: circuitsetup.6c-energy-meter-main
    version: "1.4.1"

esp32:
  board: nodemcu-32s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

captive_portal:

esp32_improv:
# Sets up Bluetooth LE to allow the user
# to provision wifi credentials to the device.
  authorizer: none

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  - platform: esphome

[... 172 lines cut ...]
[15:22:28][D][sensor:094]: 'Energy Meter CT6 Watts': Sending state 0.16288 W with 2 decimals of accuracy
[15:22:28][W][component:237]: Component atm90e32.sensor took a long time for an operation (78 ms).
[15:22:35][D][sensor:094]: 'Energy Meter Total Watts': Sending state nan W with 1 decimals of accuracy
[15:22:37][D][sensor:094]: 'Energy Meter Volts B': Sending state 117.30000 V with 1 decimals of accuracy
[15:22:37][D][sensor:094]: 'Energy Meter CT4 Amps': Sending state 0.01100 A with 2 decimals of accuracy
[15:22:37][D][sensor:094]: 'Energy Meter CT5 Amps': Sending state 0.01000 A with 2 decimals of accuracy
[15:22:37][D][sensor:094]: 'Energy Meter CT6 Amps': Sending state 0.00700 A with 2 decimals of accuracy
[15:22:37][D][sensor:094]: 'Energy Meter CT4 Watts': Sending state -0.12096 W with 2 decimals of accuracy
[15:22:37][D][sensor:094]: 'Energy Meter CT5 Watts': Sending state -0.19232 W with 2 decimals of accuracy
[15:22:39][D][sensor:094]: 'Energy Meter CT6 Watts': Sending state -0.01152 W with 2 decimals of accuracy
[15:22:39][D][sensor:094]: 'Energy Meter Freq B': Sending state 60.00000 Hz with 1 decimals of accuracy
[15:22:39][D][sensor:094]: 'Energy Meter Total Amps': Sending state nan A with 2 decimals of accuracy
[15:22:46][D][sensor:094]: 'Energy Meter Total Watts': Sending state nan W with 1 decimals of accuracy
[15:22:47][D][sensor:094]: 'Energy Meter Volts B': Sending state 117.01000 V with 1 decimals of accuracy
[15:22:47][D][sensor:094]: 'Energy Meter CT4 Amps': Sending state 0.01400 A with 2 decimals of accuracy
[15:22:47][D][sensor:094]: 'Energy Meter CT5 Amps': Sending state 0.01400 A with 2 decimals of accuracy
[15:22:47][D][sensor:094]: 'Energy Meter CT6 Amps': Sending state 0.01000 A with 2 decimals of accuracy
[15:22:47][D][sensor:094]: 'Energy Meter CT4 Watts': Sending state 0.16480 W with 2 decimals of accuracy
[15:22:47][D][sensor:094]: 'Energy Meter CT5 Watts': Sending state 0.08544 W with 2 decimals of accuracy
[15:22:47][D][sensor:094]: 'Energy Meter CT6 Watts': Sending state 0.00128 W with 2 decimals of accuracy
[15:22:47][D][sensor:094]: 'Energy Meter Freq B': Sending state 60.00000 Hz with 1 decimals of accuracy
[15:22:47][W][component:237]: Component atm90e32.sensor took a long time for an operation (75 ms).
[15:22:47][W][component:238]: Components should block for at most 30 ms.
[15:22:48][D][sensor:094]: 'Energy Meter Total Amps': Sending state nan A with 2 decimals of accuracy
```

iamwyza, 2025-01-04:
Update, I've also flashed a LILYGO T-ETH-Lite-ESP32-S3 and attempted the same thing and am getting the same behavior.  CT4-6 work, CT1-3 don't.

SzymonSlupik, 2025-01-05:
> WARNING GPIO5 is a strapping PIN and should only be used for I/O with care.

Have you tried configuring the first chip to use a different CS line?

iamwyza, 2025-01-05:
Can I specify any unused GPIO pin as a cs_pin? (keeping in mind that I'm using the configs directly from this repo and on the board directly from the store, so if cs_pin 5 is not advised, then the repo should not specify that as the pin to use).  

That said, the 2nd board (the Lilygo one) doesn't have that warning.  Here's it's config + logs:

```yaml
# CircuitSetup 6 Channel Energy Meter Main Board example config

substitutions:
# Change the disp_name to something you want  
  disp_name: Energy Meter
  friendly_name: CircuitSetup Energy Meter
# Interval of how often the power is updated
  update_time: 10s
# Current Transformers:
#  20A/25mA SCT-006: 11143
#  30A/1V SCT-013-030: 8650
#  50A/1V SCT-013-050: 15420
#  50A/16.6mA SCT-010: 41334
#  80A/26.6mA SCT-010: 41660
#  100A/50ma SCT-013-000: 27518
#  120A/40mA: SCT-016: 41787
#  200A/100mA SCT-024: 27518
#  200A/50mA SCT-024: 55036
  current_cal: '41660'
# Jameco 9VAC Transformer: 
#  For meter versions: 
#  >= v1.3: 7305
  voltage_cal: '7305'

esphome:
  name: energy-meter
  friendly_name: "${friendly_name}"
  project:
    name: circuitsetup.6c-energy-meter-main
    version: "1.4.1"

esp32:
  board: esp32s3box
  framework:
    type: arduino

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

captive_portal:

esp32_improv:
# Sets up Bluetooth LE to allow the user
# to provision wifi credentials to the device.
  authorizer: none

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  - platform: esphome

dashboard_import:
  package_import_url: github://circuitsetup/Expandable-6-Channel-ESP32-Energy-Meter/Software/ESPHome/6chan_energy_meter_main_board.yaml@master
  import_full_config: true

web_server:
  port: 80

spi:
  clk_pin: 7
[... 309 lines cut ...]
[10:44:32][D][sensor:094]: 'Energy Meter CT5 Watts': Sending state -0.29984 W with 2 decimals of accuracy
[10:44:32][D][sensor:094]: 'Energy Meter CT6 Watts': Sending state 0.12256 W with 2 decimals of accuracy
[10:44:32][D][sensor:094]: 'Energy Meter Freq B': Sending state 59.99000 Hz with 1 decimals of accuracy
[10:44:40][D][sensor:094]: 'Energy Meter Total Watts': Sending state nan W with 1 decimals of accuracy
[10:44:41][D][sensor:094]: 'Energy Meter Total Amps': Sending state nan A with 2 decimals of accuracy
[10:44:42][D][sensor:094]: 'Energy Meter Volts B': Sending state 114.19000 V with 1 decimals of accuracy
[10:44:42][D][sensor:094]: 'Energy Meter CT4 Amps': Sending state 0.01200 A with 2 decimals of accuracy
[10:44:42][D][sensor:094]: 'Energy Meter CT5 Amps': Sending state 0.01700 A with 2 decimals of accuracy
[10:44:42][D][sensor:094]: 'Energy Meter CT6 Amps': Sending state 0.02000 A with 2 decimals of accuracy
[10:44:42][D][sensor:094]: 'Energy Meter CT4 Watts': Sending state -0.01152 W with 2 decimals of accuracy
[10:44:42][D][sensor:094]: 'Energy Meter CT5 Watts': Sending state 0.12864 W with 2 decimals of accuracy
[10:44:42][D][sensor:094]: 'Energy Meter CT6 Watts': Sending state 0.41312 W with 2 decimals of accuracy
[10:44:42][D][sensor:094]: 'Energy Meter Freq B': Sending state 60.00000 Hz with 1 decimals of accuracy
[10:44:50][D][sensor:094]: 'Energy Meter Total Watts': Sending state nan W with 1 decimals of accuracy
[10:44:51][D][sensor:094]: 'Energy Meter Total Amps': Sending state nan A with 2 decimals of accuracy
[10:44:52][D][sensor:094]: 'Energy Meter Volts B': Sending state 114.24000 V with 1 decimals of accuracy
[10:44:52][D][sensor:094]: 'Energy Meter CT4 Amps': Sending state 0.01200 A with 2 decimals of accuracy
[10:44:52][D][sensor:094]: 'Energy Meter CT5 Amps': Sending state 0.01600 A with 2 decimals of accuracy
[10:44:52][D][sensor:094]: 'Energy Meter CT6 Amps': Sending state 0.01600 A with 2 decimals of accuracy
[10:44:52][D][sensor:094]: 'Energy Meter CT4 Watts': Sending state -0.12512 W with 2 decimals of accuracy
[10:44:52][D][sensor:094]: 'Energy Meter CT5 Watts': Sending state 0.24096 W with 2 decimals of accuracy
[10:44:52][D][sensor:094]: 'Energy Meter CT6 Watts': Sending state 0.03136 W with 2 decimals of accuracy
[10:44:52][D][sensor:094]: 'Energy Meter Freq B': Sending state 60.00000 Hz with 1 decimals of accuracy
[10:44:54][D][sensor:094]: 'Energy Meter WiFi': Sending state -53.00000 dBm with 0 decimals of accuracy
```

SzymonSlupik, 2025-01-05:
You select the CS for each of the 2 ICs with a jumper. The jumper selection must match what you have in the YAML. Just try a different one and make sure it matches your jumpers.

iamwyza, 2025-01-05:
There are no jumpers on the main board.  The jumpers only exist on the addon boards as far as I can tell.  
[image]

SzymonSlupik, 2025-01-05:
True... Then I still believe you have an issue with the GPIO 5 on your ESP32 variant. But my knowledge of ESP32 is too shallow to help you further.

iamwyza, 2025-01-05:
I'd agree, but given that both ESP32's (with from different mfg with different firmwares) have the same issue, i'm leaning towards it being the main board.  I suppose I could breadboard the first one so that I could move the pin that way.  But 2 different boards with the same issue?  That seems less likely.  I'm hoping @CircuitSetup will see this and respond with some further diagnostics I can take.

CircuitSetup, 2025-01-05:
The `sensor took a long time for an operation` warning is normal because of the speed of the SPI bus, and won't block actually getting data.

I don't think it would matter in this case, since you're getting connected with GPIO 4, with the Lilygo T-eth board, but take a look at the ethernet example, it uses software SPI, and a compile with the esp-idf. Try that just to make sure there isn't something limiting on the hardware SPI bus.

You very well may have a defective meter. If that's the case I'll get you a replacement.

iamwyza, 2025-01-05:
Sadly no dice.  Used the given example:

```yaml

esp32:
  board: esp32s3box
  variant: esp32s3
  framework:
    type: esp-idf

#.... cut.....

spi:
  interface: software
  clk_pin: 7
  miso_pin: 5
  mosi_pin: 6

sensor:
  - platform: wifi_signal
    name: ${disp_name} WiFi
    update_interval: 60s
#IC1
  - platform: atm90e32
    cs_pin: 8
    phase_a:
      voltage:

```

```logs
[16:22:10][E][component:082]:   Component atm90e32.sensor is marked FAILED
```

TheSp1der, 2025-03-13:
> The `sensor took a long time for an operation` warning is normal because of the speed of the SPI bus, and won't block actually getting data.

Thank you for this comment, it was driving me crazy. I love my meter and just having you confirm that this is nothing to be concerned about has helped a lot.

HelmutLinner, 2025-04-24:
I am having the same warnings. Is there a way to change the time before i get an warning? So lets say i get the warning on average between 55 and 80 ms, can i change the warning to 90ms ? 
`[21:13:26][W][component:239]: Component atm90e32.sensor took a long time for an operation (75 ms).
[21:13:26][W][component:240]: Components should block for at most 30 ms.
[21:13:38][D][sensor:094]: 'Stromzaehler2 LiLaMat Schreinerei Gesamt': Sending state 693.63074 W with 1 decimals of accuracy
[21:13:38][D][sensor:094]: 'Spannung L3': Sending state 227.91000 V with 1 decimals of accuracy
[21:13:38][D][sensor:094]: 'Zuleitung Betrieb L3 CT13 Amps': Sending state 6.80200 A with 2 decimals of accuracy
[21:13:38][D][sensor:094]: 'Zuleitung Betrieb L3 CT13 Watts': Sending state 1200.44824 W with 2 decimals of accuracy
[21:13:38][D][sensor:094]: 'LiLaMat Schreinerei L3 CT14 Watts': Sending state 6.01280 W with 2 decimals of accuracy
[21:13:39][D][sensor:094]: 'Betriebswohnung L3 CT15 Watts': Sending state 0.97536 W with 2 decimals of accuracy
[21:13:39][W][component:239]: Component atm90e32.sensor took a long time for an operation (62 ms).
[21:13:39][W][component:240]: Components should block for at most 30 ms.
[21:13:39][D][sensor:094]: 'Spannung L2': Sending state 227.53000 V with 1 decimals of accuracy
[21:13:39][D][sensor:094]: 'Zuleitung Betrieb L2 CT7 Amps': Sending state 6.64400 A with 2 decimals of accuracy
[21:13:40][D][sensor:094]: 'Zuleitung Betrieb L2 CT7 Watts': Sending state 1413.94434 W with 2 decimals of accuracy
[21:13:40][D][sensor:094]: 'LiLaMat Schreinerei L2 CT8 Watts': Sending state 655.88800 W with 2 decimals of accuracy
[21:13:40][D][sensor:094]: 'Betriebswohnung L2 CT9 Watts': Sending state -0.69440 W with 2 decimals of accuracy
[21:13:40][W][component:239]: Component atm90e32.sensor took a long time for an operation (61 ms).
[21:13:40][W][component:240]: Components should block for at most 30 ms.`

iamwyza, 2025-07-04:
Sorry for the really late reply to this.  They did send me a replacement board back in January, but I just haven't had time to actually test it in the interim.  The new board fixed the issue for me.  So I'm going to close this ticket.

closed 2025-07-04 by iamwyza