CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter #195: Negative values

ryanbuiltthat, 2025-04-07:
I currently only have two CT sensors on the board. One around each phase coming from the outside meter. I'm getting negative numbers for Harmonic power, Phase angle and peak current. Which direction should I have the CT sensors pointing?

CS Energy Meter CT1 Harmonic Power | -3.84 W |  
CS Energy Meter CT1 Peak Current | 11.82 A |  
CS Energy Meter CT2 Harmonic Power | -1.03 W |  
CS Energy Meter CT2 Peak Current | 6.54 A |  
CS Energy Meter L1 Phase Angle | 1.00 ° |  
CS Energy Meter L2 Phase Angle | -20.00 ° |  
CS Energy Meter Power Factor A | 0.90 |  
CS Energy Meter Restart | OFF | Off   On
CS Energy Meter WiFi | -58 dBm |  
CS Energy Meter ct1 FAWattHours | 6919.90 Wh |  
CS Energy Meter ct2 FAWattHours | 3100.69 Wh |  

EMON Active Power CT2 | 414.00 W |  
EMON CT1 Current | 3.89 A |  
EMON CT2 Current | 3.78 A |  
EMON Line Frequency | 60.0 Hz |  
EMON Line Voltage A | 117.12 V |  
EMON Power CT1 | 409.42 W

CircuitSetup, 2025-04-07:
What does tou config file look like?

ryanbuiltthat, 2025-04-07:
```
substitutions:
  disp_name: CS Energy Meter
  friendly_name: CircuitSetup Energy Meter
  update_time: 10s

  current_cal: '27518'
  voltage_cal: '7305'

packages:
  common: github://CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter/Software/ESPHome/6chan_common.yaml  

esphome:
  name: cs-pm
  friendly_name: cs-pm

esp32:
  board: nodemcu-32s
  framework:
    type: arduino

logger:

api:
  encryption:
    key: "***********************************"

ota:
  - platform: esphome
    password: "**************************"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  ap:
    ssid: "Cs-Pm Fallback Hotspot"
    password: "****************"

captive_portal:

spi:
  clk_pin: 18
  miso_pin: 19
  mosi_pin: 23
sensor:
  - platform: atm90e32
    cs_pin: 5
    id: chip1 #Optional    
    phase_a:
      forward_active_energy:
         name: ${disp_name} ct1 FAWattHours
         id: ct1FAWattHours

      voltage:
        name: "EMON Line Voltage A"
      current:
        name: "EMON CT1 Current"
      power:
        name: "EMON Power CT1"
      harmonic_power:
        name: ${disp_name} CT1 Harmonic Power
      gain_voltage: 7305
      gain_ct: ${current_cal}
      power_factor: 
        name: ${disp_name} Power Factor A
    phase_b:
      forward_active_energy:
         name: ${disp_name} ct2 FAWattHours
         id: ct2FAWattHours
      current:
        name: "EMON CT2 Current"
      power:
        name: "EMON Active Power CT2"
      gain_voltage: 7305
      gain_ct: ${current_cal}
      peak_current:
        name: ${disp_name} CT2 Peak Current
      harmonic_power:
        name: ${disp_name} CT2 Harmonic Power
      phase_angle:
        name: ${disp_name} L2 Phase Angle
    frequency:
      name: "EMON Line Frequency"
    line_frequency: 60Hz
    current_phases: 2
    gain_pga: 1X
    update_interval: 60s
    enable_offset_calibration: True
    peak_current_signed: True    
 

button:
  - platform: atm90e32
    id: chip1
    run_offset_calibration:
      name: "Chip1 - Run Offset Calibration"
    clear_offset_calibration:
      name: "Chip1 - Clear Offset Calibration"
```

CircuitSetup, 2025-04-07:
Thanks! Start by taking out current_phases: 2. This actually turns off proper calculations for Phase b.

Assuming you're measuring 1 voltage and on a split single phase system, the CTs should be in opposite directions from eachother.