CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter #198: Trying to get to root cause of inability to reliably update esphome on 6 channel meter.

PedanticAvenger, 2025-05-01:
So I've been trying to get to the root of this and suspect there may be some oddity in the hardware of the stack.

OTA upgrades work MAYBE 1 in 10 times, the typical failure is "ERROR Error receiving acknowledge chunk OK: [Errno 104] Connection reset by peer"
Wifi strength seems to be fine (-61dBM when I can get to the web page for the meter).  I noticed that it regularly disappears from Home Assistant so think it's reloading repeatedly.

Now, I got a couple more Node-mcu32S modules from different vendors and they all display the same behavior which is what leads me to wondering if something is up with the meter boards (one base and one expansion).  I cannot connect to the webpage to watch the OTA process, it always hangs.

Anyone have further troubleshooting ideas or steps I should go through to verify that I haven't done anything inappropriate with the hardware?

PedanticAvenger, 2025-05-01:
Further esp web page details.
When I CAN connect to it, there is almost NEVER a full sensor display, and when there is the logs don't stream either.

CircuitSetup, 2025-05-02:
I've seen this happen before when ESPHome has trouble connecting to Home Assistant. The signal strength may read high, but there could be a lot of interference/dropped packets. Are you able to change the channel on your router that it is using for 2.4GHz? Do you have an app that can analyze how many other routers around you are on a particular channel?

Also, is there an API key in your ESPHome config?

PedanticAvenger, 2025-05-02:
Yeah, so the AP thinks signal is lower than the mcu does.
Not low enough that I would be concerned in general but perhaps?
Or something is making the mcu unhappy and making it reboot a lot.

```
House Power Monitor connected to PedDev on AP-House. Connection Info: Ch. 11 (2.4 GHz, 20 MHz), -70 dBm, 192.168.55.12. | Today at 13:37
House Power Monitor disconnected from PedDev. Time Connected: 15s. Data Used: 5.72 KB (up) / 530.00 B (down). Last Connected To: AP-House at -68 dBm. | Today at 13:37
House Power Monitor connected to PedDev on AP-House. Connection Info: Ch. 11 (2.4 GHz, 20 MHz), -68 dBm, 192.168.55.12. | Today at 13:37
House Power Monitor disconnected from PedDev. Time Connected: 2m 34s. Data Used: 203.61 KB (up) / 42.70 KB (down). Last Connected To: AP-House at -69 dBm. | Today at 13:37
House Power Monitor connected to PedDev on AP-House. Connection Info: Ch. 11 (2.4 GHz, 20 MHz), -70 dBm, 192.168.55.12. | Today at 13:34
House Power Monitor disconnected from PedDev. Time Connected: 47s. Data Used: 25.92 KB (up) / 6.00 KB (down). Last Connected To: AP-House at -69 dBm. | Today at 13:34
House Power Monitor connected to PedDev on AP-House. Connection Info: Ch. 11 (2.4 GHz, 20 MHz), -70 dBm, 192.168.55.12. | Today at 13:33
House Power Monitor disconnected from PedDev. Time Connected: 22s. Data Used: 17.79 KB (up) / 4.29 KB (down). Last Connected To: AP-House at -70 dBm. | Today at 13:33
House Power Monitor connected to PedDev on AP-House. Connection Info: Ch. 11 (2.4 GHz, 20 MHz), -69 dBm, 192.168.55.12. | Today at 13:33
House Power Monitor disconnected from PedDev. Time Connected: 1m 10s. Data Used: 39.96 KB (up) / 11.75 KB (down). Last Connected To: AP-House at -68 dBm. | Today at 13:33
House Power Monitor connected to PedDev on AP-House. Connection Info: Ch. 11 (2.4 GHz, 20 MHz), -70 dBm, 192.168.55.12. | Today at 13:32
House Power Monitor disconnected from PedDev. Time Connected: 1m 1s. Data Used: 50.44 KB (up) / 14.15 KB (down). Last Connected To: AP-House at -68 dBm. | Today at 13:32
House Power Monitor connected to PedDev on AP-House. Connection Info: Ch. 11 (2.4 GHz, 20 MHz), -71 dBm, 192.168.55.12. | Today at 13:31
House Power Monitor disconnected from PedDev. Time Connected: 3s. Data Used: 0.00 B (up) / 0.00 B (down). Last Connected To: AP-House at -69 dBm. | Today at 13:31
House Power Monitor connected to PedDev on AP-House. Connection Info: Ch. 11 (2.4 GHz, 20 MHz), -69 dBm, 192.168.55.12. | Today at 13:31
House Power Monitor disconnected from PedDev. Time Connected: 1m 1s. Data Used: 70.03 KB (up) / 14.72 KB (down). Last Connected To: AP-House at -69 dBm. | Today at 13:31
House Power Monitor connected to PedDev on AP-House. Connection Info: Ch. 11 (2.4 GHz, 20 MHz), -69 dBm, 192.168.55.12. | Today at 13:29
House Power Monitor disconnected from PedDev. Time Connected: 18s. Data Used: 11.68 KB (up) / 2.58 KB (down). Last Connected To: AP-House at -71 dBm. | Today at 13:29
House Power Monitor connected to PedDev on AP-House. Connection Info: Ch. 11 (2.4 GHz, 20 MHz), -70 dBm, 192.168.55.12. | Today at 13:29
House Power Monitor disconnected from PedDev. Time Connected: 1m 16s. Data Used: 74.39 KB (up) / 21.82 KB (down). Last Connected To: AP-House at -69 dBm. | Today at 13:29
House Power Monitor connected to PedDev on AP-House. Connection Info: Ch. 11 (2.4 GHz, 20 MHz), -70 dBm, 192.168.55.12. | Today at 13:28
House Power Monitor disconnected from PedDev. Time Connected: 1m 6s. Data Used: 48.48 KB (up) / 15.79 KB (down). Last Connected To: AP-House at -69 dBm. | Today at 13:28
House Power Monitor connected to PedDev on AP-House. Connection Info: Ch. 11 (2.4 GHz, 20 MHz), -70 dBm, 192.168.55.12. | Today at 13:26
House Power Monitor disconnected from PedDev. Time Connected: 1m 7s. Data Used: 49.25 KB (up) / 26.50 KB (down). Last Connected To: AP-House at -69 dBm. | Today at 13:26
House Power Monitor connected to PedDev on AP-House. Connection Info: Ch. 11 (2.4 GHz, 20 MHz), -70 dBm, 192.168.55.12. | Today at 13:25
```

CircuitSetup, 2025-05-21:
I updated the common config file to include this line: https://github.com/CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter/blob/997eb8fab7e3f107ea5b7af9c1a7db34280e549b/Software/ESPHome/6chan_common.yaml#L90

Which will tell you the last reset reason, and if the ESP32 is, in fact, resetting or just losing connection.

If you don't see a reset reason, the next step is to try an ESP32 with an antenna to try and increase the reliability.

PedanticAvenger, 2025-05-22:
Kewl, that's very useful.
I'm narrowing in on what seems to be an environmental challenge with the install site.
There is a lot of sheet metal from ducting in challenging locations which I believe are creating a lot of noise.  It sees lots of signal but it's messy signal.
I'm going to see how far the cables will let me move the unit around to see how that changes things.

CircuitSetup, 2025-06-02:
You may want to explore ESP32 dev boards with external antennas, like this one: https://amzn.to/4n0sAAf

PedanticAvenger, 2025-06-02:
Thanks!
Do you happen to know of any pin-compatible options with wired ethernet?  Or do they all require adapter boards?

djamps, 2025-06-22:
Check here: https://github.com/CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter/issues/204