CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter #200: Only real small values reported

ehendrix23, 2025-05-16:
Just started setup of my 6-channel energy meter. Doing testing but only getting real small values, more like "calibration" and not actual measurements.

For testing I figured to measure the power going into my printer as that cable does not have a transformer first and thus for sure I know it is going 120V into it. I'm using the 20A clamp for it and it is plugged into CT1,

I would think that as the printer is on, not even in "sleep" mode I should see some higher values but all I see is like very small mainly negative numbers for Watts.

```
INFO ESPHome 2025.4.2
INFO Reading configuration /config/esphome/energy-meter-subpanel.yaml...
WARNING GPIO5 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Detected timezone 'America/Denver'
WARNING Found and merged multiple configurations for ota platform esphome port(s) [3232]
INFO Starting log output from 192.168.1.22 using esphome API
INFO Successfully connected to energy-meter-subpanel @ 192.168.1.22 in 0.111s
INFO Successful handshake with energy-meter-subpanel-2a1c7c @ 192.168.1.22 in 0.177s
[10:50:01][I][app:100]: ESPHome version 2025.4.2 compiled on May 16 2025, 09:52:38
[10:50:01][I][app:102]: Project circuitsetup.6c-energy-meter version 1.4.2
[10:50:01][C][wifi:600]: WiFi:
[10:50:01][C][wifi:428]:   Local MAC: 08:B6:1F:2A:1C:7C
[10:50:01][C][wifi:433]:   SSID: [redacted]
[10:50:01][C][wifi:436]:   IP Address: 192.168.1.22
[10:50:01][C][wifi:440]:   BSSID: [redacted]
[10:50:01][C][wifi:441]:   Hostname: 'energy-meter-subpanel-2a1c7c'
[10:50:01][C][wifi:443]:   Signal strength: -44 dB ▂▄▆█
[10:50:01][C][wifi:447]:   Channel: 11
[10:50:01][C][wifi:448]:   Subnet: 255.255.255.0
[10:50:01][C][logger:179]:   Initial Level: DEBUG
[10:50:01][C][logger:181]:   Log Baud Rate: 115200
[10:50:01][C][logger:182]:   Hardware UART: UART0
[10:50:01][C][spi:068]: SPI bus:
[10:50:01][C][spi:069]:   CLK Pin: GPIO18
[10:50:01][C][spi:070]:   SDI Pin: GPIO19
[10:50:01][C][atm90e32:277]:     State Class: 'measurement'
[10:50:01][C][atm90e32:277]:     Unit of Measurement: 'A'
[10:50:01][C][atm90e32:277]:     Accuracy Decimals: 2
[10:50:01][C][atm90e32:278]:   Power C 'SubPanel Energy Meter CT3 Watts'
[10:50:01][C][atm90e32:278]:     Device Class: 'power'
[10:50:01][C][atm90e32:278]:     State Class: 'measurement'
[10:50:01][C][atm90e32:278]:     Unit of Measurement: 'W'
[10:50:01][C][atm90e32:278]:     Accuracy Decimals: 2
[10:50:01][C][atm90e32:286]:   Frequency 'SubPanel Energy Meter Freq A'
[10:50:01][C][atm90e32:286]:     State Class: 'measurement'
[10:50:01][C][atm90e32:268]:     Device Class: 'power'
[10:50:01][C][atm90e32:268]:     State Class: 'measurement'
[10:50:01][C][atm90e32:268]:     Unit of Measurement: 'W'
[10:50:01][C][atm90e32:268]:     Accuracy Decimals: 2
[10:50:01][C][atm90e32:277]:   Current C 'SubPanel Energy Meter CT6 Amps'
[10:50:01][C][atm90e32:277]:     Device Class: 'current'
[10:50:01][C][atm90e32:277]:     State Class: 'measurement'
[10:50:01][C][atm90e32:277]:     Unit of Measurement: 'A'
[10:50:01][C][atm90e32:277]:     Accuracy Decimals: 2
[10:50:01][C][atm90e32:286]:   Frequency 'SubPanel Energy Meter Freq B'
[10:50:01][C][atm90e32:286]:     State Class: 'measurement'
[10:50:01][C][atm90e32:286]:     Unit of Measurement: 'Hz'
[10:50:01][C][atm90e32:286]:     Accuracy Decimals: 1
[10:50:01][C][atm90e32:286]:     Icon: 'mdi:current-ac'
[10:50:01][C][uptime.sensor:033]: Uptime Sensor 'SubPanel Energy Meter Uptime'
[10:50:01][C][uptime.sensor:033]:   Device Class: 'duration'
[10:50:01][C][uptime.sensor:033]:   State Class: 'total_increasing'
[10:50:01][C][uptime.sensor:033]:   Unit of Measurement: 's'
[10:50:01][C][uptime.sensor:033]:   Accuracy Decimals: 0
[10:50:01][C][uptime.sensor:033]:   Icon: 'mdi:timer-outline'
[10:50:01][C][uptime.sensor:034]:   Type: Seconds
[10:50:01][C][template.sensor:023]:   Update Interval: 5.0s
[10:50:01][C][total_daily_energy:023]: Total Daily Energy 'SubPanel Energy Meter Total kWh'
[10:50:01][C][total_daily_energy:023]:   Device Class: 'energy'
[10:50:01][C][total_daily_energy:023]:   State Class: 'total_increasing'
[10:50:01][C][total_daily_energy:023]:   Unit of Measurement: 'kWh'
[... 236 lines cut ...]
    lambda: return id(ct1Watts).state + id(ct2Watts).state + id(ct3Watts).state + id(ct4Watts).state + id(ct5Watts).state + id(ct6Watts).state ;
    accuracy_decimals: 1
    unit_of_measurement: W
    device_class: power
    update_interval: ${update_time}
#kWh
  - platform: total_daily_energy
    name: ${disp_name} Total kWh
    power_id: totalWatts
    filters:
      - multiply: 0.001
    unit_of_measurement: kWh
    device_class: energy
    state_class: total_increasing
```

I do have the Jameco 9VAC Transformer which I ordered from CircuitSetup as same time with the clamps, board, and ESP module.

This is my first foray into energy monitoring like this AND I'm a beginner in ESPHOME not long ago setup my 1st ESPHOME for a Bluetooth proxy and HA Voice Assistant.

Hence, I am pretty sure I am doing something wrong even although I followed all the steps. 
I tried 2 different clamps with same result.

Also, it seems that the module is regularly rebooting. I added the "uptime" sensor and from there I can see that it somewhat regularly reboots based on that. Similarly I can see on my router that it regularly disconnects and the time on my router how long it then has been connect is about the same as the uptime. 
Similarly in Home Assistant I can then see as well it became unavailable.

CircuitSetup, 2025-05-16:
Are you putting the CT around the whole wire, or just the hot wire? It should be around just the hot wire. 

For rebooting, add this to your config to see what's going on:

```
debug:
  update_interval: 5s

sensor:
  - platform: debug
    free:
      name: "Heap Free"

text_sensor:
  - platform: debug
    reset_reason:
      name: "Reset Reason"
```

ehendrix23, 2025-05-20:
Apologies for later reply ... had my daughter's b-day this last weekend. :-)

On CT, yep ... I just put it around the cable itself and hence not just the hot wire. Was hoping to do some "simple" testing. :-)

It has restarted multiple times since I started monitoring.
5/16 3:36PM changed to Reset due to task watchdog. 
5/17 3:01AM changed to Reset due to task watchdog. 
5/17 3:02AM changed to Reset due to task watchdog. 
5/17 1:08PM changed to Reset due to task watchdog
5/17 1:23PM changed to Reset due to task watchdog
5/17 1:52PM changed to Reset due to task watchdog
5/17 2:31PM changed to Reset due to task watchdog
5/17 2:49PM changed to Reset due to task watchdog
5/17 7:02PM changed to Reset due to task watchdog
5/17 9:38PM changed to Reset due to task watchdog
5/17 10:17PM changed to Reset due to task watchdog
5/18 7:45AM changed to Reset due to task watchdog
5/18 8:08AM changed to Reset due to task watchdog
5/18 10:04AM changed to Reset due to task watchdog
5/18 11:03AM changed to Software reset due to exception/panic
5/18 12:04AM changed to Software reset due to exception/panic
5/18 6:03PM changed to Software reset due to exception/panic
5/18 7:16PM changed to Software reset due to exception/panic
5/18 8:20PM changed to Software reset due to exception/panic
5/19 7:54AM changed to Software reset due to exception/panic
5/19 7:56AM changed to Software reset due to exception/panic
5/20 6:51AM changed to Software reset due to exception/panic

Lowest Heap free goes is 20,000 and I see no correlation between heap going down to 20,000 and restart.

[image]

CircuitSetup, 2025-05-20:
I've seen this happen before when there is something up with the memory on the ESP32. You may want to try testing a different ESP32 with the proper pinout. 

You can also try changing this part to use the ESP-IDF:

```
esp32:
  board: nodemcu-32s
  framework:
    type: esp-idf
```

ehendrix23, 2025-05-20:
The ESP32 I got is the NodeMCU-32s I got from CircuitSetup.

Is it possible I selected the wrong ESP32 in web.esphome.io for 1st install? I belief I selected 1st one which was just ESP32.

CircuitSetup, 2025-05-20:
> The ESP32 I got is the NodeMCU-32s I got from CircuitSetup.

I'll refund you if we figure out that something is wrong with it. 

> Is it possible I selected the wrong ESP32 in web.esphome.io for 1st install? I belief I selected 1st one which was just ESP32.

No, it shouldn't matter.

ehendrix23, 2025-05-22:
Can confirm that after the change I haven't had any reboots anymore. 

Thank you!

closed 2025-05-22 by ehendrix23