CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter #202: SPI and reading issues

mikelawrence, 2025-05-31:
With your  recent updates to the ESPHome ATM90E32 component I though it was to time to update to the latest yaml. But I started having serious problems. I started receiving SPI warnings like this one:

`[16:55:23][W][atm90e32:863]: [get_phase_current_()] SPI read mismatch: expected 0x0002, got 0x2413`

But the real problem was with the data.

[image]

[image]

What fixed my problem was removing `interface: software` from [6chan_common.yaml](https://github.com/CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter/blob/master/Software/ESPHome/6chan_common.yaml) line 54.. This defaults to `any` and I believe it will choose a hardware SPI on the NodeMCU.

```
spi:
  interface: software
  clk_pin: 18
  miso_pin: 19
  mosi_pin: 23
```

mikelawrence, 2025-06-02:
Yes I switched to ESP-IDF. My first configuration was to use your 6chan_common.yaml as a package which switched to ESP-IDF. As you may already know switching platforms requires a physical connection (USB) because the partition table needs to be updated and ESPHome can't do this wirelessly (at least not yet). 

I've converted most of my ESPHome devices to ESP-IDF

rykm48, 2025-06-10:
Wait a sec. I thought the advice was the other way around to switch to arduino from esp-idf.  The latest common file now references type arduino instead of esp-idf

CircuitSetup, 2025-06-10:
> Wait a sec. I thought the advice was the other way around to switch to arduino from esp-idf. The latest common file now references type arduino instead of esp-idf

If you have more than 2 add-on boards, it's necessary to use the Arduino framework using the default spi config (which is interface: any). Hardware SPI on the esp-idf can't have more than 6 devices, and software SPI produces the results above. I'm not sure why this is happening yet.