CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter #203: Adoption overwites base config

bmbouter, 2025-06-04:
# To Reproduce

I create more or less a dummy esphome entry in home assistant with [these contents](https://raw.githubusercontent.com/CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter/refs/heads/master/Software/ESPHome/6chan_energy_meter_main_board.yaml). Just enough to get it to compile and load onto the esp32.

Then I connect to its AP, give it the creds and it connects to my network (same network as home assistant). I can then "Take Control" which I can see from the diffs on the common_yaml has recently been added (kudos for going that route btw). That shared config is the whole reason I'm taking time to write about this.

Unfortunately after home assistant takes control, it completely rewrites the original contents which removes almost all of the [original non-shared config](https://raw.githubusercontent.com/CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter/refs/heads/master/Software/ESPHome/6chan_energy_meter_main_board.yaml).

The result compiles and reflashes wirelessly, the common_yaml parts are referenced correctly, but don't work (see warnings during compilation) stating "disp_name" is missing, which indeed it is ... because taking control totally rewrites it.

This work is coming along nicely, I think there's just a bit more to go.

CircuitSetup, 2025-07-16:
Going forward, firmware can be loaded from here: https://circuitsetup.github.io/ESPWebInstaller/ , and wifi credentials set
 
This firmware is built from the new (structured) config files: https://github.com/CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter/tree/master/Software/ESPHome

By default the main config file is loaded, and the only other thing to do is generate an API key for Home Assistant. If you want the other config files to be local, they can also be copied to local directories, then included like this:

```
packages:
     common: !include Software/ESPHome/6chan_common.yaml
     main: !include Software/ESPHome/meter_sensors/6chan_main_sensor.yaml
     addon1: !include Software/ESPHome/meter_sensors/6chan_addon1.yaml
```
... etc

ehendrix23, 2025-07-17:
To clarify in this as I also added it to ESPHOME builder and did updates through that, including editing the common config file. 

So from now on we should not do that anymore and instead use your installer to do updates?
How can we then add additional sensors like the positive ones for example?

The README is not clear then on this new process.

davlaw, 2025-10-17:
OK its not really the same. Otherwise I will not be able to edit offsets and stuff.. I'll start another topic.